iNavFlight/inav #6574: inav no configuration received within 10 seconds, communication failed

omesen, 2021-02-06:
Update MatekF765 Wing and Inav to 2.6.1.
Before performing the update, check versions etc. The INAV configurator working correctly at the time ..
Install the new configurator and perform "Firmware Flasher" without problem, ok. But the INAV Configuration APP 2.6.1
 send the following message:
"No configuration received within 10 seconds, communication failed" 
I went back to Flasher twice and the problem continues. The INAV does not read the plate configuration.
Have you ever had this problem?
Any help thanks.

lysronaldavin, 2021-02-07:
i have the same problem. and if you email Andrew Newton he will help you.
apnewton@gmail.com
look at his video https://www.youtube.com/watch?v=lN3cc3B_Jwo
my name is ronald  & let me know if his advice help.
nivasyl@hotmail.com

omesen, 2021-02-08:
After struggling for a few days, I proceeded to uninstall and reinstall the suggested drivers and follow the thread below: http://www.mateksys.com/?p=5159 verbatim as I had done several times but now verify with good results the correct driver installation, this must be done when connecting the Matek to the PC.

Muchas gracias !!!!!

Enviado el: domingo, 7 de febrero de 2021 12:31 AM
Para: iNavFlight/inav <inav@noreply.github.com>
CC: omesen <omesen@indeinsa.com>; Author <author@noreply.github.com>

 

apnewton@gmail.com <mailto:apnewton@gmail.com> 
nivasyl@hotmail.com <mailto:nivasyl@hotmail.com>

closed 2021-04-07 by DzikuVx