iNavFlight/inav #6585: Release hotfix 2.6.1

digitalentity, 2021-02-08:
We have some small fixes in `master` worth merging into a hotfix branch. Please nominate PRs to cherry-pick.

Let's aim for `2.6.1` by February 21st.

digitalentity, 2021-02-08:
Baro median filtering - #6584

stronnag, 2021-02-08:
#6418 please

DzikuVx, 2021-02-08:
#6320 with #6327 and #6462

rts18, 2021-02-09:
@digitalentity Was your invitation for anyone? I'd find this #6499 and this #6477 useful early releases.

Pairan, 2021-02-09:
I would like #6459 to enter the hotfix

MrD-RC, 2021-02-09:
#6581 #6499 #6333 #6361

breadoven, 2021-02-11:
https://github.com/iNavFlight/inav/pull/6430

kasatka60, 2021-02-12:
[#6333](https://github.com/iNavFlight/inav/pull/6333) Is it possible to increase the maximum to 1-2 km?
The surrounding fields are overgrown with bushes and in case of FS, I can damage the airplane or someone .

MrD-RC, 2021-02-12:
> [#6333](https://github.com/iNavFlight/inav/pull/6333) Is it possible to increase the maximum to 1-2 km?
> The surrounding fields are overgrown with bushes and in case of FS, I can damage the airplane or someone .

That is not how SAFEhomes is supposed to work. The safehome should be within your line of sight, to make sure the landing area is empty and SAFE to land in. You can’t do this 1-2km away.

kasatka60, 2021-02-12:
That is, I have to run 1-2 kilometers for the plane to land in the field in FS mode?

MrD-RC, 2021-02-12:
> That is, I have to run 1-2 kilometers for the plane to land in the field in FS mode?

This has been discussed before with you in the safehomes issue. Safehomes is not a remote landing solution. It is to be used for alternate landing spots where you can still see the model. It won’t suite everyone. However, the safety of the general public comes first. Please stop cluttering up this unrelated issue.

shellixyz, 2021-02-14:
#6602

digitalentity, 2021-02-15:
#6603

OptimusTi, 2021-02-19:
#6598  :)

DzikuVx, 2021-02-23:
@digitalentity are we doing this?

DzikuVx, 2021-02-23:
Thanks gents, the list is closed now. We will not release in 2.6.1 PR that were not either merged to master or approved or are changing the behavior that was not a bug.
So unfortunately some of the proposals were rejected.

DzikuVx, 2021-02-23:
The full list is:

https://github.com/iNavFlight/inav/pull/6584
https://github.com/iNavFlight/inav/pull/6418
https://github.com/iNavFlight/inav/pull/6320
https://github.com/iNavFlight/inav/pull/6327
https://github.com/iNavFlight/inav/pull/6462
https://github.com/iNavFlight/inav/pull/6499
https://github.com/iNavFlight/inav/pull/6477
https://github.com/iNavFlight/inav/pull/6602
https://github.com/iNavFlight/inav/pull/6603

avsaase, 2021-02-24:
No #6430?

closed 2021-02-25 by DzikuVx