iNavFlight/inav #6591: Target for iFlight SucceX-E mini F7

ptheodosis, 2021-02-10:
## Current Behavior

I noticed that there is a target for iFlight SucceX-E mini F4 (that does NOT have a barometer)

## Desired Behavior

Could I get support for a target for iFlight SucceX-E mini F7 (that does have a barometer)?

## Suggested Solution

I think this flight controller is pretty new and might be used by a lot of folks for iNAV now that there is a barometer in the upgraded version.

## Who does this impact? Who is this for?

I'm a beginner in the quad space but plan to help add features as I familiarize myself with the code base in the future.

## Additional context

450fpver, 2021-02-13:
Looking to use succex-e mini f7 for long range quad.  A target for inav would be great.

jawa6988, 2021-02-28:
I 2nd that.  While I'm not impressed with iflight quality control or customer service response, they make lower cost options and are available on Amazon prime so you can have it in your hand within a couple days.  Sometimes you might get one with gyro oriented incorrectly but hey, once you bang your head on the table for a few hours and find you have to rotate it 270deg in firmware, it works haha  

Than again Amazon has a GREAT return policy LOL
(seriously losing my patience with iFlight to be honest and might give up on this board.  I mean that is some seriously bad quality control and seems to pretty much confirm they never even power it up for factory testing)

NSCOTTSTEVENS, 2021-03-02:
Ive built a number of  INaV quads and was surprised that this board does not have a target in INAV.  I have several Mamba F405 MK2 boards that I am trying get an I2C mag and baro rigged up for.  The 405 has multiple INAV targets and is not nearly as well suited to the software as the Succex F7.  It would be great to have a target set up in the near future.

i-build, 2021-07-27:
I definitely would appreciate it.  I am new to quads flying and was good with a DJI Mavic... built and loaded Betaflight and was disapointed by the GPS abilities... got excited when I saw what iNav can do and planned install tonight but fell flat when I saw no target for iflight_succex_e_f7....  PLEASE set it up.....!

Billbo911, 2021-08-03:
I too would like to set up the F4 with an iNav target. BF really is disappointing for me, an iNav user.

fanedi, 2021-08-04:
I think there is a specific target on iFlight site .. isn't this working ?

450fpver, 2021-08-04:
iFlight site only list a betaflight target for the succeX-E mini F7.

fanedi, 2021-08-04:
on his drive : https://drive.google.com/drive/folders/1qb6bVJd4WSG7zr1_si5DizUYNBwZRnIY

i-build, 2021-08-04:
AWESOME! Thank You!   

Why isn't this like more public 450fpver?

G9xy, 2021-08-04:
I have successfully flashed my iFlight SucceX-E mini F7 with INAV target provided by iFlight, however I have to use iFlight SucceX-E  F722 target since the "mini" version did not work (baro and acc sensors not recognized). See my post: [https://github.com/iNavFlight/inav/issues/6711#issuecomment-892387900](url)

fanedi, 2021-08-04:
Good to know ... G9xy ... and what was the version of the configurator ?

G9xy, 2021-08-04:
You would need [INAV 3.1.0 configurator](http://seyrsnys.myzen.co.uk/inav-configurator-next) for flashing SucceX-E Mini F7 FC (Chimera 5 - analog) with this particular INAV 3.1.0 target posted on [iFlight Support Google drive](https://drive.google.com/drive/u/0/folders/1r3MNTBNYSzGeRXQg45j4IJVoW1Xjk7Y0)

450fpver, 2021-08-04:
I prefer not  to use a development build (inav 3.1.0).  I will stick with Betaflight for my Explorer LR until there is an official target.

cazungas, 2021-08-06:
I can no longer find this target...is it still available?

i-build, 2021-08-09:
G9xy did you find that by using the development build you were having any significant issues or any major functionality that was not working?  I want to pull the trigger however I dont want to damage or brick my FC or go through the effort to find that there is major functionality lacking by not waiting for an official release if there is one.  Please let us know as I have 3.1.0 and the hex file you mentioned ready and waiting.  Thanks all for the feedback.

G9xy, 2021-08-10:
The INAV development build 3.1.0 for iFlight SucceX E F7 works for me without any issues, while testing I have flash it multiple times - Betaflight <---> INAV) and so far I have performed 5 flights with my Chimera 5 with this INAV 3.1.0 development build without any major issues excluding some configurations errors from my side.

My findings so far: Angle, Acro and Nav Altitude Hold modes works flawlessly, there is some drifting issues in Nav Position Hold mode (it might have to do with magnetic compas position - it is not levelled with FC). I have yet to test RTH mode.

cazungas, 2021-08-10:
This is great news. 
Any I2C faults?
I compiled my own firmware/hex and had a consistent I2C 45 fault, occasional HARDWARE HEALTH pre-arming flag set (power cycle/reboot cleared)…everything else seemed to work ok.  Hoping I don’t have the same once I download the “official” firmware.  I do have a MATEK 5883 compass tied to SCA/SDL pins.

If you have any tips/things to check, please share.

cazungas, 2021-08-10:
This was using 3.1.0

G9xy, 2021-08-10:
I have soldered iFlight M8Q-5883-GPS Module V2.0 SDA, SCL pins to FC and I2C as well, I am still testing it, magnetic compass perceive to me that it has some kind or sometimes delay/lag of showing correct heading during flight - this might be caused by I2C bus error. I have set I2C bus clock speed 400kHz(default), I will experiment with some other settings.

i-build, 2021-08-10:
![image](https://user-images.githubusercontent.com/49788937/128929277-af274ae9-8df6-426b-bd36-36dd9329793f.png
So Far so good.  Thanks

pedram28july, 2021-08-25:
> 450fpver on his inav google drive : https://drive.google.com/drive/folders/1qb6bVJd4WSG7zr1_si5DizUYNBwZRnIY
> 
> ... but iNav 3.1.0 configurator is needed !

Hello guys, I have looked for "inav_3.1.0_IFLIGHT_SUCCEX_E_F722_V1.hex" a lot, but failed to find it. Could anyone please upload it on google drive and share the link here. Or just email me at pedram28july@gmail.com. I really appreciate your kind help.

zoobs, 2021-08-30:
I have tried flashing the firmware from iFlight and it goes through the procedure, but then when complete it opens in the CLI tab and will not open any other tabs. When you flash the firmware, which target board do you select, or do you just leave this blank? I'm obviously missing something, any help would be much appreciated.
Cheers Zoobs

i-build, 2021-08-30:
> I have tried flashing the firmware from iFlight and it goes through the procedure, but then when complete it opens in the CLI tab and will not open any other tabs. When you flash the firmware, which [...]

I left it as is (which was not loaded)... only selected the file to flash.

pedram28july, 2021-08-31:
HI

You should use iNAV configurator version 3.1.0 which is still unofficial.
The latest official version is 3.0.2. I have used the 3.1.0 and it worked,
but have not tried the version 3.0.2. You can get the software from iNAV
page and try. If you still encounter problem, I will share you the 3.1.0
software.

Regards
Pedram

zoobs, 2021-09-01:
Thanks Pedram, that would be great if you could send/link me to version 3.1.0 that would be great.
Cheers Zoobs

pedram28july, 2021-09-02:
I only have Windows 64 version. I hope it is helpful to you.
 INAV-Configurator_win64_3.1.0.zip
<https://drive.google.com/file/d/1loPSCrRK4j3efRUin2UoeI8A8e60tL_F/view?usp=drive_web>

-- 

*With My Best Regards*

*-----------------------------------------------------------------------------------"COMMITTED
TO BRING CHANGES IN
EDUCATION"-----------------------------------------------------------------------------------*

*Dr. Pedram Kashiani*

*Genetics and Molecular Breeding*

FB-4-8, Level 4, Block B,

Faculty Of Technical and Vocational Education (FPTV),

Sultan Idris Education University (UPSI),

35900 Tanjong Malim, Perak, Malaysia

Tel: 006-05-4505403

Fax: 006-05-4585893

Email: ***@***.***

           ***@***.***

closed 2022-02-03 by DzikuVx