iNavFlight/inav #6597: Select variable to log to blackbox

avsaase, 2021-02-10:
## Current Behavior
Depending on the configuration, the selection of blackbox variables is fixed. Some flight controllers only have 16MB of onboard storage for blackbox logs so when logging at high rates, the logs can fill up the flash memory quite quickly.

## Desired Behavior
[ElvinC/gyroflow](https://github.com/ElvinC/gyroflow) is a project that allows you to stabilize onboard footage using the gyro data logged by the flight controller. This requires a logging rate of about 500Hz, but only of the gyro data. I have heard that Betaflight has (or will have in a coming version) the option to select which variables are logged, so that you can limit the storage space required for the logs. It would be nice if INAV has this feature as well.

## Who does this impact? Who is this for?
Anyone who want to use Gyroflow or similar software and that has a flight controller with limited storage.

avsaase, 2021-02-16:
This is the PR from betaflight: betaflight/betaflight#9726

akbarfoto, 2021-08-14:
Would be handy if this is implemented. Would just copying what the betaflight codes changes to inav work ?

- src/main/blackbox/blackbox.c
- src/main/blackbox/blackbox.h
- src/main/blackbox/blackbox_fielddefs.h 
- src/main/cli/settings.c