iNavFlight/inav #6601: Manual Altitude and Cruise heading adjusment

padcom, 2021-02-12:
## Current Behavior
Currently, when in 3D Cruise mode changing the altitude and heading is done using the pitch and roll stick.

## Desired Behavior
For long flights, with graduate climbs, descents and heading changes, it'd be great to have the option to adjust the heading and altitude just like e.g. VTX power can currently be adjusted.

For cruise, it could be 1 step 1 degree.
For altitude, it could be 1 step 10m.

Preferably adjustable using some variables. Also, if this is not already the behavior, control smoothness would influence how sharp the plane pitches when target altitude changes.

## Suggested Solution
On the Adjustments tab in the configurator add two options:

- Cruise heading Adjustment
- Altitude Adjustment

## Who does this impact? Who is this for?
This is for long-range mission pilots that prefer to steer the plane by letting it know the heading and altitude instead of manually pointing the airship into the desired direction using stick input. So, more like a real autopilot than a helper to keep the current altitude and heading that we have right now.

## Additional context
It is crucial that in the OSD the element being altered (altitude or heading) blinks while adjusting - just like any other element now.

I think since there already is a way to fly missions over waypoints it wouldn't be hard to implement it. If I understand things correctly, inav does 3D Cruise by creating a point that's in front of where the plane is heading. This would just mean modification of that point left/right. Also, if I get it right, the altitude hold just holds the currently set as target altitude which means the adjustment would just mean changing whatever the target altitude is.

padcom, 2021-02-13:
When flying long range missions it is crucial to keep the altitude in check and to perform altitude changes gradually instead of climbing with max throttle. This is because battery needs to be conserved or otherwise the flight will be shorter.

Although the heading change isn't all that important (the way of changing the heading works quite well), long altitude changes aren't easy to implement in the current state.

The way I've done it is by altering the pitch curve in OpenTX so that I can manually move the center point up and down. To make the plane start/end the climb gradually I have configured OpenTX so that the direction (climb or descent) is controled by a 3 position switch and the amount of climb/descent is controlled by a pot. This means the procedure for climbing and/or descending is as follows:

- move the pot all to the left
- flick the 3 position switch to the desired state (up for descent, down for climb - to correspond with the desired stick movement)
- increase the pot to get the desired stick deflection to keep the plane climbing or descending
- once the target altitude is almost reached gradually decrease the pot to zero
- flip the 3 position switch to the middle position

It works, but it is more of a hack than actual navigation controls.

The way to change the pitch curve in OpenTX involves creating a GV that stores the pot value and then, depending on the 3 position switch, setting that GV to either negative pot value or positive pot value. Then the center point of the channel is set to that GV. This has this nice property that the stick still allows for full control surface deflection in both directions allowing emergency control when climbing/descending.

padcom, 2021-02-13:
The main reason I'd really like to be able to just set the target altitude and heading instead of using stick controls is to be able to set them more precisely. Right now, with the stick deflection adjustment, it is more of a "let's see if I can hit that value" game which in turn means the pilot is more occupied with trying to hit the target value than observing the path ahead to avoid collision.

So you might say, in long-range semi-automated flights, this would be a feature to minimize the risks.

wx4cb, 2021-03-22:
> The main reason I'd really like to be able to just set the target altitude and heading instead of using stick controls is to be able to set them more precisely. Right now, with the stick deflection [...]

yes but the pilot should always be paying attention to the surroundings, and that said, in log semi automonous flying (eg wp misson) you don't want to do that because what happens if you then have to avoid something, the mission is going to want to try and go back to that point. so you would constantly have to be adjusting anyway. 

as for "trying to hit that value" if you're going more than a few hundred feet in a straight line, you're going to have to adjust anyway. especially in a crosswind scenario. because "hitting" a heading of say, 087 is not necessarily going to get you to the same point in space where you want to go. you could be trying to fly 087 but actually flying sideways

padcom, 2021-03-23:
yes, but I think you're missing my point here. The current state of things is that if you're flying in cruise mode the heading lock is already in please. The only thing that is missing is the expected altitude hold adjustment and allowing those values to be displayed and adjusted in the Adjustments tab. It's not such a big deal I'd hope since all the ingredients are already in place.

And most of all it is done especially so that the pilot can be aware of its surroundings while flying long-range rather than constantly observing the altimeter and trying to hit a target.

closed 2022-02-10 by DzikuVx