iNavFlight/inav #6617: Display telemetry from arming location. not SafeHome

Jetrell, 2021-02-17:
@tonyyng I would like to run an idea past you.

At present, any `SafeHome` within 200m of were arming occurs, now becomes the new arming location.
This can be seen on the OSD via `Distance to home and the Home arrow.`

I was try to work out the discrepancy today, when a friend and I were calling out distance from home, while fly together.

We know from the arming page, that a `SafeHome` is detected and it will be the landing location in an emergency.
But would it be possible to keep the Arming location as the reference for `Distance to home and the Home arrow` instead of the SafeHome? to avoid confusion in various cases.

tonyyng, 2021-02-17:
The way I use it, my safehomes are typically 100m or less away from the arming point, so it doesn't make much of a difference if the OSD shows home or safehome.  I'm aware that it is using a safehome.  However, given that the safehome can now be up to 650m away, and that the safehome was intended to be used for an emergency, I agree that having the safehome being always shown as home isn't ideal.

While looking at the code, there are a number of OSD elements that make use of the home location: crosshairs, radar, home map, home direction and home distance.  These would use the safehome location instead of the original home location.

I could address this by not replacing the original home location with the safehome location until RTH is activated.  With this approach, the OSD information would almost always show the original home.  It would use the safehome's location only after RTH is activated.  If RTH is deactivated, the safehome would remain as home, unless I can find an obvious way to detect the transition from RTH ON to RTH OFF.

This opens up other scenarios.  What if there were no safehomes, within the safehome_max_distance at the time of arming, but at the time RTH is activated, there was a safehome near the current location?  Should we divert there or continue to the arming home?  If I've flown out 5km and there was a safehome at 4.5km, I'm not sure I would want to land there.

My inclination is to stick with the current safehome selection approach (selected during arming), and then divert there with RTH.  However, as I've found out, there are lots of different opinions about how safehomes should work.  I'll put this out there for discussion.  I can't promise I'll be able to implement what is discussed (given my knowledge of the code), but I'll listen.

@MrD-RC, @Mateyhv, @stuwe911, @stronag, @digitalentity, @giacomo892, @kasatka60, @rts18, @DzikuVx: Comments?

Jetrell, 2021-02-17:
> I could address this by not replacing the original home location with the safehome location until RTH is activated. With this approach, the OSD information would almost always show the original [...]

Thanks for taking the time to look at the idea. You brought up some good points. 

I like the idea of allowing RTH/FS to reset the armed location coordinates, to the (stored near launch) SafeHome coordinates.
Something that would need to be addressed if doing it that way Would be to leave a buffer zone near the Armed location. So an RTH/FS test could be performed after launch. To ensure RTH is working, before a long range trip.

The only other way I could think of. Would be to introduce a second variable for the Armed location, that is introduced when a SafeHome is detected?
This variable could drive the telemetry data, while calculating angular offset difference between the Armed coordinates and SafeHome coordinates.

tonyyng, 2021-02-18:
> I like the idea of allowing RTH/FS to reset the armed location coordinates, to the (stored near launch) SafeHome coordinates.
> Something that would need to be addressed if doing it that way Would be [...]

I don't see a simple to do this.  If I proceed with my proposed approach, when you enable RTH (to test or otherwise), the original arming home would be replaced with the selected safehome for the remainder of the flight.  This would negate what the changes are trying to achieve.  However, beyond initial flights after setting up a plane, I can't say I've ever found the need to "test" RTH.

tonyyng, 2021-02-18:
iNav has an option to reset home during flight.  @rts18 suggested using that as a means of setting a safehome that is over [1km away.](https://github.com/iNavFlight/inav/issues/6301#issuecomment-778477601)

If this is activated during flight, I think I would also clear any safehome that was saved during arming.  RTH would return to the new home location.

jreise-d, 2021-02-18:
There is another point to consider with safehome. If you use _failsafe_min_distance_ and _failsafe_min_distance_procedure = NONE_ to prevent the motor from starting on the ground when the rc transmitter is accidentally switched off, this no longer works if a safehome is nearby. I think this is a safety issue that should be named in the safehomes explanations.

tonyyng, 2021-02-18:
> There is another point to consider with safehome. If you use e.g. _failsafe_min_distance = 5000_ and _failsafe_min_distance_procedure = NONE_ to prevent the motor from starting on the ground when the rc transmitter is accidentally switched off, this no longer works if a safehome is nearby. I think this is a safety issue that should be named in the safehomes explanations.

Just to be certain I understand, the situation you've outlined is: 
- failsafe_min_distance set to 50m
- failsafe_min_distance_procedure set to NONE
- armed with a safehome chosen that is beyond 50m
- turned off the transmitter
This causes the failsafe_min_distance parameters to be ignored, and the normal failsafe procedure to be activated.  Is this right?

I'm not certain, but if I implement the logic I've described above, the home location wouldn't get replaced with the safehome until after the logic that checks the failsafe_min_distance is done, so it should behave the way you expect.

This is a real outlier use case and it seems like there is a flaw in the implementation of the RTH procedure.  It the aircraft is sitting stationary on the ground, it shouldn't activate.

Jetrell, 2021-02-18:
@tonyyng I haven't looked through the code. And don't know the complexity of what I'm asking. But I'll throw this out there. 
Would it be possible to have two sets of variables. One for the Armed location coordinates and the other for SafeHome.
i.e. One drives the navigation and the other drives the telemetry data?

tonyyng, 2021-02-18:
Before safehomes, there was just the one home location used for both navigation and telemetry.
The current implementation of safehomes, replaces the home location when arming and it continues to be used for navigation and telemetry.
What I'm proposing above is to keep track of the safehome during arming, keep the arming location as home for navigation and telemetry, and only replace arming location with the safehome when RTH is activated.

I'm not familiar enough with the code to consider making the logic for navigation and telemetry to use separate locations.  Too complicated for me, but I've cc'd a bunch of other experienced devs.  Perhaps they have some ideas.

jreise-d, 2021-02-18:
Again, because I may not have made it clear above.

My current situation is as follows:

- failsafe_min_distance set to 50m
- failsafe_min_distance_procedure set to NONE
- Armed, with no safehome nearby.

I am flying normally and land at home, disarm the plane and accidentally turn off my transmitter before turning off my plane. The aircraft is within 50m of home, so FS does not kick in. The motor does not turn, it's safe.

And now the same situation but arming with safehome 150m away. Now after flying and normal landing at takeoff point and disarming when I turn off my tx before aircraft, failsafe_min_distance is calculated from safehome and not from my takeoff point. So failsafe_min_distance is exceeded and therefore ignored and FS occurs and the motor starts spinning.

This is my current problem with safehome as a simple replacement of homepoint.

tonyyng, 2021-02-18:
> And now the same situation but arming with safehome 150m away. Now after flying and normal landing at takeoff point and disarming when I turn off my tx before aircraft, failsafe_min_distance is [...]

With the proposed changes, the home location won't be replaced by the safehome location, until RTH is invoked.  The check for min distance would work as before.

Jetrell, 2021-02-18:
From your question above. I have seen RTH/FS fail between flights, due to a barometer hardware issue. That's why I always do FS check before a long range flight.

> 
> I think I've identified the point when RTH kicks in (navOnEnteringState_NAV_STATE_RTH_INITIALIZE), when I would replace the home location with the selected safehome. In theory, if you cancelled RTH, I could revert home back to the original arming location, but how I detect "leaving" RTH isn't obvious to me.

If you can find an exit point. That would be great. It would allow for an RTH/FS test, while not resetting the Arming location.

Jetrell, 2021-02-18:
From your question above. I have seen RTH/FS fail between flights, due to a barometer hardware issue. That's why I always do a FS check before a long range flight.

> I think I've identified the point when RTH kicks in (navOnEnteringState_NAV_STATE_RTH_INITIALIZE), when I would replace the home location with the selected safehome. In theory, if you cancelled RTH, [...]

If you can find an exit point. That would be great. It would allow for an RTH/FS test.

tonyyng, 2021-02-19:
@Jetrell Here's the logic that checks the failsafe_min_distance.

```
   // Craft is closer than minimum failsafe procedure distance (if set to non-zero)
    // GPS must also be working, and home position set
    if ((failsafeConfig()->failsafe_min_distance > 0) &&
        ((GPS_distanceToHome * 100) < failsafeConfig()->failsafe_min_distance) &&
        sensors(SENSOR_GPS) && STATE(GPS_FIX) && STATE(GPS_FIX_HOME)) {
        // Use the alternate, minimum distance failsafe procedure instead
        return failsafeConfig()->failsafe_min_distance_procedure;
    }

```

It is using GPS_distanceToHome, which is the distance to the current home.  As you've pointed out, this could be a safehome, or a different GPS location that was set during flight using the "home reset" feature.  Either would likely make the logic above misbehave.

As part of this change, I should be able to save the original arming location and change the logic above to use the distance to the arming location, instead of distanceToHome.

stuwe911, 2021-02-19:
On a LR flight within a formation I've used the home-offset feature but my wingman did not. when shouting out distances, altitude and speeds to each other we thought we were safe. the HD video after the flight revealed that we had a near miss by only a few meters vertically. I regarded this as my own mistake and don't blame the software for it. of course it will show a different distance on the two OSD's if i set my "home" 300m further out than my buddy that had the default "armed" position as a reference. Since formation LR flying only makes up about 5% of my activity this is neglectable for me but this might be different for other pilots. On the other hand I've once learned that you should trust your instruments and it can become confusing if you have to add or substract numbers during a flight especially in stress situations. Hence I like Tony's proposal to use the "armed" location as a reference until RTH is activated. Once RTH is deactivated it should go back to referencing to the "armed" location. 

Like Jetrell I do test my RTH regularly especially when going on a longer mission or with newer planes, or after some major changes. Flip the switch and let the plane turn around. check if it takes the correct heading, speed and altitude. If all looks good, I disable the RTH switch and turn around towards my initial heading. However this is not a priority to me personally - I'm just happy that the 200m limit is gone :)

breadoven, 2021-02-19:
> In theory, if you cancelled RTH, I could revert home back to the original arming location, but how I detect "leaving" RTH isn't obvious to me.

Check the Navstate:

`navGetStateFlags(posControl.navState) & NAV_AUTO_RTH`

Also `posControl.flags.forcedRTHActivated` will tell you if failsafe RTH is active.

`selectNavEventFromBoxModeInput `in `navigation.c` is useful for this info.

wilco1967, 2021-02-20:
This is similar to #6541

tonyyng, 2021-02-20:
> This is similar to #6541

Yes, I think an option like "safehome_on_failsafe_only" should do the trick (along with the changes proposed above).

breadoven, 2021-02-22:
It would be useful to be able to disable Safehomes using the OSD CMS for the situation where the safehome isn't safe when you get to the flying site leaving a choice of either not flying or finding another arming/launch point that is safe in itself. Could use the options given in https://github.com/iNavFlight/inav/issues/6541 or just safehome_max_distance (reduce below nearest safehome distance) ?

tonyyng, 2021-02-24:
I've made the following changes and bench tested them.

- The safehome does not replace the arming location until it is needed.  Also, if you cancel RTH or recover from RX failsafe, the original arming location is restored as the home location.
- New `safehome_usage_mode` option:
  -   OFF - Safehomes are not used
  -   RTH (default) - Safehomes will be used for both manual RTH and RTH RX failsafe
  -   RTH_FS - Safehomes will be used only forRX failsafe
- The Safehome usage mode can be set using the OSD menu
- New OSD message when RTH to safehome is in progress: "DIVERTING TO SAFEHOME"

It will be awhile before I can flight test them.  We're probably two months away from flying season at my location.

Jetrell, 2021-02-25:
@tonyyng looks good!
Have you made a build with it?
I can try it on my test rig - `Omnibus F4pro`.

breadoven, 2021-02-25:
Looks good. Just a couple of thoughts:

1. Doesn't appear to check if landing is set for RTH/RTH FS. Would be useful to add in a landing check (`navigationRTHAllowsLanding() `) to control whether or not it only diverts to the safehome if setup to land otherwise it would return to the arming home and loiter. Certainly makes sense to do this for a normal RTH where you have control, For a FS RTH perhaps better to loiter over the safehome (a safer place to crash when the battery runs out). Could be yet another option ...
2. Would it be better to check if a safehome is available even when set to OFF and flash a warning on the OSD arming screen to show that a safehome is available but that safehomes is set to OFF ? This would be useful where you've disabled safehomes using CMS at a particular site where a safehome can't be used and then forget to re-enable next time you're flying.

tonyyng, 2021-02-25:
@breadoven Thanks for taking the time to look over the code.

> 1. Doesn't appear to check if landing is set for RTH/RTH FS. Would be useful to add in a landing check (`navigationRTHAllowsLanding() `) to control whether or not it only diverts to the safehome if [...]

The safehome logic isn't involved in the decision to land when it gets to home.  It only decides whether home is the safehome or arming location.  

If you are proposing for the following situation:
- safehome_usage_mode = RTH and safehome found and normal RTH in progress
  - nav_rth_allow_landing = TRUE -> divert to safehome
  - nav_rth_allow_landing = FALSE -> continue to arming point

I'm not sure I like this.  It would add a confusing connection between the two settings.
I think you could get what you want by setting safehome_usage_mode = RTH_FS at the same time you are setting nav_rth_allow_landing to FALSE.

> 2. Would it be better to check if a safehome is available even when set to OFF and flash a warning on the OSD arming screen to show that a safehome is available but that safehomes is set to OFF ? [...]

This makes sense, and I've changed the logic to display ```SAFEHOME FOUND; MODE OFF``` when this happens.

tonyyng, 2021-02-25:
> @tonyyng looks good!
> Have you made a build with it?
> I can try it on my test rig if you can supply me with build target - `OmnibusF4pro`.

I think I've built this based on release 2.6.1 adding the safehome changes, but I'm a git novice.  Check the 2.6.1 release notes and use at your own risk.

[OMNIBUSF4PRO.hex.zip](https://github.com/iNavFlight/inav/files/6044040/OMNIBUSF4PRO.hex.zip)

breadoven, 2021-02-25:
> The safehome logic isn't involved in the decision to land when it gets to home. It only decides whether home is the safehome or arming location.
> 
> If you are proposing for the following situation:
> 
> * safehome_usage_mode = RTH and safehome found and normal RTH in progress
>   
>   * nav_rth_allow_landing = TRUE -> divert to safehome
>   * nav_rth_allow_landing = FALSE -> continue to arming point
> 
> I'm not sure I like this. It would add a confusing connection between the two settings.
> I think you could get what you want by setting safehome_usage_mode = RTH_FS at the same time you are setting nav_rth_allow_landing to FALSE.
> 
> > 1. Would it be better to check if a safehome is available even when set to OFF and flash a warning on the OSD arming screen to show that a safehome is available but that safehomes is set to OFF ? This would be useful where you've disabled safehomes using CMS at a particular site where a safehome can't be used and then forget to re-enable next time you're flying.
> 
> This makes sense, and I've changed the logic to display `SAFEHOME FOUND; MODE OFF` when this happens.

As far as normal RTH is concerned that's the logic I was thinking of. However, as you say it can be controlled by changing settings. Auto landing would only be a failsafe option for me, better than nothing unless you have somewhere really flat to land, so I'd only use RTH_FS.

And thanks for adding the other point.

Jetrell, 2021-02-26:
> 
> I think I've built this based on release 2.6.1 adding the safehome changes, but I'm a git novice. 
> 

@tonyyng Thanks 

I went out for a test run. 
Firstly. This was tested with a quad. I would normally use this feature with a fixed wing. But I think its relevant for a quad if the takeoff location was very close to objects or people. Although it wouldn't need to be set as far away.

I set the `Safehome` at 23m from the Home/Arming location. With the default `Safehome` settings.
 
These are some CLI settings on this quad -
```
failsafe_procedure = RTH
failsafe_min_distance_procedure = RTH
nav_min_rth_distance = 1000
nav_disarm_on_landing = ON
nav_rth_allow_landing = ALWAYS
```

* On arming. The `Safehome` was detected and displayed on the arming screen.
* The telemetry data (_distance from home_) is now based on the Arming location. Not `Safehome `as before.
* When the RTH mode switch was enabled. The _distance to home_, is now displayed as the distance to the `Safehome`.
* `Diverting to Safehome `also appears on the OSD messages.
*  When RTH is disabled again, the _distance to home_ reverts back to the Armed/home location.
* It also landed and the _Nav System_ disarmed it.

There is only one edge case bug that I found. It has to do with the modes tab selection.
Return to home can be selected by two sources. 
* NAV RTH
* Failsafe

Both these mode selections will trigger RTH by a switch.  
_Although Failsafe mode selection has the added advantage of not just triggering RTH by the switch input.  But also simulates an RX Failsafe signal loss or RX wiring failure._ 

When Failsafe mode was selected for return to home and the test switch was enabled. The `distance to home` would reset to Zero and the quad would initiate a landing procedure (wherever it was).
If the switch was disabled again, the _distance to home_ would then revert back to the distance to the Arming location.

tonyyng, 2021-02-26:
@Jetrell It was a good idea to test with a quad.  As opposed to fixed wing, you don't have to wonder if it is going to land in one piece.

With the bug you described, I'm not sure that is safehome logic causing that behavior.  The setting for nav_min_rth_distance would cause it to immediately land if you were within 10m of home.

If you were outside nav_min_rth_distance, could you try to reproduce the bug without safehomes in use?   One of the following approaches should work:
- set safehome_usage_mode to off
- set safehome_max_distance to 1
- don't define any safehomes

If the behavior still happens when safehomes isn't involved, I don't think it is bug that safehomes introduced.

Another test you could do is to trigger RTH by the switch and then RX failsafe by actually turning off your radio.  That would indicate whether the "failsafe mode" really does simulate RX failsafe accurately.

Jetrell, 2021-02-26:
@tonyyng  Hi, I did actually test all those things. I didn't want to get too long winded in the explanation though:-) 

My tests were always outside `nav_min_rth_distance`. e.g. 60m from arming and 83m from `Safehome`. 
I believe its a logic bug; because _distance from home_ on the OSD would go to Zero when FS mode was triggered by the switch. That's why the quad went into auto land. Because it then thought it was within `nav_min_rth_distance`.
But as soon as the Failsafe mode switch was disabled. The OSD _distance to home_ figure, returned to its original `Armed location` value.

I know it didn't behave this way with Safehome turned OFF. Because I always used Failsafe mode as means of RTH before `Safehome ` existed.

The time that I tested auto land. Was done by turning OFF my radio control.  It would fly to the `Safehome` location and land.
But its not conclusive concerning Failsafe. Due to my Crossfire using `FS POS `and not `FS CUT`
`FS Pos`, effectively just moves the channel switch to the Failsafe mode active position. Instead of sending an RX signal loss data byte. But the effect should be the same in the way iNav see;s it.

tonyyng, 2021-02-26:
@Jetrell Let me make sure I understand the bug. 

You have a switch mapped to enable Failsafe mode and when you flip that switch, your quad immediately lands in its current location.   

This is a concern and I want to confirm that my changes didn't cause this behavior.

> I know it didn't behave this way with Safehome turned OFF. Because I always used Failsafe mode as means of RTH before Safehome existed.

So:
Prior to release 2.6:  No safehome feature, Failsafe mode RTH returns to home
Test version I gave you: 2.6.1x:  Safehome V2, Failsafe mode RTH lands in current location (assume with safehome enabled)

What I'd like to know is if Failsafe mode RTH lands in the current location with safehome disabled.

Jetrell, 2021-02-26:
> 
> 
> 
> You have a switch mapped to enable Failsafe mode and when you flip that switch, your quad immediately lands in its current location.

  Yes, that's correct.

> So:
> Prior to release 2.6: No safehome feature, Failsafe mode RTH returns to home
> Test version I gave you: 2.6.1x: Safehome V2, Failsafe mode RTH lands in current location (assume with safehome [...]

No it doesn't. 
If Failsafe mode is set for RTH. And` Safehome` is OFF. The arming screen says - `SafeHome Found : Mode OFF`
Once the FS switch is enabled, the quad`returns to home` and lands at the armed location.  I just went out and tested it with the V2 `Safehome` release.

tonyyng, 2021-02-27:
I'm confused.  Can you clarify this statement?
> `safehome_usage_mode = RTH` or `safehome_usage_mode = RTH_FS` both act the same, regardless of whether Failsafe mode or Nav_RTH mode is used to trigger them.

This is what I think you've told me so far:
- `safehome_usage_mode = RTH`, Trigger Nav_RTH -> returns to safehome
- `safehome_usage_mode = RTH`, Trigger Failsafe mode -> lands at current location
- `safehome_usage_mode = RTH_FS`, Trigger Nav_RTH -> returns to arming point
- `safehome_usage_mode = RTH_FS`, Trigger Failsafe mode -> lands at current location

Is that correct?

I setup my radio/FC with the FAILSAFE_MODE on a switch and I can't reproduce your behavior on the bench.  When I flip the switch for FAILSAFE_MODE, I see `DIVERTING TO SAFEHOME` and the distance to the safehome in my OSD.

I've looked over the code and don't see anything obvious that would cause the behavior you've described.  However, my knowledge of this code is at a novice level, so I could be missing something.

At this point, before I can submit my changes, I need to wait on:
- help with the code from the devs (not sure how I get this)
- more beta testers
- my own flying season to start so I can test myself (if I can't reproduce your problem, not sure about this)

Maybe you could bench test this on your fixed wing?

Jetrell, 2021-02-27:
> 

> This is what I think you've told me so far:
>
>     * `safehome_usage_mode = RTH`, Trigger Nav_RTH -> returns to safehome
>
>     * `safehome_usage_mode = RTH`, Trigger Failsafe mode -> lands at current [...]
  Yep. That's correct.  
  It is odd that you can't see it on the bench.  I can do another test with a fixed wing.  Target MatekF411_SFTSRL2

  Here is a short video clip from the DVR. This may help you visualize it. 
[Safehome RTH_FS - FSmode switch trigger.zip](https://github.com/iNavFlight/inav/files/6055520/Safehome.RTH_FS.-.FSmode.switch.trigger.zip)

tonyyng, 2021-02-27:
Looking at video, I can see the message: DIVERTING TO SAFEHOME.  
[image]

This message is displayed based on a flag that is set when home has been set to the safehome (determined during arming).  This means my logic has executed that detects that RTH or RTH_FS has occurred and replaced the arming location with safehome.

I do see that the distance to home changes to zero.

The only reasons I can think of are:
- Your quad is right over the safehome position - seems unlikely
- something else in the firmware code has reset the home location to the current location.  Is this what an emergency landing is supposed to do?

Here is the Matek firmware with the same changes: 
[inav_2.7.0_MATEKF411_SFTSRL2.zip](https://github.com/iNavFlight/inav/files/6055794/inav_2.7.0_MATEKF411_SFTSRL2.zip)

Jetrell, 2021-02-27:
> 
> The only reasons I can think of are:
> 
>     * Your quad is right over the safehome position - seems unlikely
      50mtrs away, as the OSD shows before FS is triggered and after it is disable.

>     * something else in the firmware code has reset the home location to the current location. 
>       Is this what an emergency landing is supposed to do?
> 
       
  From the CLI wiki -
  **nav_min_rth_distance** 
```
Minimum distance from homepoint when RTH full
procedure will be activated [cm]. Below this distance, the
mode will activate at the current location and the final
phase is executed (loiter / land). Above this distance, the
full procedure is activated, which may include initial
climb and flying directly.
```

It must be landing because the home value becomes zero.  Which places the quad within the `nav_min_rth_distance` set value.

As you said - _something else in the firmware code has reset the home location to the current location._ 

Thank you Tony.  For the time you are putting in to track this down :+1:

tonyyng, 2021-02-27:
I see some code that is checking the distance to home during RTH and checks that the distance isn't increasing.  If it does, it goes into emergency landing mode.  If the safehome is farther from the quad than the arming location, the distance to "home" would increase when home is set to the safehome.

Can you try something with a quad test flight?  Fly out so that the safehome is between you and the quad (it might be easier to move the safehome), and then trigger Failsafe.   If this successfully returns to the safehome, that would confirm that logic is what is causing the behavior you are seeing.

breadoven, 2021-02-28:
Check nav_rth_abort_threshold. Default is 500m so distance to home would have to have increased by 500m more than the minimum distance achieved during that RTH event to trigger emergency landing. You can also disable nav_rth_abort_threshold check by setting it to 0 so it no longer checks if RTH distance is increasing.

Jetrell, 2021-02-28:
>
> 
> Can you try something with a quad test flight? Fly out so that the safehome is between you and the quad (it might be easier to move the safehome).

These are the results I had when I ran tests with Safehome between the Armed location and the trigger location. They were the  same as the results before.
* `safehome_usage_mode = RTH_FS`, Trigger Nav_RTH - returns to arming point.
* `safehome_usage_mode = RTH_FS`, Trigger Failsafe mode - lands at current location.

I also tried another variant. `nav_rth_allow_landing = Never`. 
It would reset the D.T.H. to zero, then climb to the fly home altitude and just park there. Saying DIVERTING TO SAFEHOME and HOVERING.

That all sounds the same. Until I tested the fixed wing target you posted. 

I used - `safehome_usage_mode = RTH_FS` with Failsafe mode as the trigger. As I did with the Quad.
The test was done on the ground. With me walking around to simulate the plane in flight.
**Everything worked correctly!**

So I did a compare of the two _diff all_ files. And only found two possible discrepancies. One was as @breadoven and you stated. And the other was `nav_rth_allow_landing = Never`.
Altering both, still made no difference on the quad.

Is it possible that something is buried in `failsafe.c` or `navigation_multicopter.c` that effects the way iNav handles FS differently with MR's, compared to FW's?

breadoven, 2021-02-28:
This isn't an Emergency Landing because there's no message. Given that home distance resets to 0  I'm guessing the problem is related to:
https://github.com/iNavFlight/inav/blob/cb2fccb7a0a0c012337726e6f7d7ed24c6b822b7/src/main/navigation/navigation.c#L1142

The fixed wing difference possibly caused by:
https://github.com/iNavFlight/inav/blob/cb2fccb7a0a0c012337726e6f7d7ed24c6b822b7/src/main/navigation/navigation.c#L1123

Also home distance is reset to 0 when the home position is set:
https://github.com/iNavFlight/inav/blob/cb2fccb7a0a0c012337726e6f7d7ed24c6b822b7/src/main/navigation/navigation.c#L2303

And only gets recalculated in:
https://github.com/iNavFlight/inav/blob/cb2fccb7a0a0c012337726e6f7d7ed24c6b822b7/src/main/navigation/navigation.c#L2425

So could be a sequencing issue related when `checkSafeHomeState()` is called during failsafe or a general issue related to changing the home position when armed, i.e. changing the home position when armed is precluded for FS and RTH modes if the Home Reset mode is used:
https://github.com/iNavFlight/inav/blob/cb2fccb7a0a0c012337726e6f7d7ed24c6b822b7/src/main/navigation/navigation.c#L2411

tonyyng, 2021-02-28:
> Also home distance is reset to 0 when the home position is set:
> https://github.com/iNavFlight/inav/blob/cb2fccb7a0a0c012337726e6f7d7ed24c6b822b7/src/main/navigation/navigation.c#L2303
>
> And only gets [...]

Based on your comments, I see that sequencing is likely the issue.  I'll try reinitializing home distance/direction after changing the home location.  That would hopefully prevent other code from seeing that home distance is 0, and thinking it has reached home.

> The fixed wing difference possibly caused by:
> https://github.com/iNavFlight/inav/blob/cb2fccb7a0a0c012337726e6f7d7ed24c6b822b7/src/main/navigation/navigation.c#L1123

This code would prevent fixed_wing from activating right?  The problem @Jetrell is seeing is that multirotors aren't working.

tonyyng, 2021-02-28:
@Jetrell This is the fw with updates.  Please repeat the tests with both fixed wing and quad.

Thanks
[inav_2.6.12.zip](https://github.com/iNavFlight/inav/files/6057015/inav_2.6.12.zip)

Jetrell, 2021-03-01:
@tonyyng Good news. It now works with both MR's and FW's

Although. There is a variation in the way `safehome_usage_mod` works. RTH and RTH_FS act the same. 

* `safehome_usage_mode = RTH` , Nav_RTH mode - The model will return to the Safehome under mode trigger and TX off signal loss.

* `safehome_usage_mode = RTH_FS` , Failsafe mode - The model will return to the Safehome under mode trigger and TX off signal loss.

 Other combinations prevent Safehome from working except this -

* `safehome_usage_mode = RTH_FS`, Nav_RTH mode and CRSF FS-CUT (instead of FS-POS)
The model would fly to the Armed location with switch trigger and fly to the Safehome with TX off signal loss. 
 
This one is more like the way you designed - `safehome_usage_mode = RTH` to work.

I have written up a document for the Wiki. With recommendation for RTH/Failsafe setup surrounding Safehome V2 with the last changes. If your happy to leave it as is, Then I can post it here for you. 

Personally. I feel that its exceptable to have it fly to the Safehome under both Switch trigger and TX signal loss. There maybe a user case when they want the model to fly back to the Armed location via Switch. But why would they bother using Safehome in the first place. 

I also think its good to leave both RTH and RTH_FS. Even though RTH doesn't totally work as it should.
 Mainly because some users are set in their ways of prefering to use Nav_RTH mode. Even though Failsafe mode has always been a safer choice.
WDYT

tonyyng, 2021-03-01:
So FS-CUT with TBS CRSF will result to RTH mode in iNav and if safehome_usage_mode is RTH_FS, won't use the safehome.  Is that right?

Options are:
- use safehome_usage_mode = RTH
- use FS-POS, configured to set the mode to FAILSAFE

>Personally. I feel that its exceptable to have it fly to the Safehome under both Switch trigger and TX signal loss. There maybe a user case when they want the model to fly back to the Armed location via Switch. But why would they bother using Safehome in the first place.

If you read some of the earlier posts, people like to use RTH to bring their aircraft to the arming point.

> I have written up a document for the Wiki. With recommendation for RTH/Failsafe setup surrounding Safehome V2 with the last changes. If your happy to leave it as is, Then I can post it here for you.

Sure.

Thanks for your testing.  I don't think I'm going to make any more changes to the behavior.  

Thanks also to @breadoven for his help in diagnosing the problem.

Jetrell, 2021-03-01:
@tonyyng 

> 
> 
> So FS-CUT with TBS CRSF will result to RTH mode in iNav and if safehome_usage_mode is RTH_FS, won't use the safehome. Is that right?
> 
No.
CRSF FS-CUT, and I imagine any other RX that outputs NO pulses. Will accomplish Safehome return via FS and normal return to arming location via Switch.

The document explains the options. 
I think its good for a PR ??

[Safehome v2 doc.pdf](https://github.com/iNavFlight/inav/files/6058910/Safehome.v2.doc.pdf)

I know the variations are getting hard to wrap your head around. I wrote down each combination of settings on each DVR file. Otherwise I would have lost were I was up too.. Lol

breadoven, 2021-03-01:
> Personally. I feel that its exceptable to have it fly to the Safehome under both Switch trigger and TX signal loss. There maybe a user case when they want the model to fly back to the Armed location [...]

You seem to have an odd understanding of the modes here Jetrell. I don't see switched RTH as a failsafe mode, it's just RTH, the same as any other NAV mode you might use whilst you have control. The fact Failsafe can be setup to use RTH doesn't make Failsafe a RTH mode which is what you imply when you say users prefer using switched RTH to achieve RTH rathe than Failsafe set up to use RTH. Also as I understand it switched Failsafe mode should act exactly the same as Failsafe caused by loss of signal.

I was never sure why there is a Failsafe mode other than for testing Failsafe without actually turning the TX off. It can be used to trigger failsafe on some radios that don't have No pulses (eg Flysky doesn't) or the ability to set the Failsafe on the Rx but only if an unused Aux channel defaults to some setting when the signal is lost that can be used to trigger the mode. 1 Aux channel on my radio consistently goes from 1500 to >2000 us when the signal is lost and back to 1500 when it's regained. I did use this use this to trigger Failsafe mode until I realised you could set the throttle signal to a value below 1000 by a configured failsafe setting on the Rx which triggers Inav Failsafe using the invalid signal method (`rx_min_usec`). 

I use RTH all the time to bring the craft back even when I can see it and could fly it back manually. I usually cancel the RTH when it gets close enough ... which is why I wouldn't want it flying to a Safehome, I want it to come back to where I launched from. The way I normally fly I would only use Safehomes for the situation where I've lost the control link, it's in Failsafe and I can't control the landing. So you need somewhere better/safer than the launch point to allow it to autoland. I can land a plane no problem from the places I launch from (generally uneven places with obstacles) but an auto land there would never work, it would crash every time. However, there's usually somewhere not too far away which has a better chance of auto landing.

Jetrell, 2021-03-01:
@breadoven Have you read through the PDF document I left a couple of posts back?
It certainly is possible to do as you requested. To use RTH mode switch to fly back to the Arming location.  And a signal loss FS would bring it back to the Safehome.  

I don't believe I referred to RTH as a failsafe mode. It is simple a means of bring the craft back to home via a mode switch. 
Maybe your confusing my mention of `safehome_usage_mode = RTH_FS`
I did state that Failsafe mode can be used to return the craft to the takeoff location. And also simulates a RX signal loss. 
If it only simulated a RX signal loss, and didn't not return the craft to takeoff. Then I would not use or recommend it.

I use Failsafe mode switch in place of RTH mode, because it does the same and more. 
 I doesn't take up an extra channel position in my setup's. It's my Return to Home switch because it performs the same task.
I have used both over the years. But found Failsafe mode to have no disadvantage at all.  I would still not be using it, if it had caused an issue for me or anyone else I fly with.

Back when I used PPM. I noticed that a signal below `rx_min_usec` would initiate failsafe.  This could happen as a result of an RX FS condition, when outputting `no pulses` or the RX signal wire came loose.

tonyyng, 2021-03-01:
@Jetrell Thanks for the writeup.  I opted to incorporate some of the suggestions directly in the safehome doc.

I'm creating the pull request now.

closed 2021-03-03 by Jetrell