iNavFlight/inav #6618: PosHold not working

Jakub-Espandr, 2021-02-18:
Hi, today I have set up my second fpv quad. On quad, there is Mamba f405 mk2 + 40A stack, BN-880 GPS and magnetometer.

I wired up all cables and checked all components and then calibrate accelometer and compass. I also assigned PosHold to aux channel. 

Then I went on field with my laptop and there I chacked gps sats and all measured values. 

I power on quad, wait for gps fix, then arm and get it into the air. When I change position of switch that is assigned to aux that controls PosHold, nothing happen. 

I tried PosHold in Acro, Horizon and Angle but the quad is behaving the same with poshold on and with poshold off.

I tried calibrating everything and rechecking, but nothing helps to solve my issue.

When i put quad on my table at home, arm (without props) and switch aux to poshold, the coursor shows that poshold should be activated but it is still greyed out.

Thanks a lot for your advices and helps.
Regards Jakub Espandr from Czech Republic.

avsaase, 2021-02-18:
Can you share a log and diff file? And what does the OSD say when switching into POS HOLD?

Jakub-Espandr, 2021-02-18:
> Can you share a log and diff file? And what does the OSD say when switching into POS HOLD?

Hi. In a while I will send diff. Tomorrow I´ll send log file (now it´s dark outside for test flight)
OSD says nothing about changing mode to PosHold.

Jakub-Espandr, 2021-02-18:
> Can you share a log and diff file? And what does the OSD say when switching into POS HOLD?

I tried to test PosHold now and recorded log - https://drive.google.com/file/d/1Y3nfC9TuzHQ-4TjcAf4vOIu4uRbgmNDb/view?usp=sharing              (after flight I attached drone to computer and hit Save flash to file)
here is dump: https://docs.google.com/document/d/1ayyA939LzeqyAEtyBtEiPFJdlsLhnkwxM42IpnIj5VY/edit?usp=sharing

some basic notes about channels: ch5 - arm; ch6 - nav rth; ch7 - (acro)/angle/horizon; ch8 - nav PosHold

10s before flight I had this number of satellites and HDOP, EPH, EPV values: https://drive.google.com/file/d/1Etv5VUXY_JS9aT3yWoPRq6KPEgEidvSg/view?usp=sharing

**here is diff:**
# diff

# version
# INAV/FURYF4OSD 2.6.0 Dec  4 2020 / 11:22:52 (cada68e4)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000  1.000
mmix 1  1.000 -1.000 -1.000 -1.000
mmix 2  1.000  1.000  1.000 -1.000
mmix 3  1.000  1.000 -1.000  1.000

# servo mix

# servo

# safehome

# logic

# gvar

# feature
feature SOFTSERIAL
feature GPS
feature RSSI_ADC
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 2 2 115200 57600 0 115200
serial 30 2048 115200 115200 0 115200

# aux
aux 0 0 0 1300 2100
aux 1 1 2 1700 2100
aux 2 2 2 1300 1700
aux 3 11 3 1700 2100
aux 4 10 1 1700 2100
aux 5 46 3 1325 1700

# adjrange
adjrange 0 0 0 900 2100 12 0

# osd_layout
osd_layout 0 0 24 2 V
osd_layout 0 1 2 1 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 23 12 V
osd_layout 0 9 1 2 H
osd_layout 0 11 2 3 H
osd_layout 0 12 1 5 V
osd_layout 0 13 24 10 V
osd_layout 0 14 23 1 V
osd_layout 0 15 12 2 V
osd_layout 0 20 2 11 V
osd_layout 0 21 2 12 V
osd_layout 0 23 11 1 V
osd_layout 0 28 22 11 V
osd_layout 0 32 2 2 V
osd_layout 0 37 1 6 V
osd_layout 0 106 2 7 H

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 110
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_hz = 250
set gyro_stage2_lowpass_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 200
set rssi_adc_channel = 2
set current_adc_channel = 3
set acc_hardware = MPU6000
set acczero_x = 44
set acczero_y = 1
set acczero_z = -63
set accgain_x = 4107
set accgain_y = 4063
set accgain_z = 4018
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 270
set magzero_y = 106
set magzero_z = -222
set maggain_x = 427
set maggain_y = 460
set maggain_z = 1013
set baro_hardware = NONE
set pitot_hardware = NONE
set min_check = 1050
set serialrx_provider = FPORT
set blackbox_rate_denom = 8
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT600
set throttle_idle =  12.000
set failsafe_delay = 7
set failsafe_procedure = RTH
set failsafe_min_distance = 2000
set failsafe_min_distance_procedure = SET-THR
set vbat_scale = 1104
set current_meter_scale = 252
set model_preview_type = 3
set applied_defaults = 2
set rpm_gyro_min_hz = 150
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set mc_airmode_type = THROTTLE_THRESHOLD
set mc_airmode_threshold = 1000
set nav_use_midthr_for_althold = ON
set nav_rth_abort_threshold = 5000
set nav_rth_altitude = 8500
set nav_fw_launch_max_angle = 180
set telemetry_halfduplex = OFF
set osd_rssi_alarm = 48
set osd_left_sidebar_scroll = ALTITUDE
set osd_main_voltage_decimals = 2
set osd_coordinate_digits = 8
set debug_mode = ERPM
set name = FCC
set vtx_channel = 7
set vtx_low_power_disarm = UNTIL_FIRST_ARM

# profile
profile 3

set mc_p_pitch = 44
set mc_i_pitch = 60
set mc_d_pitch = 25
set mc_i_roll = 50
set mc_p_yaw = 45
set mc_i_yaw = 70
set mc_p_level = 50
set dterm_lpf_hz = 100
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 150
set dterm_lpf2_type = PT1
set yaw_lpf_hz = 30
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

set battery_capacity = 1550
set battery_capacity_warning = 1085
set battery_capacity_critical = 1317

# end the command batch
batch end

#

stronnag, 2021-02-18:
Looks mis-configured to me:
```
$ parse-aux.rb 
aux 0 0 0 1300 2100
aux 1 1 2 1700 2100
aux 2 2 2 1300 1700
aux 3 11 3 1700 2100
aux 4 10 1 1700 2100
aux 5 46 3 1325 1700
ARM                  CHAN 5 1300 2100	(aux 0 0 0 1300 2100)
MANUAL               CHAN 6 1700 2100	(aux 4 10 1 1700 2100)
HORIZON              CHAN 7 1300 1700	(aux 2 2 2 1300 1700)
ANGLE                CHAN 7 1700 2100	(aux 1 1 2 1700 2100)
Unknown              CHAN 8 1325 1700	(aux 5 46 3 1325 1700)
BEEPER               CHAN 8 1700 2100	(aux 3 11 3 1700 2100)

```

stronnag, 2021-02-18:
This doesn;t help either:
```
set baro_hardware = NONE
```

stronnag, 2021-02-18:
Moving to discussion, this is not firmware issue

closed 2021-02-18 by stronnag