iNavFlight/inav #6629: Change all distance units to consistent units and scale (eg SI units - meters)

fernandodlc, 2021-02-22:
Currently many of the distance units are in the very unintuitive centimetres. I find this confusing and it has caused me a few problems when certain aspects are out by a factor of 10 either way. CM are great for measuring triangles in school, less good for defining distances in the 10s to 100s of meter ranges.

It would be far easier to work in the standard SI distance unit of meters. This is the most obvious to a new user, it has a close imperial equivalent (the yard) for our freedom loving friends.

Stretch request: Allow users to select the desired units and scale, eg mm, cm, m, inches, yards, miles, furlongs, leagues, barelycorns, chains or whatever.

bishko, 2021-03-06:
Fully agree with you on this one!!!

SI units (e.g.: [m], [kg], [l]) are defined by The International System of Units. I don't see why [cm] have been used to begin with for this project but maybe the developer has a reason and would like to comment / elaborate for our better understanding. 

Thus, in another [thread](https://github.com/iNavFlight/inav/issues/5246) that has been closed on this same topic, @DzikuVx commented that it has to be changed from root, but I don't think that is true. 

All that needs to happen is to add a small division / multiplication to the input boxes and change the text displayed in the units, so for example:

Variable: **Loiter_Radius (r_L) = 100.000 cm** ; where 100.000 is a raw input value defined by the user

change to:

Variable: **Loiter Radius (r_L) = 1000 m** ; where (r_L = r_L * 100), so this little division takes care of the units and nothing needs to change in the main code. ;) ;)

I think that's easy-peasy and there is only a had full of input boxes - a half-day's work at the most! ;)

PS: I'm not sure what they called the variable so I just used r_L for Loiter Radius.

avsaase, 2021-03-06:
AFAIK the reason for the sometimes strange units is to use integer math for performance reasons as much as possible. I assume that argument still stands so then changing the units internally is a no-go. Changing the units in the configurator is possible, but that will result in seemingly inconsistent values between the configurator and the cli settings. Perhaps it is possible to also change the units in the cli and convert them internally, but that will be a ton of work, and break compatibility with older diffs. 

@DzikuVx?

bishko, 2021-03-06:
I get your point @avsaase, but rounding to the second digit is also an option. While you are right in saying that working with integers will be easier on the processor, having [cm] instead of [m] is a nightmare for this type of application and makes no sense to begin with. Having an engineering background, I stand by SI units no matter the cost and complexity of the rework because this is the right way to go about this. Most models fly at least 5km these days and expressing a maximum distance for example as 5000m is far more user-friendly and logical to most people, as opposed to 500.000cm. 

It's like having a bank account that is expressed in pennies or cents, as opposed to pounds and euros, hahaha :D

wx4cb, 2021-03-22:
i totally agree... but why setting on metric? why not use imperial... feet, inches, miles per hour, i mean... most of aviation in general use feet and miles per hour in some form (mph vs kn) :D

that said, i do agree that it's pretty stupid to have 1 option to be centimeters and the next option to be meters.

bishko, 2021-03-23:
That's true, it shouldn't be metric only. There should be options for the user to chose from so that they can use whatever they are most comfortable with. Consistency is a must though regardless of what the processor and the formulas calculate internally.

fernandodlc, 2021-03-23:
> That's true, it shouldn't be metric only. There should be options for the user to chose from so that they can use whatever they are most comfortable with. Consistency is a must though regardless of what the processor and the formulas calculate internally.

I'd agree with this in principle but I'm not sure how much real value it would have vs just using SI units. Changing cm -> m would drive considerable value for relatively low complication. Adding in user selectable units would be the gold standard but the effort level would be disproportionate for the value.

digitalentity, 2021-03-23:
Changing to unified SI units will require very large effort but will bring very little real value. It's a nice to have at the UI/UX level, but bears almost zero value on the firmware level.

This feature request should be moved to Configurator which delivers UI/UX, @DzikuVx WDYT?

fernandodlc, 2021-03-23:
I would emphasise the UI/UX value here. I have lost a model to stupid mistakes here and I'm sure I'm not alone. The consequence of error here can easily become a lost model or worse a dangerous one. A model RTHing at 4000m or 400m could easily be a safety issue.