iNavFlight/inav #6630: Add vbat compensation to navigation throttle values

fernandodlc, 2021-02-22:
Navigation throttle values are fixed at a certain throttle %. As the power source voltage changes the effective thrust/power/speed of these values will change dependent on the power voltage.

Add vbat compensation to navigation so that this variation can be at least partially offset

Voltage tends to affect power by the SQUARE of the difference.
For example:
#1 5v + 1000kv
#2 10v + 1000kv.

The power alone is doubled from the voltage but for the same throttle value the motor is also trying to spin twice as fast. The expected max wattage from #2 is expected to be V-diff*V-diff or in this case #2 = 4x#1

I would suggest any compensation is just a mathematical formula based on the configured value being set for max cell voltage.
(I'm sure someone smarter than me figured this out better somewhere else but issues should be stand-alone)

This should be useful for anyone using fixed wing navigation modes, especially those with low power models on li-ion cells (eg ZOHD drift and 250g).

avsaase, 2021-02-22:
There already is throttle voltage compensation. This increases the throttle output when the battery gets empty to maintain the same thrust. You can enable it on the Configuration tab. Or do you want to keep output power constant between batteries with different voltages/cells?

b14ckyy, 2021-02-22:
> There already is throttle voltage compensation. This increases the throttle output when the battery gets empty to maintain the same thrust. You can enable it on the Configuration tab. Or do you want to keep output power constant between batteries with different voltages/cells?

As far as I am concerned that is NOT used for Fixed wing but only on quads. I had it enabled on all my planes when I thought it would work for it but on every plane I flew the current went down near the end of the battery as well as the voltage. So the overall power went down too. 

the only reason why the planes still flew where was, because the lower speed caused an altitude loss and the altitude controller raised the throttle to match the new pitch angle. But in climbs the power really lacks on li-ion when the battery voltage goes down. and the flight controller always keeps the same throttle output value no matter what voltage is left.

fernandodlc, 2021-02-22:
This (for me at least) is specifically for my zohd drift. Cruise throttle ranges from 32%-45%.

My understanding of VBAT compensation was that it was more tired PID functions than throttle gain.

b14ckyy, 2021-02-22:
@fernandodlc there is PID vbat compensation and Throttle vbat compensation. But yeah still I think this is just used on a multirotor setup. Not fixed wing.

b14ckyy, 2021-02-22:
If it really is also active on fixed wing it might e too weak. Then we might play around with 
thr_comp_weight

see https://github.com/iNavFlight/inav/blob/master/docs/Battery.md#automatic-throttle-compensation-based-on-battery-voltage

avsaase, 2021-02-22:
As far as I can see there is nothing in the code that limits this feature to multirotors. I am also pretty sure I have seen this working when I was experimenting in #6478.

b14ckyy, 2021-02-22:
> As far as I can see there is nothing in the code that limits this feature to multirotors. I am also pretty sure I have seen this working when I was experimenting in #6478.

Thanks for that info. I will try some bench testing I think. By enabling Compass I should be able to enable navigation modes on the bench.

avsaase, 2021-02-22:
The voltage compensation applies to any flight mode so no need to enable a nav mode for bench testing.