iNavFlight/inav #6639: DEBUG OSD Element is wider than OSD available space [2.7.0]

b14ckyy, 2021-02-24:
## Current Behavior
Adding the OSD Item 78 for DEBUG info, starting at column 0 of the OSD area, will still crop the right side of the element.

## Steps to Reproduce
1. Enable OSD ID 78 with column coordinate 0
2. Look at the OSD on goggles or monitor
3. Right side of the OSD element crops 1 or 2 digits

## Expected behavior
DEBUG OSD should be visible completely to be usable

## Suggested solution(s)
lower the number of visible digits of all debug values by 1 to save width

## Additional context
none

---
- FC Board name and vendor: MATEKF405SE / F405 Wing and also F411-WSE
- INAV version string: 2.7.0 Feb 18 2021 / 09:51:07 (fc7aeaf2)

breadoven, 2021-02-24:
I just removed the square brackets except 1. Just as readable without them:

`tfp_sprintf(buff, "%u=%11ld [%u=%11ld", bufferIndex, debug[bufferIndex], bufferIndex+1, debug[bufferIndex+1]);`

b14ckyy, 2021-02-24:
awesome. Then I can continue to test the new board align feature with Pawel :)

PR #6660 (2021-03-02): Limit the size of OSD DEBUG element
fixes #6639
closed 2021-03-02 by DzikuVx