iNavFlight/inav #6640: PIFF tuning in OSD - no correct value adjustment possible [2.7.0]

b14ckyy, 2021-02-24:
## Current Behavior
PIFF values on wings (maybe also copter PIDs but not tested) cannot be adjusted correctly. Roll right does not change value at all. Roll left will jump huge values and makes tuning the values impossible. 

## Steps to Reproduce
1. Enter OSD menu
2. Navigate to PID tuning
3. Select any item with pitch input
4. Try to change any value with roll input

Behavior as described above can be seen

## Expected behavior
usual step by step adjustment as before in 2.6

## Suggested solution(s)
fix adjustment bug

## Additional context
This was discovered in a 2.7 test build

---
- FC Board name and vendor: Matek F405 Wing and also F411-WSE
- INAV version string: 2.7.0 Feb 18 2021 / 09:51:07 (fc7aeaf)

Jetrell, 2021-02-27:
Are you by chance using an OpenTx global function to control the iNav adjustments tab for inflight tuning?

I have experienced the same thing you described when using this combination. 
It was caused by the TRIM channel being held at 2000uS. Which prevent the adjustments for working. And as a side effect, also cause the OSD CMS to make large jumps in values one way and do nothing the other way.

b14ckyy, 2021-02-27:
No I used the OSD menu that is controlled by the sticks. INAV adjustments. These work fine. Trims are centered

Jetrell, 2021-02-27:
> 
> 
> No I used the OSD menu that is controlled by the sticks. INAV adjustments. These work fine. Trims are centered

Mine looked fine in the Radio. It all seemed to work. I only picked up on it in the Configurator receiver tab.  
In my case CH12 was being held at 2000uS. (CH12 was used to change values in the OpenTX GV/ iNav adjustments tab.

It turned out that the CRSF was outputting RSSI on CH12, which overrode the values in the OSD stick menu.

b14ckyy, 2021-02-28:
I think we are not talking about the same thing here. 
I am not talking about in flight adjustments where you assign a channel to select them and another channel to change the values. I am talking about the OSD Menu! This one: https://youtu.be/Eg32BRqLsj0

this has nothing to do with any channel beside the main control channels and is controlled by roll+pitch stick.

b14ckyy, 2021-02-28:
Here you go. When I go to the PIDs and push the stick to the right, nothing happens, and when I push it to the left, the values change as you can see. Other sub menus work fine. 
https://www.youtube.com/watch?v=bA1BFaEJGIU

avsaase, 2021-02-28:
When was this bug introduced? I remember it working correctly in the past, but it might be I was on 2.4 the last time I tried changing the PIFFs via the OSD menu.

Jetrell, 2021-02-28:
@b14ckyy 
That's what I experienced with 2.7 a few ago. Can't remember the exact time. I thought I worked around it.
Any how.. Just tested it on another model. Its still there in the later 2.7 development FW.

b14ckyy, 2021-02-28:
> When was this bug introduced? I remember it working correctly in the past, but it might be I was on 2.4 the last time I tried changing the PIFFs via the OSD menu.

Up until 2.6 it was fine. I first recognized it in Pawels test build for the auto board align and the same happens in your build for auto rates. So at least it is nothing that you both changed. Must have happened earlier in the 2.7 development.

OptimusTi, 2021-03-02:
I was able to narrow it down around these commits but needs checking: 97eea57e9e01a754df06d0d16c2e3c26fbb4c2a3 or 4f3abed3ef07809f62f4e778d52f2d781d03040d

DzikuVx, 2021-03-02:
If anybody can test if #6658 fixes that would be awesome. No way to test by myself right now

b14ckyy, 2021-03-02:
> If anybody can test if #6658 fixes that would be awesome. No way to test by myself right now

got another Build from @avsaase today and will flash later on my 2 test rigs. Both fixed from you and also boith new tuning features included. I can tell you after work.

OptimusTi, 2021-03-02:
> If anybody can test if #6658 fixes that would be awesome. No way to test by myself right now

Worked correctly on the bench for me

PR #6658 (2021-03-02): Change PIDFF settings in CMS to UINT16
fixes #6640
closed 2021-03-02 by DzikuVx