iNavFlight/inav #6645: beeper RX_SET & RX_LOST do not work with ESC buzzer (DSHOT600)

chillside, 2021-02-24:
## Current Behavior

I have a built-in buzzer and it works well. But I would like to turn on the ESC buzzer too. ESC only beep when turned on, but do not beep when beeper is turned on or power is lost

Dshot600 ESC protocol is on. I try CLI commands: `beeper ALL` & `beeper RX_SET` `beeper RX_LOST` and save. But to no avail

---

version
INAV/KAKUTEF7HDV 2.6.0 Dec  4 2020 / 11:41:25 (cada68e4)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

- FC Board name and vendor: Kakute F7 hdv
- INAV version string: 2.6.1

Jetrell, 2021-02-25:
@chillside  iNav doesn't have DShot Beacon at the flick of a switch, like BF does.

chillside, 2021-02-25:
> @chillside iNav doesn't have DShot Beacon at the flick of a switch, like BF does.

why? it is sad. will there ever be?

Jetrell, 2021-02-25:
iNav doesn't have` DShot Command`  required to externally control BLheli.
There has been interest. Not so much in this feature, but in `Flip Crash` 'turtle mode' that also requires `DShot Command` to operate.

poi2poi, 2021-02-27:
Well.. why can't we take a few code-snippets from the Betaflight sourcecode and make our ESC go beep-beep.. ? Would be great..

Jetrell, 2021-02-27:
> 
> 
> Well.. why can't we take a few code-snippets from the Betaflight sourcecode and make our ESC go beep-beep.. ? Would be great..

@poi2poi Sounds ideal to me.

@DzikuVx I remember you talking about this on your live stream. Is there any reason @poi2poi could not port DShot command?

Jetrell, 2021-03-10:
@chillside @poi2poi  If this gets implemented #6685. It will go most of the way towards having the motor/ESC as a beeper.

harry1453, 2021-04-10:
I'm working on this now.

PR #6812 (2021-04-16): DShot Beeper
Closes #6645. Motors beep in `RX_LOSS` and `RX_SET` beeper modes.
closed 2021-04-16 by DzikuVx