iNavFlight/inav #6647: Servo autotrim settings delay / fade

poi2poi, 2021-02-25:
## Current Behavior
when activating auto-trim, it will take a few seconds and iNav stores the new servo center settings. when the mid-points were off by a lot, or you used trimmer on your TX, the plane will all of sudden change direction because the trim settings / stick inputs are then added to the new center points. this obviously only happens when in Manual mode.

## Desired Behavior
Give pilots not flying in stabilized mode more time to react / adapt to the new servo settings

## Suggested Solution
- not apply new server center positions until disarmed (= calibration complete)
- or -
- add a "fade-to-new-values" algorithm so the pilot has enough time to react

b14ckyy, 2021-02-25:
I understand your point but there is an easier way. 
Your Scenario ONLY applies for Autotrim in manuak mode by using radio trims (what is not recommended anyway). 
But you can also just trim on the radio, land without disarm, enable autotrim and wait 5s. Then disarm and reset trims. No reason to use autotrim mid air with the radio trim method. 

and if the trims are so much off that the plane suddenly would do rapid movements, then you should not autotrim anyway but mechanically trim first.

poi2poi, 2021-02-25:
Okay well that's pretty clever, yeah I had no reference on how to sat the elerons on my S800 so it was pure guessing on the first flight, but I guess it will work as it is for most people.

It's just unusual for people building a plane with Gyro-FC for the first time to use the stabilization systems, I guess it would be handy to have an official quick start manual on how to set up a plane with a iNav compatible FC.

I'm aware there are many youtube videos but I'm more into text-based, step-by-step manuals where you can easily skip what you already know.

b14ckyy, 2021-02-26:
@poi2poi like here? :) 
https://inavfixedwinggroup.com/inav-guides/guides-for-getting-started/

poi2poi, 2021-02-26:
Thanks blacky

closed 2021-02-26 by poi2poi