iNavFlight/inav #6651: FR: Calculated glideslope

wx4cb, 2021-02-28:
I was thinking that we have access to the vertical speed and obviously the ground speed, i would be a great addition imho to be able to calculate the glide slope as an osd item.

glideslope in general is the averaged result of sink rate (- Vspd) over ground speed. 

we already have all the necessary variables and also would be a good thing for tuning things like the linear descent etc (again another item we already have).

Just a thought.

avsaase, 2021-04-27:
Is the glide slope not the other way around, the horizontal speed over the sink rate?

wx4cb, 2021-04-27:
@avsaase it may well be, have to look it up, but when im out in full size, ground speed in Knots *5 is the way i usually do it, that will get you the vspd descent rate you need

breadoven, 2021-11-30:
Closed by https://github.com/iNavFlight/inav/pull/6939.

closed 2021-11-30 by breadoven