iNavFlight/inav #6652: Launch mode doesn't arm

bover992, 2021-03-01:
I have a MATEK F765 and a TARANIS Q X7
If i try to arm my motor it work well, but if I try launch mode a blue led start to flashing and I cannot arm anymore.  
Below you can find my DIFF ALL and also a video that show the issue, I hope someone can help me.

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 -50 0 -1
smix 1 3 1 -50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo

# safehome

# logic

# gvar

# feature
feature MOTOR_STOP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1450 2100
aux 1 1 2 900 1300
aux 2 2 2 1300 1700
aux 3 12 2 1700 2100
aux 4 36 1 1650 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set acc_hardware = MPU6000
set acczero_x = 51
set acczero_y = -19
set acczero_z = -50
set accgain_x = 4070
set accgain_y = 4098
set accgain_z = 4009
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = ADC
set motor_pwm_protocol = STANDARD
set failsafe_mission = OFF
set align_board_yaw = 1800
set vbat_scale = 2480
set platform_type = AIRPLANE
set has_flaps = ON
set model_preview_type = 8
set small_angle = 180
set applied_defaults = 3
set nav_wp_radius = 3000
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 5000
set nav_fw_control_smoothness = 2
set nav_fw_launch_accel = 1001
set nav_fw_launch_thr = 2000
set nav_fw_launch_motor_delay = 300

https://user-images.githubusercontent.com/70479670/109429378-bbd54300-79fb-11eb-9a32-875555a07ae6.mp4

# profile
profile 1

set rc_expo = 30
set rc_yaw_expo = 30
set pitch_rate = 15
set yaw_rate = 9

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

set battery_capacity = 1300

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

https://user-images.githubusercontent.com/70479670/109471391-fb8a4200-7a70-11eb-9a36-00cb7fdbc00d.mp4

breadoven, 2021-03-02:
Do you have any OSD or log info ? Try connecting Configurator and checking status. Might also help if you had a buzzer attached, it does provide useful audio feedback.

bover992, 2021-03-02:
no, I don't have any OSD or log info...I've checked the status before writing this post, but all the light were green, no problem detected

bover992, 2021-03-18:
Someone could help me?

bover992, 2021-03-18:
below the #status detected when the problem occours:
# status
System Uptime: 278 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 12.44V (3S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, PITOT=ADC
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x20020000, Heap available: 1820
I2C Errors: 50, config size: 6538, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = ADC 4
System load: 2, cycle time: 1001, PID rate: 999, RX rate: 111, System rate: 9
Arming disabled flags: CLI ROLL/PITCH
VTX: not detected

bover992, 2021-03-18:
below the #status when the problem occours:

# status
System Uptime: 47 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 27.98V (7S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, PITOT=ADC
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x20020000, Heap available: 1820
I2C Errors: 50, config size: 6538, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = ADC 4
System load: 2, cycle time: 1002, PID rate: 998, RX rate: 111, System rate: 9
Arming disabled flags: ARMSW CLI ROLL/PITCH
VTX: not detected

avsaase, 2021-03-18:
> Arming disabled flags: ARMSW CLI ROLL/PITCH
This suggests the roll and pitch channels are not centered, but it would really help if you hooked up a vtx so you can see the osd.

breadoven, 2021-03-18:
As said before ... no beeper, no OSD and no Log makes it very difficult to work out what's wrong. The lights on the FC tell you very little. Your board has an SD card for a Log so put a card in and see what the Log says or post it here so others can help.

bover992, 2021-03-18:
> > Arming disabled flags: ARMSW CLI ROLL/PITCH
> 
> To me, this suggests the roll and pitch channels are not centered. Do you have the launch mode switch interacting with one of these channels? In either case, it would really help if you hooked up a vtx so you can see the osd.

solved!!!!! The problem was simply that the roll and pitch channles were not centered!!! Thank you!

avsaase, 2021-03-18:
> As said before ... no beeper, no OSD and no Log makes it very difficult to work out what's wrong. The lights on the FC tell you very little. Your board has an SD card for a Log so put a card in and see what the Log says or post it here so others can help.

Doesn't the log only start recording once you arm?

breadoven, 2021-03-18:
> Doesn't the log only start recording once you arm?

Hmmm, you're right. Does it not log failed arming attempts, never checked, I'm guessing not. So OSD or Configurator status if you can't arm I guess.

blastedgolem, 2021-03-20:
Прошу помощи по армингу в INAV. В разделе Mode первая строчка ARM всегда серая. Какой бы канал я не ставил для арминга - строчка не работает, остаётся серого цвета. ПК Matek405SE, пульт FlySky. Need help.