iNavFlight/inav #6657: iFlight SucceX-E F7

IngressoDev, 2021-03-02:
Is it possible to use my fc (iFlight SucceX-E F7) with iNav?
There is no selection in the "Flash Firmware" tab for my fc like it is in betaflight.

Is it possible to choose a similar firmware?
I don't know how to create a new firmware spicific for my fc, but i have no fear to be a guinea pig :)

Regards

G9xy, 2021-08-04:
If there is no published firmware target for your FC you can flash it from local file.

See my post: [https://github.com/iNavFlight/inav/issues/6711#issuecomment-892387900](url)

RipperGOT, 2022-01-15:
> If there is no published firmware target for your FC you can flash it from local file.
> 
> See my post: [https://github.com/iNavFlight/inav/issues/6711#issuecomment-892387900](url)

Can you post it again?

I bought a Nazgul!!!!😬