iNavFlight/inav #6663: matek F722 mini se,crazy baro in al inav version,too in 2.6.1 for micro quad nav modes

Molinero1, 2021-03-03:
hello team, i mount a matek F722 mini se from bangood in a micro quad with gps beitian bn 880.
confIgurated with all inav version and last 2.6.1 with same PROBLEM BARO NOT WORKS OK.
# version
# INAV/MATEKF722MINI 2.6.1 Feb 25 2021 / 10:53:27 (64249b1a)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
logic 5 0 -1 0 0 0 0 0 0
logic 6 0 -1 0 0 0 0 0 0
logic 7 0 -1 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
logic 12 0 -1 0 0 0 0 0 0
logic 13 0 -1 0 0 0 0 0 0
logic 14 0 -1 0 0 0 0 0 0
logic 15 0 -1 0 0 0 0 0 0
logic 16 0 -1 0 0 0 0 0 0
logic 17 0 -1 0 0 0 0 0 0
[... 1263 lines cut ...]
set pitch_rate = 50
set yaw_rate = 50
set manual_rc_expo = 20
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch

BARO NOT WORKS ,please help.thanks from Spain.

Molinero1, 2021-03-03:
link of video:
https://youtu.be/xFfDHq4iXEA

mrcottonmouth, 2021-03-05:
I'm having the same issue I think. My video is here https://youtu.be/5w2i8sI1WlcI have tried several things to fix this, I have replaced the foam over the barometer 3 times. It's thick foam now. I looked at my log file and it seems the baro is looking mostly right but the reported altitude is crazy. I'll post more videos of this issue, it happens on each flight.
[INAV_cli_Bobcat_20210304_171840.txt](https://github.com/iNavFlight/inav/files/6088061/INAV_cli_Bobcat_20210304_171840.txt)

mrcottonmouth, 2021-03-05:
Another video displaying the issue https://youtu.be/vBVHZpu_se8
I just set the quad up with betaflight using the 4.2.8 custom firmware from Matek and the altimeter seems to be working right. I'll post that video soon.

mrcottonmouth, 2021-03-05:
Video of the exact same hardware with Betaflight https://youtu.be/5YrOj5qsGwo
GPS, compass, baro, altimeter, etc seem to be right. I think there is a iNav software bug that is causing our altimeter issues.

OptimusTi, 2021-03-06:
Are you guys using the target from Matek's website or from here? Just curious.

mrcottonmouth, 2021-03-06:
I was using the one from here

Molinero1, 2021-03-06:
Sorry Bobcat, the same problem as me and at the moment, the same solution to some fc MATEK that says inav that supports the quad in x: NONE.
IT IS A MATEK PROBLEM WITH INAV or the other way around, because this failure with the baro also happens to the matek f411 mini se, and just like the MATEK F722 MINI SE works ok the baro in betaflight with rescue operating but there is your ship if you actuate rth in inav ...
waiting for these matek not to become space junk.
post: the kakute f7 v3 mini if ​​they have solved THE SAME PROBLEMS OF BARO, in version 2.6.1 and it works rth perfect.
waiting for some wise man to bring the matek back to life ...

Molinero1, 2021-03-06:
report to matek

Your message has been sent to service@mateksys.com

rts18, 2021-03-07:
@Molinero1 The kakute f7 v3 has a BMP280, while the F722-mini SE has a DPS310 barometer.
The DSP310 should be a better unit, with less drift due to temperature.

I'm using an F722-mini SE in one of my builds. It drifts more than my BMP280 boards. But i'm not experiencing the large drift that either of you are.

May be certain small air frames create pressure vortices around the FC/baro, affecting altitude data.  
iNav sensor fusion filtering for the DSP310 may not be good enough. As @mrcottonmouth video's show its more accurate with Betaflight.

Molinero1, 2021-03-07:
IF WITH THE SAME QUAD X ,THE CONFIGURATION IS CHANGED to:
-inav 2.6.1 = BARO READING FAIL = FLIGHT STABILIZED NOT WORKING

BETAFLIGHT BARO WORKS OK = RESCUE MODE OK.
the answer is clear ...
Some inav teacher should appear in this forum and take note ...

Molinero1, 2021-03-08:
tic tac,tic tac...

Molinero1, 2021-03-08:
I have reinforced the insulation of the plate and it remains the same BAD READING OF THE BARO ... IS THERE NO SOLUTION, THE SAME PROBLEM OCCURS IN TWO UNITS.THANK YOU.
[url=https://subefotos.com/ver/?9e7e588c554d9a7f04ca3ad14d11bb7bo.jpg][img]http://thumbs.subefotos.com/9e7e588c554d9a7f04ca3ad14d11bb7bo.jpg[/img][/url]

Beardyblair, 2021-03-10:
I can confirm that I am also getting wild ALT figures during flight with the Matek F722 mini se. Due to this I did not try height hold flight.

Molinero1, 2021-03-10:
So how does this work now? Are they trying to modify inav for matek or do we have to return the plates to the manufacturer for misleading advertising?

rts18, 2021-03-10:
> 
> 
> So how does this work now? Are they trying to modify inav for matek or do we have to return the plates to the manufacturer for misleading advertising?

iNav only has a few developers. They add features and fixes in their free time and aren't payed for it, as Matek are.
Be patient and respectful. A fix may take a long while to happen. 

If you can't wait, contact Matek.

OptimusTi, 2021-03-10:
Are you guys having the same issue with the target provided in Matek's website? Has anyone tested it?

rts18, 2021-03-11:
@OptimusTi  My board was purchased the week they were released. I ran Matek's 2.5.1 build, until the DPS310 driver was merged into development. I notice no difference between it and 2.6 or 2.6.1; even with the baro median filter turned off.
The most I experience is a few meters drift from time to time. Never as bad as these guys are seeing. Although there certainly seems to be an issue here..

I compared my Dump files to both posted here. There are some small differences. Depending on how much accelerometer filtering and GPS altitude data play a part in the final altitude reading.

set gyro_stage2_lowpass_hz = 260
set gps_provider = UBLOX7    **(10hz update rate)**
set i2c_speed = 800KHZ         **(not sure if Matek uses the SPI or I2C version)**

Most of my PID filter settings are extremely close to @mrcottonmouth

OptimusTi, 2021-03-11:
I don't have one of those boards, but I'd check "baro_bustype" in BF. If it's i2c, I'd try lowering i2c_speed in Inav to 200 and test?

Molinero1, 2021-03-11:
I am being respectful enough to advertise MATEK that their mini boards work with INAV AND THAT'S A LIE.
I DO NOT TOLERATE THE LIE AND MORE IF THERE IS MONEY IN THE MIDDLE.

Molinero1, 2021-03-11:
AND THIS THE SAME PROBLEM WIYH MATEK AND F411 MINI SE...TWO TIMES IS MORE...

Molinero1, 2021-03-11:
I appreciate the effort and respect of the inav programmers for providing us with a system TO FLY OUR MODELS. Thanks from the heart

Molinero1, 2021-03-11:
a LIAR is caught sooner than a LAme man.
Tips

INAV2.5.1 and Betaflight 2.4.x downloaded from configurator don’t support new barometer DPS310.
DPS310 will be supported officially by INAV2.6 / BetaFlight 4.3 and new configurator.
For now, pls download customized INAV2.5.1 and BetaFlight 2.4.0 from here.
inav_2.5.1_MATEKF722MINI_DPS310.he

In INAV CLI mode, type command

set baro_hardware = DPS310
save
betaflight_4.2.0_MATEKF722SE_DPS310.he

In BF CLI mode, type command

set baro_bustype = I2C
set baro_i2c_device = 1
set baro_i2c_address = 118
set baro_hardware = DPS310
save
Tips

INAV2.5.1 and Betaflight 2.4.x downloaded from configurator don’t support new barometer DPS310.
DPS310 will be supported officially by INAV2.6 / BetaFlight 4.3 and new configurator.
For now, pls download customized INAV2.5.1 and BetaFlight 2.4.0 from here.
inav_2.5.1_MATEKF722MINI_DPS310.he

In INAV CLI mode, type command

set baro_hardware = DPS310
save
betaflight_4.2.0_MATEKF722SE_DPS310.he

In BF CLI mode, type command

set baro_bustype = I2C
set baro_i2c_device = 1
set baro_i2c_address = 118
set baro_hardware = DPS310
save.

Beardyblair, 2021-03-11:
For clarity - Matek make hardware, they offer warranty on the hardware. They ship with software but its not guaranteed. INAV is a free open source project that can be flashed to that hardware and others. As INAVs licence states, it comes with no warranty, the same for betaflight.

The sites usually state its compatable with xxx. And it is, but thats where things get tricky. Matek dont work for INAV and INAV does not work for Matek. Two different "products". So naturally there will be bumps along the way. You cant get mad as neither guarantees the other. 

Saying things like "tic tac,tic tac..." and spamming these issue pages with things in caps does not help.

Take a breath, understand what this community and project is and then come back. Report issues and wait patiently for a voulenteer to answer and help. I guarantee this will be fixed but behaving this way is a sure fire way to be ignored or put to the back of the line.

Just my 2c.

Molinero1, 2021-03-11:
I see an ad (bangood) for matek in which it indicates that its mini boards work with INAV.
ONE ENTERS THE INAV CONFIGURATOR and is financed with matek ... it appears at your logo ...
Trust and buy 7 matek plates, five f411 mini se and two f722 mini se and WARNING THAT THEY DO NOT WORK WITH INAV because matek does as you well know "deaf ears" to my notice. I am always a beta tester of inav and faithful defender of el.Many pilots fly by you.
I test ALL versions of INAV with MATEK MINIS and try to configure some boards that do not work in INAV.
and for reporting this situation so that you and all the world pilots know it, MATEK'S ATTITUDE OF PASSING THREE PEOPLES OF MY / FUTURE PROBLEM OF ANYONE You tell me that my "attitude" does not help the community and that I will be banned .. .
FORWARD THAT THE PULSE DOES NOT SHAKE ...
goodbye and good flights.

Molinero1, 2021-03-11:
DO NOT BUY THESE PLATES, THE BARO DOES NOT WORK WITH INAV.
NEITHER MATEK, NOR INAV, NOR BANGOOD WILL BE RESPONSIBLE FOR THE PROBLEM AND YOU WILL ENCOUNTER WITH SPACE GARBAGE.

Molinero1, 2021-03-11:
UNTIL ALWAYS BECAUSE YOU TELL ME TO GO.

DzikuVx, 2021-03-11:
1. This board work with INAV just fine. One of my drones is a working proof of that
2. In this specific case it might be: hardware failure, not protected baro that catches air gusts from props, poor altitude estimation from gps, vibration that are fed into accelerometer
3. We will not tolerate abusive and spammy behavior and as such, ticket is closed and abusive behavior reported to GH

closed 2021-03-11 by digitalentity