iNavFlight/inav #6675: Please Update Flywoo Goku F745

BenWThomas, 2021-03-05:
## Current Behavior
iNav FlywooF745 and FlywooF745Nano firmware is available only on Flywoo site.
Behaved flawlessly (until voltage mishap clearly damaged my Nano).
2.5.2 lacks tweaks, defaults of 2.6.x, which appear to improve my other models

## Desired Behavior
Flywoo F745 support directly in iNav, up to date
Goku F745Nano 16x16

## Suggested Solution
Work with developer to update and incorporate F745 and F745Nano firmware.
I will happily submit feedback from new F745Nano

## Who does this impact? Who is this for?
This is for all iNav users, to bring F745 software mainstream.
This would (of course) be for me, too.  iNav is my pref, and this board is worthy.

## Additional context
It's definite that I damaged my first Flywoo board, but I will be picking up a new F745 16x16 35A stack tomorrow.
It behaved very well in 130mm Phantom w/Matek I2C GPS-mag and Bluetooth -- until I decided that the LEDs on the ESC should be incorporated, but not paying proper attention.

The hardware quality is very fine, cleanly and densely packed and laid out with 6 serial ports, with nearby signal and power pads and signals.  It's a great little board to work with, especially as a stack, easy to wire logically in small space.  Zero issues during setup, iNav 2.5.2 defaults for 3" were flyable.  Great bling - a fully programmable RGB LED at each corner of each board.

With proper support the F745Nano should do well for the iNav community, as there is a lot of interest in smaller craft.
Although I've only asked Flywoo one easy question, I was impressed that it was answered next day.

d4rken, 2021-03-08:
Dup of #6522 ?

BenWThomas, 2021-03-11:
Hi, Matthias.

Yes, definitely a dupe.  Absolutely.

I would be happy to test an iNav 2.6 hex file.

I’ve just received two new F745 nano stacks.

Thanks and regards

Ben

Marcus-Barcus, 2021-03-21:
I finally got my F745 working in BF as UART2 was stuffed for TBS nano rx and used UART7 next to it and 
receiver finally works 
Since i fitted a Matek GPS/Compass for iNAV ive flashed the Flywoo Hex - everything all worked in Configurator
I fitted a lipo and it wouldnt Arm even tho port worked fine - (i know GPS works in BF and Arms after 5 Sats)

Now i cant reconnect to iNAV to check as wont open usb even after loading ImpulseRC???
To get back in i have to reflash plus Cli to go in which is a time waste.

Why isnt there a UART3 and why do Ports start at 5????? Also my Vista only works with Lipo but works off USB in BF???

Marcus-Barcus, 2021-03-25:
I got F745 working in iNave but Flywoo said to disable Bluetooth msp and now can get beck into iNav 
after flashing - uart3 has been fixed

Marcus-Barcus, 2021-04-01:
BenWThomas - How do you program the corner LEDs as one corner is just red 
- does that indicate anything untoward??

BenWThomas, 2021-04-02:
[image]

Marcus:

The top row (0-3) are the four corners of the FCU
The bottom row (4-7) are the ESC board.
Center pairs are port side (left side) and ends are starboard (right side).
(ESC board LEDs are only active when ESC board is powered through battery connector.)

It doesn't matter where on the grid you put them; it's the "wiring order" that matters.  They could all be in a line of 8, but I checked the default Betaflight config on my new stack before I burned iNav onto it, and this is how Flywoo had it set up.  Having one "board" (sequence of 4) atop the other makes it easier to visualize against the hardware, so I followed that convention.

As configured here, the FCU left-side LEDs are both red (navigational convention for port) and the right-side LEDs are green.  The ESC board LEDs are white blinkers, with a quick double-flash once a second.  FCU LEDs are not blinkers, so 'ship' orientation is correct and solid-on.  Running lights!

BenWThomas, 2021-04-02:
I'm currently working up a GEPRC Phantom with iNav 2.5.2, the most recent firmware on Flywoo's site.

iNav 2.6 and 2.7 F745Nano firmware Herpderpula offered for testing in "New Target Request; FLYWOOF745 #6522" works fine with all the on-board FCU sensors, but doesn't have support for I2C2 (where the F745Nano has it); the FlywooF745 (not Nano) uses I2C1.

I want compass support, so I dropped back to Flywoo's F745Nano 2.5.2... which seems to support all the hardware; MSP BlueTooth works at 9.6kbps.

Before discovering I'd made a wiring error while I still had the Flywoo BTNano attached (it defaults to 115kbps), I'd decided to try a 9.6 kbps JTY-18, and... that worked.  The wiring error I made was messing with supply voltage to the FCU, so I really don't know about the BT Nano yet.

With wiring now corrected, I intend to give the BTNano another go, and will comment on that here when I have more reliable info.  

Flywoo TS had advised me to run BT Nano at 3.3v; I'd had some pretty wild symptoms that I now know were due to my wiring problem.

I went back to BetaFlight to explore that, and confirmed that BT Nano with MSP was just fine at 115kbps and the official latest BetaFlight release. BT Nano with BetaFlight worked equally reliably at 3.3v, 4.5v (USB powered) and 5v, which is what the info on the product page says.  While my wiring problem didn't seem to bother BetaFlight, I couldn't make it work in iNav 2.5.2: BTNano worked like a champ all through final configuration while USB-tethered, but I kept ending up in DFU after removing and re-applying power.

This model is still benchwork; I was ready for a test flight this evening, but had a settings error that kept it from arming.

Marcus-Barcus, 2021-04-02:
BenWThomas
Thanks for reply - ive tried setting up LEDs as above and dont get a blink anymore and that 
right rear corner led is still fixed red 
Im using Flywoos update iNav Hex as they reinstated Uart3 ha ha was missing
before installing and flashing both betaflight and iNav i tested controller and it blinked
normally
My F745 functions except i couldn't get Crossfire Nano to work on uart1 so used uart7 behind it 
They also have a correct and wrong wiring chart - The Manual for Vista Tx & Rx is wrong - yes i used that 
and moved plus to get OSD to work on DJI Goggles in iNav you have to set DJI in Peripherals which
Flywoo didnt know or tell me but found myself so have good OSD and home arrow - cool
Im using Matek GPS Compass.

BenWThomas, 2021-04-04:
@Marcus-Barkus,

I hope this isn't too much information, but you've raised a couple questions... and I may have some observations that will clarify.

-   Have you flown the F745 Nano with iNav, yet?
-   Is there some version of FLYWOOF745NANO that Flywoo has posted somewhere that's newer than inav_2.5.2_FLYWOOF745NANO?
-   How long have you had the F745 Nano with the dead right rear LED?
-   Are you using a stack version of the F745 Nano?
     (Photo on Flywoo site shows no connector on separately ordered F745 Nano FCU, so they may not be identical)

Comment on LEDs:
   If you still bench-testing with USB, and don't remove and reapply power, LED settings may not completely function until you've done the no-power, power thing.
   If you have a stack, the ESC LEDs will only operate if the ESC board is powered at the battery connector.

I've been testing two different F745 Nano stacks.  Enchanted with the elegant design, great layout, stack integration, UARTS, and all that, I went all-in on Flywoo.
First board was original build of GEPRC Phantom 120:
   F745 Nano
   Flywoo 20A (stack-connected)
   Flywooo BT Nano
   Flywoo VTX625, andI just [correctly] re-flashed inav_2.5.2_FLYWOOF745NANO with iNav Configurator 2.5.
   Matek M8Q-5883

When I first powered up this first board, it had Flywoo's BetaFlight on it.
(I did not note the firmware version, because I was about flash iNav.)
All 4 onboard LEDs, by default from the factory, worked fine as alternating Blue/Green Larsen scanner.

I re-flashed iNav 2.5.2, did the ports, calibration, config, added and verified all my peripherals functioned.
I configured the LEDs (haphazardly; I hadn't figured that out yet)
In addition to the stack connector, I'd confused a wire between the FCU and ESC boards.  The online image for the pins on the ESC show CUR, +5, LED at top right; printed instructions received later with second stack show CUR, LED IN, 5V IN.
It was obvious I'd got something wrong when I powered that first stack back up -- the ESC LEDs flashed and went out, and the FCU LEDs -- that back right LED on the FCU (let's call it M1) was off, stayed off, and on that board, has not worked since.

Thinking I'd really killed it, I ordered a second and third stack.
While I waited for those, I discovered that with iNav 2.5.2, or HerpDupula's iNav 2.6 or 2.7, I could configure everything while connected to USB, and everything worked fine, apart from no I2C2 in 2.6 and 2.7 (so no mag.)
But if I removed power, and then started it back up, it would boot, appear to initialize the board -- and enter DFU.
Reflashed, reconfigured, consistent -- pull power then fire it back up: DFU
=>   This problem turns out to be iNav with the BTNano configured for MSP.
        BetaFlight does not have the DFU after power off problem that iNav has, with the BTNano
        Substituted my standard JDY-18 (9600 bps vs 115 kbps) and MSP (at 9600) is just fine
        This is very reproducible, now with two different stacks: can't do MSP at 115 kbps.
           Whether it's the BT Nano, or the un-vetted iNav firmware unable to handle the connect speed, I cannot say
           (I haven't been able to change the port speeds on either BT device, so it's apples and oranges)
 
Back to the LEDs:  the M1 LED on the first board, which is older, has never worked since that first electrical mishap.
The M1 LED on the newer second board, and all the other LEDs, work as expected.

Now, about the third stack (recall, I got 2 more a couple months after the first):  I haven't done anything except plug it into Beta Flight... and collect a full DIFF. 

If you restore this DIFF to a fresh BetaFlight burn, you will be able to see your LEDs set up the way they originally shipped.
My objective is iNav on F745 Nano... not quite there yet.

[Flywoo F745 BetaFlight DIFF.zip](https://github.com/iNavFlight/inav/files/6254001/Flywoo.F745.BetaFlight.DIFF.zip)

edyno, 2021-04-04:
For others it may be a matter of course, but for the LEDs to work, "feature LED_STRIP" had to be added. 
My settings are: 
led 0 6,6::CB:5
led 1 7,6::CB:2
led 2 8,6::CB:2
led 3 9,6::CB:5
So far, I'm flying with inav 2.5.2, where everything works OK.

BenWThomas, 2021-04-06:
@edyno Interesting.  Maybe it's the order of running through iNav Configurator steps, but I haven't had to make that declaration in CLI.  I have noticed that if I set and save the LED GUI config before enabling LED Strip in Configuration/Other Features, the LEDs will work as expected, after the board is rebooted.  (I think iNav Config runs that when you enable the the feature.).

Settings that I have are; last pair is for the VTX625 pair, but not checked as I still need to run the LED wire to that board.:

led 0 6,6::C:6
led 1 7,6::C:2
led 2 8,6::C:2
led 3 9,6::C:6
led 4 6,7::CB:1
led 5 7,7::CB:1
led 6 8,7::CB:1
led 7 9,7::CB:1
led 8 7,8::CB:4
led 9 8,8::CB:4

BenWThomas, 2021-04-06:
@edyno.  Flying OK is the best news.  Is your board F745Nano hardware?  External peripherals connected and working?  (Somebody commented on one particular UART not working earlier - that may have been in the non-Nano thread, #6522)

edyno, 2021-04-06:
Yes, I use 745nano ::
https://flywoo.net/products/goku-f745-2-6s-16x16-stack-combo-f745-fc-20a-4in1-esc-1?variant=37828024631491

Peripherals work without problems - except LED on ESC. I have a GPS and a compass connected. I don't have VTX and camera connected yet. All the flight modes I've tried work OK. Including RTH. 
UART1 = Crossfire RX 
UART2 = GPS
I2C = compass

BenWThomas, 2021-04-18:
!!!  Flywoo.net has just posted iNav 3.0.0.hex for their entire lineup.
flywoo.net/support/downloads

BenWThomas, 2021-04-18:
Anxious to test; burned OK on Le Phantóm 125, my full-config test platform.  No time now, but will anxious to see if Flywoo's BT Nano will work with this iNav build.  DEFINITELY got I2C compass.  (GPS and DJY-18 BT not talking yet, but check that out when I can.)

edyno, 2021-04-30:
I just performed my first test flight with INAV 3.0.0
Gyro, Accelerator, Baro, Magneto and GPS work OK.
RTH with no problem. I'll try more over the weekend.

# status
System Uptime: 13 seconds
Current Time: 2041-06-28T03:04:00.000+02:00
Voltage: 0.02V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x20010000, Heap available: 1716
I2C Errors: 50, config size: 7059, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 5, cycle time: 502, PID rate: 1992, RX rate: 49, System rate: 9
Arming disabled flags: RX CLI
VTX: not detected

BenWThomas, 2021-05-23:
Here's a correct diagram from the 20/35A stack combo

BenWThomas, 2021-05-23:
[image]

closed 2021-05-23 by BenWThomas

BenWThomas, 2021-05-23:
Reasonable weather permitting, first flight tomorrow.  Just finished wiring, testing, and basic calibration.  F745 4-in-1 stack w/JDY-18 BT on top, Fli14+ Rx behind with VTX625 on top.  HolyBro micro GPS/compass squeezed into holder secured with heatshrink.  Apart from the receiver, GPS, and BT, it's all Flywoo, but built on the Explorer 4" frame kit.  Gave up second guessing motors, and ordered Flywoo's jewel-like NIN 1404 3750kV, hoping they do well with GF4024 on 3s.

[image]

BenWThomas, 2021-05-23:
iNav 3.0.0, the reason for the JDY-18.  Absolutely cannot get BT Nano to work with iNav after initial configuration; FCU goes DFU after first removal and restoration of power.   (Reproduced with 2 F745 and a Kakute F7 with iNav 2.6 and 3.0.0.  F745 is supported by Betaflight and BT Nano works like a champ with it.)

Marcus-Barcus, 2021-05-23:
I think Flywoo has redone Hex to get the BT Nano to work in iNav but save your uart as SpeedyBee iNav works a treat using a patch cable.

BenWThomas, 2021-05-23:
iNav 3.0.0 with 3" GP defaults flies butter-smooth on F745Nano Stack.  Everything works fine, although I forgot to foam the baro.  Some PID tuning to be done, of course.  NIN 1404s have a nice smooth sound and responsiveness.  Hover and light maneuvering in front yard under fail-safe live oak went well; was even able to catch a little video one-handed.  LEDs on FCU are set for navigational standard red=port, green=starboard; LEDs on ESC double=flash bright white once a second.

BenWThomas, 2021-05-23:
https://user-images.githubusercontent.com/4916041/119268033-a87ae380-bbb6-11eb-94f4-a6b72f94fe81.mov

BenWThomas, 2021-05-23:
Initial GPS lock took 1 1/2 min+.  550mAh 3s started at 4.10/cell, after ~4 min testing, 3.79.  Lookin' good!

kingasa47, 2021-05-30:
Ended up going back to betaflight tried adjusting the pids just wouldn't fly right if anyone has better pids to share I would be willing to give it a try again

BenWThomas, 2021-06-17:
@kingasa47  --  
iNav Configurator 3.0.0 was posted a couple days ago, with official Flywoof745nano support.

The 4" Explorer in the hover movie was the unsigned firmware, and beta 3.0.0 Configurator.
I am using default iNav PIDs from an iNav preset, on F745 Nano 35A stack with GPS and a separate I2C compass module with 3750 kV motors and 4024 2-blade props

Stabilization (Angle Mode) was enabled.  It is smooth and level without stabilization, but tuning will be required to fly it easily in Acro: PID adjustments for Basic/Acro mode are still needed, and after I get those better set, I will need to work with the Baro/Mag PIDs.  (When in full Nav and stabilization modes, the automatics smooth it out very well.  The FC is so fast and smooth that it's not really obvious that this costs power.)

Only one hand on the transmitter while filming with my phone, so it was drifting around a 1-2 meter area while on Nav Pos Hold.  I got too close and had to dodge it once...

PIDs are a black art, but what works will be related to the mass, mass distribution, the moment arms, power system, and props.

iNav's several Quad Presets are intended for 3-6" models of various sorts.

I chose the Generic 3".  (I'd also tried the 3" Cinewhoop, which was nowhere near close.)
I built my Flywoo 4" Navigator from a bare frame kit:

   197.4 grams with battery total all-up weight
   550 mAh 70c 3s lipo 
   3750 kV Flywoo NIN 1404 motors
   4" 2.4 pitch 2-blade props
   
The dead cat frame design puts the center of mass without the battery very far aft.
To get CG at center point between the four motors, which is about where the middle standoff posts of the frame are, I had to strap the battery rather far forward.  Balance is important.
The FCU (with its gyros and accelerometers) is actually about 15 cm forward of the CG.

Here a screen shot of those default PID settings for the model described above.  They would probably work pretty well as a starting point for another 3-4" quad of similar size, mass, geometry, and power systems.  If your model is much different than mine, these PIDs might work even better -- or not as well.

[image]

Something else that may help you.  There is an awesomely mathematical Wiki page on PID controllers, here:

    https://en.wikipedia.org/wiki/PID_controller

Hilariously, for many of us, the great part of this article is a single small paragraph entitled "Manual Tuning" about halfway down the page, buried inside all that calculus and talk of specialized tuning hardware for industrial systems.

    https://en.wikipedia.org/wiki/PID_controller#Manual_tuning

In 5 meaty sentences, it outlines a practical strategy for tuning PIDs, from scratch.  (It assumes all PIDs at '0' as a staring point.  And of course, you have to do these one axis at a time... but understanding this strategy helps somewhat with the intuition of how all those other factors affect the behavior and settings.

I hope this is helpful.