iNavFlight/inav #6680: Pitch / turn assist behaviour Matek F765Wing versus F405wing , somehow related to pitot_hardware = ADC

wilco1967, 2021-03-06:
version
- INAV/MATEKF405SE 2.6.1 Feb 25 2021 / 10:39:37 (64249b1a)
- INAV/MATEKF765 2.6.1 Feb 25 2021 / 10:55:13 (64249b1a)

Today I changed the flightcontroller on my MiniTalon from a Matek F405Wing, to a Matek F765Wing

Did the recalibration, and made the CLI configuration the same as on the F405W. Only specific F765W parameters are different.

Everything seemed to work on the bench.
On the first flight, noticed some very weird response, especially when flying in acro.
As soon as I roll a little, the plane pitches up violently.
Landed in one piece.

Also seems to happens in angle, or cruise mode, only not as clear (the stabilization tries to deal with it, but it is still noticable).

Back on the bench, while in Acro, I noticed, that if I roll the plane, the elevators pitch up strongly. At 45 deg roll, the elevators are already full up.

Checked with the F405 again, and an the F405, the effect it is a lot less. I need about 80 deg roll, before full pitch up.
Pitching up happens because I have turn assist mode active. But the effect is a lot stronger on the 765, as it did on the F405, everything else being the same.

Went through the differences in the CLI parameters, and found one that stood out.

The F765 has
set pitot_hardware = ADC

the F405 had
set pitot_hardware = none

I don't even have a pitot tube installed.

Not knowing what caused it, I simply set it to set pitot_hardware = none on the F765 as well, not expecting anything to change.
However, it made a clear difference.

When rolling the plane now, the elevators are much less agressive in pitching up. Now I need to roll the plane to approx 80deg before the elevators are full up. This is about the same as the F405 responded.

So looks a lot better now....

Also tried to set the F405 to set pitot_hardware = ADC, but that doesn't seem to make a difference on the F405.

Haven't flown it again. Will try tomorrow if the weather cooperates, but pretty confident it will act 'normal' again.

What I don't understand is WHY would setting pitot_hardware = ADC or NONE have anything to do with pitch response ?

I also don't have any yaw mixed in the elevators (MT is a V-Tail). only stabilized pitch.

If I deactive the turn assist on the modes tab, there is no elevator action at all, as expected, but I prefer to have a little turn assist active.

avsaase, 2021-04-03:
Turn assist uses the bank angle and the airspeed to calculate what rate of turn in the earth frame it should target. This is then rotated to the pitch and roll axes resulting in elevator and rudder movement. When you have a pitot tube, turn assist will use the actual airspeed in this calculation, otherwise it uses a fixed reference airspeed, `fw_reference_airspeed`. This explains the difference your observe. 

In the next release, turn assist will use the target bank angle instead of the actual bank angle for its calculation. This means turn assist will no longer be compatible with acro mode since that doesn't have a target bank angle. This is actually the first time I hear of someone using turn assist with acro mode so I didn't think this would be a problem. If it is, let me know then I'll figure something out to keep compatibility with acro.

DzikuVx, 2021-04-07:
If you do not have a Pitot tube, Pitot tube sensor should be disabled.

closed 2021-04-07 by DzikuVx