iNavFlight/inav #6682: WP mission landing overrides RTH landing setting

breadoven, 2021-03-06:
I was playing with WP mission landings today and noticed the following issue.

- Run a mission and set the last WP as LAND (not RTH landing)
- Switch out of WP mode before the landing completes (fixed wing so didn't want it to actually land)
- Switch to RTH mode, plane returns to launch and then tries to land even though normal RTH landing is set OFF

I'd already noticed the following code in `navigationRTHAllowsLanding` which doesn't appear to have any function since it's not called from anything related to WP missions only RTH_HOVER_PRIOR_TO_LANDING.:

https://github.com/iNavFlight/inav/blob/3c863c02591a272bd99b71d297d6e2cc10c15657/src/main/navigation/navigation.c#L3597

It caused the unwanted RTH landing because the active WP action remained set on landing after cancelling WP mode triggering a landing when `navigationRTHAllowsLanding ` was called from RTH hover prior to landing.

Does this code serve some purpose or is it left over from something else ?

stronnag, 2021-03-07:
It serves no purpose and looks like a left over or misunderstanding. I think it should be removed.

breadoven, 2021-03-08:
Code has been removed in https://github.com/iNavFlight/inav/pull/6524.

I'll close this shortly if there's no other feedback.

closed 2021-03-12 by breadoven