iNavFlight/inav #6684: INAV + Kakute F7 AIO 1.5 + ALTHOLD = White Smoke

blind-oracle, 2021-03-07:
I'm a bit confused, but here it is.

**Two of my Kakute F7 were fried when I activated ALTHOLD**

It flies ok in angle mode, but then, when I activate ALTHOLD - the copter immediately shuts down and drops.
After that I see the smoke coming out of it - after unsoldering it looks like that internal FC's BEC is fried, but hard to say for sure.

Hardware:
- 5" quad
- Kakute F7 AIO 1.5
- GPS + Mag (i2c)
- Tekko32 Metal ESCs (latest BLHeli32)

Also, not that it should matter:
- Atlatl VTX
- Frsky R-XSR RX
- Taranis X9D

## Steps to Reproduce
- Assemble, flash latest INAV
- Setup, calibrate ACC/MAG etc
- Assign ANGLE/HORIZON/NONE to one 3-pos switch
- Assign NONE/ALTHOLD/POSHOLD to another
- Wait for GPS lock, arm, take off
- Fly around a bit, enable ALTHOLD
- Watch copter fall in white smoke

When it happened for the first time I thought that I might've fucked up some wiring or had caught a short circuit somewhere, but coincidence with enabling ALTHOLD was strange. Then I've ordered the 2nd FC, carefully soldered everything again - and *bam*, same shit.

Before iNAV I flew with this setup for quite some time on Betaflight without issues.

I'd be grateful for any lead. For now I've ordered Matek F722-SE just to be sure that hardware isn't to blame.

blind-oracle, 2021-03-07:
There's a blackbox log, but it looks strange to me due to very high gyro values: https://cloud.novg.net/s/FW5GsD4DC9zP5cc/download

stronnag, 2021-03-07:
Your log is sufficiently corrupt as to be unreliable for any kind of analysis.

blind-oracle, 2021-03-07:
@stronnag Yep, that's what I thought. Any ideas what might caused this? It's configured to log 25% of loops.
And stick positions seem to be logged correctly...

stronnag, 2021-03-07:
Probably the SD is not of good enough quality. For example, some of the metadata at the begining is missing:
```
H Firmware revision:INAV 2.6.0 (cada68e4) KAKUTEF7
H Firmware date:Dec  4 2020 11:38:46
H Log start datetime:2021-03-06T17:12:0H thr_expo:0
H tpa_rate:0
```
Notice how the `H thr_expo` value is joined to the  `H Log start datetime` value. There is also data missing, here's an example of the same section from a good log:
```
H Firmware revision:INAV 2.5.0 (2501949e9) MATEKF405
H Firmware date:May 21 2020 14:01:17
H Log start datetime:2020-05-22T13:35:19.870+01:00
H Craft name:Miss Piggy
H P interval:1/16
H minthrottle:1060
H maxthrottle:2000
H gyro_scale:0x3f800000
H motorOutput:1060,2000
H acc_1G:2048
H vbat_scale:112
H vbatcellvoltage:33,35,42
H vbatref:1263
H looptime:1000
H rc_rate:100
H rc_expo:70
H rc_yaw_expo:20
H thr_mid:50
H thr_expo:0
```

If I run your log through a tool to show state changes:
* There is no ALTHOLD indicated
* The final entries (iteration 14592) are corrupt nonsense.
```
$ inav_states.rb /tmp/LOG00001.TXT 
2.6.0 using states for 2.5.0
LOG00001.TXT: INAV 2.6.0 (cada68e4) KAKUTEF7
Iteration	Time(s)	Elapsed(s)	State	FltMode
        0	  90.9	(   0.0)	nav_state_undefined (0)	 ARM|ANGLE	
    14592	   1.8	( -89.1)	FAILSAFE=>RX_LOSS_DETECTED (1030423,-1008459,8,-30);(18446744073709551615,18446744073709551615)
    14592	   1.8	( -89.1)	 (-52)	 ARM|HEADFREE|HEADADJ|NAVRTH|BEEPERON|LEDLOW|LIGHTS	
    14592	   1.8	( -89.1)	end of log
```
The penultimate line in the decoded log had an iteraton count of 127808, the final entry is just random bits; the mode is junk, time has gone backwards, the nav state value (-52) is not valid. There is no `End of log` entry.  Some of is most likely  a consequence of the fire.

Still, I can't image how the firmware could cause the board to smoke.

blind-oracle, 2021-03-07:
@stronnag 

Thanks, I'll switch to another SD next time.
Though Betaflight logged correctly with this one, but who knows what happened to it during these white smoke events...

> The penultimate line in the decoded log had an iteraton count of 127808, the final entry is just random bits; the mode is junk, time has gone backwards, the nav state value (-52) is not valid. There is no End of log entry. Some of is most likely a consequence of the fire.

After the crash FC rebooted itself, so the BEC probably isn't dead completely. Probably this lead to these last entries (they were added after reboot?) Or it should have produced another log file (there is none)?

> Still, I can't image how the firmware could cause the board to smoke.

Same thing, if I didn't have this experience two times in a row I would not believe it... 

Just unsoldered the 2nd FC - the damage seems to be exactly the same.

These two square-shaped elements are inflated a bit and have burn marks on the side. What are they I don't know as Kakute is closed-source HW.

Also I found this topic: https://discuss.ardupilot.org/t/kakute-f7-aio-power-problem/53626
It looks like that this is a more or less common problem... the same square element, same white smoke.
But those guys got the smoke on the 1st battery plug, not switching modes like me.

Behavior like mine is also here: https://www.dronetrest.com/t/kakute-f7-v1-5-smoked-should-i-keep-flying-it/9318
Some speculate that it might be 5V overload.

Probably switching mode to NAV causes some sort power surge or something like this which blows the FC. But what exactly happens is a mystery...

[image]

[image]

shellixyz, 2021-03-07:
I cannot see how this could be software related. Closing

closed 2021-03-07 by shellixyz