iNavFlight/inav #6688: RPLidar A1 integration

J40yke, 2021-03-08:
No facility to include a 360 degree lidar sensor. Hardware connection is possible via UART ports. However iNav doesn't currently support setup. 

## Desired Behavior

To either include a simple graphical representation on an OSD tab with distance figures displayed on screen (Birdseye view, with a display similar to automobile parking sensors), and/or (but less importantly) maybe some obstacle avoidance for use in Rover modes. 

## Suggested Solution

Release a beta version for testing. I have a rover (boat) setup ready to be used for testing.

## Who does this impact? Who is this for?

Aimed for iNav Rover users, for visual navigation in daylight/GPS denied environments. For example, boats in tunnels, or cars/tanks in confined spaces.

Kableking, 2021-03-08:
I have been looking at putting on RPLidar A1 on my Rover for sometime, but having to move to Ardurover put me off.

J40yke, 2022-01-09:
Will there likely be any moves towards implementation of the RPlidar for the use on Rovers? Even just from a Telemetry viewing perspective? Be great to see a "map" of the scanned data similar to ardurover.

b14ckyy, 2024-03-23:
There is just not enough storage and calculation power to handle the data of a 360° scanning lidar, let alone to project any kind of augmented reality in our OSD feed. This would need a companion computer that does all the work and then controls inav. Has nothing to do with the firmware itself.

closed 2024-03-23 by b14ckyy

J40yke, 2024-03-23:
> There is just not enough storage and calculation power to handle the data of a 360° scanning lidar, let alone to project any kind of augmented reality in our OSD feed. This would need a companion computer that does all the work and then controls inav. Has nothing to do with the firmware itself.

Fair enough. I only asked because the same F405_Std I'm currently using the RPLIDAR on with AP firmware, can also run inav. The hardware is capable as it's doing it already, but just wondered if inav was capable.