iNavFlight/inav #6698: Incorrect max speed on summary screen after flight

Beardyblair, 2021-03-10:
## Current Behavior
Following a flight, the maximum speed obtained does not always match the speed in the summary. 

[image]

[image]

## Steps to Reproduce

1. Fly
2. Land and disarm to view summary

## Expected behavior
Speeds to be the same in both locations

## Suggested solution(s)
I will go check code but expect this is due to rounding on one variable and removing decimals on another.

## Additional context
Dump not required as its not settings based.

---

# INAV/MATEKF722MINI 2.6.1 Feb 25 2021 / 10:53:27 (64249b1a)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

erstec, 2021-11-15:
@Beardyblair Can you show where you see 'rounding' on one variable and ' removing decimals' on another?

As I see from the code there is fundamental difference between 'live' OSD and 'stats' screen (in your case)
'Live' shows max_speed
'Stats' shows 'max_3D_speed', which includes calculations of Z-axis speed.

MrD-RC, 2021-11-15:
The summary screen has always shown the maximum speed in 3D space. The speed on your OSD is ground speed.

closed 2023-10-27 by DzikuVx