iNavFlight/inav #6702: Does osd_row_shiftdown work any more ?

breadoven, 2021-03-11:
osd_row_shiftdown doesn't seem to have any affect.

It's included in settings.yaml but doesn't appear to be used anywhere in the code in any obvious way.

Does it still work ?

breadoven, 2021-03-11:
Seems perhaps it's a phantom setting:

https://github.com/iNavFlight/inav/issues/3322

OptimusTi, 2021-03-11:
Looks like this setting was from a previous version. Removed in #2028

breadoven, 2021-03-11:
I was trying to use it to avoid a cut off Safehome message on the arming screen but it appears it's the start row alignment on the arming screen that's probably the issue since it allows the top row to start on row 0 for NTSC (least on mine) which means half the text is off the top of the screen. The Disarm Stats screen is hard coded to start on row 1, the first full row, so it's odd the Arming screen doesn't do the same.

The Arming screen sets the top row using:
https://github.com/iNavFlight/inav/blob/ee062031515f4ca6c775a793daf565185e888bb5/src/main/io/osd.c#L3121

which doesn't make much sense. If you want 12 visible rows then it would be easier to use the following instead which ensures the top row starts on at least the first full row, 1 and otherwise centres in the middle of the screen.
`uint8_t y = osdDisplayPort->rows > 13 ? (osdDisplayPort->rows - 12) / 2  : 1;`

closed 2021-03-12 by breadoven