iNavFlight/inav #6706: Expected/Measured RPM mismatch alarm

expipiplus1, 2021-03-12:
## Desired Behavior

It would potentially be quite useful for inav to be able to inform the pilot of a mismatch between the expected RPM from a throttle command and the measured RPM. 

This could be used to detect:

- Loose/detached props
- Malfunctioning motors
- ESC temperature throttling
- ESC current throttling 

The ESC throttling ones are obviously better done in the ESC firmware and reported over telemetry, but with many closed source ESC firmwares what can you do :shrug: ...

See also: https://github.com/iNavFlight/inav/issues/5815

## Suggested Solution

- inav somehow has a measure of the expected RPM for a given power
- inav watches for gross deviation of the RPM reported by the ESC from this value
- inav reports this to the user via OSD or telemetry

## Who does this impact? Who is this for?

Anyone with a telemetry enabled ESC

DzikuVx, 2021-03-12:
The problem is that FC does not know the expected RPM.

expipiplus1, 2021-03-13:
Yeah, that is a challenge, hopefully this issue can help nail down
something sensible.

I wonder if it's possible to learn this relationship either during flight,
from blackbox logs or on a test stand.

One would also expect the throttle->RPM function to be non-decreasing (more
or less).

expipiplus1, 2021-03-31:
Of course embedding this in the ESC telemetry would be best. Where is the best place to request such a feature for DShot?

closed 2022-02-10 by DzikuVx