iNavFlight/inav #6707: Auto disarm on crash for autolaunch

fernandodlc, 2021-03-13:
When using autolaunch, especially with large models, you often don't have the radio actually in your hands for launch. If there is an issue the motors will not disarm and will potentially damage themselves, the props and the plane on an unscheduled landing event. Getting to the disarm switch fast enough in such situations whilst in a panic is not particularly easy.
 
Add a feature that detects a crash and auto disarms if the autolaunch procedure is active.

Suggest detecting gyro acceleration over a certain level that would be consistent with a hard landing or impact.

Useful for anyone using autolaunch.

Some care might need to be taken to avoid a missed launch event crash actually starting the launch procedure. For example when the launch acceleration is not high enough to trigger the launch and the model is effectively just thrown to the ground.

avsaase, 2021-03-13:
Flying/not flying and crash detection is discussed in #2331.

fernandodlc, 2021-03-13:
@avsaase Great, that was a bit buried to show up in my search. Just trying to contribute.

FPVZaphod, 2021-05-24:
I think this is not exactly same like #2331 - it only applies for the short period of the Autolaunch. 
Possible solution: After launch detect, a 2 secs grace period starts. If climb rate after this 2 secs is not positive (or smaller than X m/s), then overwrite throttle to zero for 5 secs. If then still no positive climb rate, then disarm. 
The 5 secs of throttle zero before full disarm it to prevent problems in a short loss of altitude while launching, the first to seconds is because even with bungee launch the plane drops a bit in the first second.

closed 2022-02-10 by DzikuVx