iNavFlight/inav #6711: Add support for iFlight Succex-E Mini F7 2-6S Flight Controller

GQ4Life, 2021-03-16:
**## Current Behavior**
The Succex-D F7 Twing though same resources the gyro doesnt work if used on this flight controller. 

**## Desired Behavior**
Everything to work. I have GPS with Compass, Baro on board,

**## Suggested Solution**
I tried to make myself by editing other iFlight target but I cant get my system to compile targets

**## Who does this impact? Who is this for?**
For long range pilots. Same Flight controller is in Chimera5 LR . iFlight says there is a target for it. But I can not find one. 

**## Additional context**
If someone could help if would be great. 
My **iFlight Succex-E Mini F7 v1** dump (Not sure why no resource for Baro I had it clicked on. Drone isnt flyable yet. So not fully setup and tested. 

# 

# dump

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.4 Oct 20 2020 / 08:20:06 (fbcaf8c50) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F7, version: c7fde51e, date: 2020-08-26T04:30:34Z

# start the command batch
batch start

board_name IFLIGHT_SUCCEX_E_F7
manufacturer_id IFRC

# name: -

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
[... 1043 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1750
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

DivingFalconFPV, 2021-03-17:
I second. I have a LR drone with this flight controller. Needed 20x20, baro, and enough uarts

zoobs, 2021-08-01:
I third. I have Chimera 5 with this controller and require firmware for decent GPS.

G9xy, 2021-08-04:
iFlight posted INAV target for their line of FC at their official support Google Drive : [https://drive.google.com/drive/u/0/folders/1r3MNTBNYSzGeRXQg45j4IJVoW1Xjk7Y0](url)

Working INAV target for my SucceX-E Mini F7 FC (Chimera 5 - analog - BNF) is "inav_3.1.0_IFLIGHT_SUCCEX_E_F722_V1.hex". 

(be aware that "inav_3.1.0_IFLIGHT_SUCCEX_E_MINI_F722_V1.hex" file does not work for my SucceX-E Mini F7 - gyro and baro sensors are not recognized)

You would need INAV 3.1.0 configurator found at [http://seyrsnys.myzen.co.uk/inav-configurator-next](url) for flashing SucceX-E Mini F7 FC (Chimera 5 - analog)

[image]

cazungas, 2021-08-06:
I cannot find this target in the google drive folder...is it still available?
The only thing I see are betaflight targets...
Please help

fanedi, 2021-08-06:
no more iNAV targets in iflight google drive ... strange !

cazungas, 2021-08-07:
Would someone mind sharing the target
"inav_3.1.0_IFLIGHT_SUCCEX_E_F722_V1.hex".

G9xy, 2021-08-08:
You can't send a direct private message to another GitHub user. Some folks share their Twitter @username on their GitHub profile...

cazungas, 2021-08-08:
Ok...I updated my profile...thanks

fanedi, 2021-08-10:
I am interested by this target .. conversion of my Chimera 5 analog BF to Inav ... but newbee

pedram28july, 2021-08-25:
Hello guys, I have looked for "inav_3.1.0_IFLIGHT_SUCCEX_E_F722_V1.hex" a lot, but failed to find it. Could anyone please upload it on google drive and share the link here. Or just email me at pedram28july@gmail.com. I really appreciate your kind help.

pedram28july, 2021-08-25:
> iFlight posted INAV target for their line of FC at their official [iFlight Support Google Drive](https://drive.google.com/drive/u/0/folders/1r3MNTBNYSzGeRXQg45j4IJVoW1Xjk7Y0)
> 
> Working INAV target for my SucceX-E Mini F7 FC (Chimera 5 - analog - BNF) is "inav_3.1.0_IFLIGHT_SUCCEX_E_F722_V1.hex".
> 
> (be aware that "inav_3.1.0_IFLIGHT_SUCCEX_E_MINI_F722_V1.hex" file does not work for my SucceX-E Mini F7 - gyro and baro sensors are not recognized)
> 
> You would need [INAV 3.1.0 configurator](http://seyrsnys.myzen.co.uk/inav-configurator-next) for flashing SucceX-E Mini F7 FC (Chimera 5 - analog) with this particular INAV 3.1.0 target.
> 
> [image]

Hello guys, I have looked for "inav_3.1.0_IFLIGHT_SUCCEX_E_F722_V1.hex" a lot, but failed to find it. Could anyone please upload it on google drive and share the link here. Or just email me at pedram28july@gmail.com. I really appreciate your kind help.

Odin263, 2021-09-26:
[inav_3.0.2_IFLIGHT_SUCCEX_E_F7_MINI.zip](https://github.com/iNavFlight/inav/files/7232309/inav_3.0.2_IFLIGHT_SUCCEX_E_F7_MINI.zip)
Here is a target with fixed turtle mode for iNAV 3.0.2

Odin263, 2021-09-27:
> Thank you very much Bro for sharing the file with me. Warm Regards Pedram
> […](#)

But be careful! Apparently I've rotated the alignment by 90 °. If necessary, I can re-upload it with the correct orientation.

fpvphil, 2021-10-31:
Works, fc and mag off by 90. Thx for sharing

G9xy, 2021-10-31:
Besides proper configuration for particular orientation of GPS, it is essential that GPS is flat  to the FC. With this iNAV 3.1.0 target none of GPS magnetic offset works tilted position of GPS. I have struggled for couple of weeks, than decided to 3d print new GPS frame with flat elevation to the FC and it worked flawlessly for me.

DivingFalconFPV, 2021-10-31:
> Besides proper configuration for particular orientation of GPS, it is essential that GPS is flat to the FC. With this iNAV 3.1.0 target none of GPS magnetic offset works with tilted position of GPS. I have struggled for couple of weeks, than decided to 3d print new GPS frame with flat elevation to the FC and it worked flawlessly for me.

Really?? Shit. Normally I do -300 roll and it helped. Thpugh my gps was titled like 45° so even in forward flight it was always on an angle

aniksuper, 2021-12-25:
Can anybody post the firmwares here? For im looking for the firmware for the succex e f7. @G9xy @Odin263

G9xy, 2021-12-26:
The target is posted 7 posts up here by Odin263.

Odin263, 2021-12-26:
[inav_4.0.0_IFSUCCEX_E_F7.zip](https://github.com/iNavFlight/inav/files/7777583/inav_4.0.0_IFSUCCEX_E_F7.zip)
should run but haven't tested it

aniksuper, 2021-12-26:
> The target is posted 7 posts up here by Odin263.

I assumed it works only for the mini version. BUT I flashed it and tried, work fine on the big board.

> [inav_4.0.0_IFSUCCEX_E_F7.zip](https://github.com/iNavFlight/inav/files/7777583/inav_4.0.0_IFSUCCEX_E_F7.zip) should run but haven't tested it

will try next week, thank you!

Odin263, 2021-12-29:
[inav_4.0.0_IFSUCCEX_F7_E.zip](https://github.com/iNavFlight/inav/files/7790344/inav_4.0.0_IFSUCCEX_F7_E.zip)
Since the turtle mode was introduced, i often had the issue that after using the turtle mode, the motors did not all turn in the right direction. I have not found any reports with the same issue. It may be due to this FC or the IFlight ESC. I was able to solve this issue for myself and create the hex file for it. So far, I have not been able to determine any disadvantages due to the adjustment.

QVirus, 2021-12-30:
Hi @Odin263 ,

I'm trying your INAV 4.0 FW with my Nazgul 5 Evoque.
I have permanently I2C Error code: 45.
The Baro doesn't work (DPS310). After enabling the baro I receive the I2C Error code: 50.

> status
System Uptime: 21 seconds
Current Time: 2041-06-28T02:04:00.000+01:00
Voltage: 22.50V (6S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1820
I2C Errors: 50, config size: 8792, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 14, cycle time: 528, PID rate: 1893, RX rate: 76, System rate: 9
Arming disabled flags: CAL HWFAIL CLI DSHOTBEEPER
VTX: not detected

#

Odin263, 2021-12-30:
Hi @QVirus,

I also have I2c Error Code 45 permanently and I suspect that it is related to the non-existent speed sensor or something similar. First I tried to copy all pin assignments from betaflight without changing them. But since everything else works for me, I have not yet tried to eliminate this error. But I'll try it soon.
With the Baro, I don't quite understand why it doesn't work for you, because I have the same Baro with the same FC (DPS310) and as long as the batteries are connected, it works properly.

# status
System Uptime: 15 seconds
Current Time: 2041-06-28T03:04:00.000+02:00
Voltage: 7.57V (2S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=DPS310, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1924
I2C Errors: 45, config size: 8792, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 13, cycle time: 513, PID rate: 1949, RX rate: 49, System rate: 9
Arming disabled flags: NAV RX CLI
VTX: band: F, chan: 1, power: 1 (25 mW), freq: 5740 MHz

Odin263, 2021-12-30:
@QVirus

I commented out the temperature sensor in target.h as follows.
[image]
I exchanged the new .hex file above. This means that the I2C error code: 45 no longer appears for me.

fanedi, 2021-12-30:
what a mess ... which one is the latest F7 mini  compliant ?

Odin263, 2021-12-30:
@fanedi 

Above is version 3.0.2 that worked. Version 4.0.0 is a little deeper, as it was taken over from betaflight. In the last version 4.0.0 I manipulated the FW a bit and eliminated the I2C error. All 3 versions work for me. As far as I know, there are no differences between the mini and the large board. Just choose what you want!

fanedi, 2021-12-30:
thanks Odin263

OptimusTi, 2021-12-30:
What does I2C Error 45 means?

Odin263, 2021-12-30:
@OptimusTi 
I suspect I2C Error 45 relates to the temperature sensor which does not exist and therefore no communication is possible.

OptimusTi, 2021-12-30:
I thought that was a counter of some sort. Still trying to figure out 50 and 32.   Anyway, found this little gem in the docs:  
[image]

Odin263, 2021-12-31:
You're not wrong about the counter. But different modules return different numbers of errors. For example, if the rangefinder is defined in INAV, but none is available, 5 more I2C errors are displayed. With the temperature sensor there would be 50 errors. Compass were 10 I2C errors. I haven't tried how many errors the other interfaces generate. But it is actually not that easy to identify clear causes. (It's still just my guess. I don't really have a clue about it.)

QVirus, 2022-01-03:
Hi @Odin263 ,

Thank you for the updated firmware.

I've checked my FC (SucceX-E Mini F722 V1.4) there is no Baro sensor on the board :(
I ordered a Baro sensor (BMP280 on the board with 5V support) on Ali and will connect it additionaly to the I2C.

xchipx, 2022-01-04:
Has anyone had a successful flight with the nazgul Evoque using these targets? Anyone able to share their tune?
 Thanks

xchipx, 2022-01-04:
Just to update everyone.. I'm using nazgul Evoque f5d analog , and the above 4.0.0 file allowed my barometer to work out of the box. (DSP)

Odin263, 2022-01-09:
Can someone tell me whether the targets I have uploaded are fully functional or whether problems have arisen?

MeapVR, 2022-01-17:
> Just to update everyone.. I'm using nazgul Evoque f5d analog , and the above 4.0.0 file allowed my barometer to work out of the box. (DSP)

I'm using iFlight Nazgul Evoque F5D Digital and when I flash the firmware the gyro does not work. Do you know how to fix this? What firmware .hex file are you using?

RipperGOT, 2022-01-18:
Thanks for your work friends! Too bad I can't be of help!

xchipx, 2022-01-18:
> I'm using iFlight Nazgul Evoque F5D Digital and when I flash the firmware the gyro does not work. Do you know how to fix this? What firmware .hex file are you using?

I just used the 4.0.0 one.
I have single gyro analog nazgul Evoque.
Gyro worked out of the box and barometer too.
I have not flown it yet.

MeapVR, 2022-01-18:
> > > Just to update everyone.. I'm using nazgul Evoque f5d analog , and the above 4.0.0 file allowed my barometer to work out of the box. (DSP)
> > 
> > 
> > I'm using iFlight Nazgul Evoque F5D Digital and when I flash the firmware the gyro does not work. Do you know how to fix this? What firmware .hex file are you using?
> 
> I just used the 4.0.0 one. I have single gyro analog nazgul Evoque. Gyro worked out of the box and barometer too. I have not flown it yet however.
> 
> I think the digital nazgul uses twin gyro FC?

I think I see the problem here.
The two Nazgul's have different flight controllers.
The iFlight Nazgul Evoque F5D Analog has the SucceX-E Mini F7
While the iFlight Nazgul Evoque F5D Digital has the SucceX-D Mini F7 (which is what I need)

MeapVR, 2022-01-18:
If anyone looking at this post has a Nazgul Evoque F5D Digital (DJI Caddx Vista/Polar) then note that it is using the iFlight Succex-D Mini F7.

I have opened a feature request for the digital version here: https://github.com/iNavFlight/inav/issues/7765

Odin263, 2022-01-18:
Hi @MeapVR, 
according to the description from Banggood, Nazggul5-Evoque F5D changed FC from Succex E mini to BLITZ mini in October 2021. This has slight differences in the pin assignment.
[image]

However, the target for the BLITZ mini is available in iNav
[image]
Myself I use the last version 4.0.0. But I didn't have any problems with the previous ones either, except for the turtle mode.

@RipperGOT,
As far as I know, there are no differences between the mini and the big board as far as the FW is concerned.

RipperGOT, 2022-01-19:
@Odin263  Thanks men!

I bought it at Bangood and it arrived on Friday.  
On the Iflight page it says that in December 2021 they changed the SucceX-E for the Blitz, but surely these bangood assholes send me the SucceX-E... 
It also says on the Iflight website that the two boards do not have baro, but I see that yours (2.0 boards) do have them.  

Does the Blitz also have baro?  I hope so, I love Inav and I would like to be able to use its navigation functions.

MeapVR, 2022-01-19:
> Hi @MeapVR, according to the description from Banggood, Nazggul5-Evoque F5D changed FC from Succex E mini to BLITZ mini in October 2021. This has slight differences in the pin assignment. [image] [...]

Hello @Odin263 I tried flashing the Blitz firmware as you suggested and the gyro seems to be working now. I have done some setting up and will attempt to fly it when I get some time on my hands, I'll keep you posted.

Odin263, 2022-01-19:
@RipperGOT

i think iFlight shipped some succex f7 e boards with and some without baro. Some have the BMP280 and some have the DPS310. Look here:

[image]

I got this one myself. 

[image]

This baro was not recognized with the published version of Betaflight, but I was able to remedy the situation. With the created target in iNav, it was recognized immediately.

According to Banggood's description, the BLITZ F7 BMP280 baro has:
[image]

But no baro can be found on the pictures for this FC:

[image]

Apparently some were delivered with and some without a baro.

RipperGOT, 2022-01-19:
> @RipperGOT
>
> i think iFlight shipped some succex f7 e boards with and some without baro. Some have the BMP280 and some have the DPS310. Look here:
>
> [image]
>
> I got this one myself.
>
> [image]
>
>
> This [...]

🤦🏻‍♂️ No comments...

OptimusTi, 2022-01-19:
[image]

Odin263, 2022-01-19:
> [image]

I don't think an exact date can be set. The problem is that the products are usually not bought from the manufacturer but from the retailer. Not every retailer will sort their inventory by date and sell everyone on time. It is therefore possible that FCs will still be delivered with a baro next year, even though the manufacturer no longer installs them. This is also the reason why RipperGOT received a Nazgul5 with SucceX-E FC on Friday. This is even more common with small parts.

RipperGOT, 2022-01-21:
> I don't think an exact date can be set. The problem is that the products are usually not bought from the manufacturer but from the retailer. Not every retailer will sort their inventory by date and [...]

Well, today I received the nazgul with the Succex E F7 board and the DPS310 baro. I guess the Blitz will be delivered in the next few months... Fu**
This weekend I will mount the receiver, gps, etc and I will test the Hex te Inav

RipperGOT, 2022-01-27:
Guys, Can you look at this?  #5549 
I don't know why the bidirectional dshot is not implemented in Inav.  We could use the rpm filter for nazgul5 .  I also have a hobbywing esc rotor 60a blheli_32 but it has no telemetry output.  Perhaps if many people ask for it, it will be implemented.

Odin263, 2022-02-09:
[inav_4.1.0_IFLIGHTF7_SUCCEXE.zip](https://github.com/iNavFlight/inav/files/8031629/inav_4.1.0_IFLIGHTF7_SUCCEXE.zip)
in case anyone wants to update

xpertcrush, 2022-02-09:
I need your help Odin. How to contact you?

Odin263, 2022-02-09:
Hello @xpertcrush,
you want to add baro qmp6988 to iNav? But it's not installed on your rush blade F7 board?

fpvphil, 2022-02-09:
> [inav_4.1.0_IFLIGHTF7_SUCCEXE.zip](https://github.com/iNavFlight/inav/files/8031629/inav_4.1.0_IFLIGHTF7_SUCCEXE.zip) in case anyone wants to update

Thanks for your efforts!

xpertcrush, 2022-02-09:
> Hello @xpertcrush,
> you want to add baro qmp6988 to iNav? But it's not installed on your rush blade F7 board?

It's on board, i have it working, problem is motor sound only upto idle rpm in motors tab, not higher even if i set it to 100%

xpertcrush, 2022-02-11:
> @xpertcrush 
> I am trying to create a target. If it works I'll upload it. You should open a new issue. Maybe someone has already dealt with this and has the right solution.
> If it works, this baro should be added to iNav via pull requests.

https://github.com/iNavFlight/inav/issues/7818
Here

xchipx, 2022-03-16:
Friends , im sorry to be a super noob, but PLEASE would someone recommend if the default inav pids/tune etc is suitable for my nazgul evoque?

Ever since flashing to inav in january ( see my comments above, i have the nazgul and working barometer etc ) I have been afraid to test fly my nazgul as I have not been sure if the default inav settings will be okay - ofcourse ive set up the failsafes and all those options, but i couldnt work out if the inav stock tune was anywhere near to the betaflight tune these drones shipped with ( This is the first time i set up an inav quad, I have set up a few wings )

supernova1969a, 2022-05-06:
> [inav_4.0.0_IFLIGHT_SUCCEX_F7_E(fixed i2c error 45).zip](https://github.com/iNavFlight/inav/files/7834011/inav_4.0.0_IFLIGHT_SUCCEX_F7_E.fixed.i2c.error.45.zip)
> 
> Since the turtle mode was introduced, i often had the issue that after using the turtle mode, the motors did not all turn in the right direction. I have not found any reports with the same issue. It may be due to this FC or the IFlight ESC. I was able to solve this issue for myself and create the hex file for it. So far, I have not been able to determine any disadvantages due to the adjustment.

Hello. I also am going to use the succex e f7 v 1.4 target file. Do you have it for use with INAV 4.1? Thanks

supernova1969a, 2022-05-11:
> I second. I have a LR drone with this flight controller. Needed 20x20, baro, and enough uarts

I need one as well

> [inav_4.1.0_IFLIGHTF7_SUCCEXE.zip](https://github.com/iNavFlight/inav/files/8031629/inav_4.1.0_IFLIGHTF7_SUCCEXE.zip) in case anyone wants to update

Thanks Odin!!!

supernova1969a, 2022-05-16:
> Friends , im sorry to be a super noob, but PLEASE would someone recommend if the default inav pids/tune etc is suitable for my nazgul evoque?
> 
> Ever since flashing to inav in january ( see my comments above, i have the nazgul evoque analog and working barometer etc ) I have been afraid to test fly my nazgul as I have not been sure if the default inav settings will be okay. Ofcourse ive set up the failsafes and all those options, but i couldnt work out if the inav stock tune was anywhere near to the betaflight tune these drones shipped with ( This is the first time i set up an inav quad, I have set up many wings in inav and ardupilot ) Thanks

I have the same quad Iflight Nazgul5 v2 .bn-880 with bm-280 baro..INAV 4.0. yes it works great!!! There are a few bugs....but that's the fun!!!!

supernova1969a, 2022-05-22:
> Can someone tell me whether the targets I have uploaded are fully functional or whether problems have arisen?

I used your INAV 4.0 Iflight succex  E F7 target..works awesome!!! Could you make one for INAV 5.0? That would be fantastic!!

[20 more replies]

closed 2022-09-03 by DzikuVx