iNavFlight/inav #6715: NAV Waypoint mission resets after FC reconnects in failsafe mode

HerraMakkara, 2021-03-17:
Current Behavior

failsafe_mission is set to OFF
Mission keeps going when I switch RC link off, but I get ”move sticks to exit FS” and the mission resets and start fron the beginning.

Steps to Reproduce

1 - RX link lost - failsafe in action, waypoint mission continues
2 - RX link restored - WP mission reset, asks to move sticks to exit failsafe mode 
3 - WP mission starts from the beginning

Expected behavior
Failsafe is set to continue mission, which is does and continue mission when FC reconnects from failsafe.

Suggested solution(s)

- FC Board name and vendor: Matek wing F765
- INAV version string: Inav/matekf765 2.6.0 Dec 4 2020 / 12:27:03 (cada68e4)
- Gcc-9.2.1 20191025

HerraMakkara, 2021-03-17:
[
[image]
[image]
[image]

](url)

breadoven, 2021-03-17:
A log would be useful.

Have you checked what the Aux channels do during Failsafe, see if they cause modes to do unexpected things ?

stronnag, 2021-03-17:
IIRC, it has always been the case that restarting a mission restarts from the begining, by design.

avsaase, 2021-03-17:
> IIRC, it has always been the case that restarting a mission restarts from the begining, by design.
But after a failsafe recovery when a mission is active, it should continue the mission instead of starting over. That makes more sense to me than what @HerraMakkara observed.

stronnag, 2021-03-17:
From the report, I understood the OP moved to sticks to exit F/S which caused the mission to restart. Had the sticks not been touched, the mission should have completed. 

https://www.rcgroups.com/forums/showpost.php?p=46747363&postcount=27589 tells a possibly more complete version of events.

Whether the extant design, which precedes "continue on F/S" is still valid when  "continue on F/S" is enabled is another question.

HerraMakkara, 2021-03-17:
It’s like moving throttle stick with WP mission on going, it doesn’t affect the mission. But I think in my case the mission is resetted right away RC link is restored.

HerraMakkara, 2021-03-17:
> From the report, I understood the OP moved to sticks to exit F/S which caused the mission to restart. Had the sticks not been touched, the mission should have completed. [...]

It’s like moving throttle stick with WP mission on going, it doesn’t affect the mission. 

Comfirmed from the flight recrording: In my case the mission is resetted right away RC link is restored. Touching sticks is not the reason.

HerraMakkara, 2021-03-17:
> A log would be useful.
> 
> Have you checked what the Aux channels do during Failsafe, see if they cause modes to do unexpected things ?

Flight log enclosed:
https://www.dropbox.com/s/fl4loadrb5qhq2y/INAV.TXT?dl=0

HerraMakkara, 2021-03-17:
This all happens when RC link is restored

[image]

stronnag, 2021-03-17:
```
$ inav_states.rb INAV.TXT 
2.6.0 using states for 2.5.0
INAV.TXT: INAV 2.6.0 (cada68e4) MATEKF765
Iteration	Time(s)	Elapsed(s)	State	FltMode
        0	 117.0	(   0.0)	nav_state_undefined (0)	 ARM|ANGLE|BLACKBOX	
    63328	 180.5	(  63.6)	nav_state_waypoint_in_progress (17)	 ARM|BLACKBOX|NAVWP	
    99872	 217.2	( 100.2)	nav_state_idle (1)	 ARM|BLACKBOX	
   102112	 219.4	( 102.5)	nav_state_waypoint_in_progress (17)	 ARM|BLACKBOX|NAVWP	
   145408	 262.9	( 145.9)	FAILSAFE=>RX_LOSS_IDLE (1495,1505,1499,1497);(0,1)
   188800	 306.4	( 189.5)	nav_state_idle (1)	 ARM|BLACKBOX	
   188864	 306.5	( 189.5)	nav_state_waypoint_in_progress (17)	 ARM|BLACKBOX|NAVWP|GCSNAV	
   209856	 327.6	( 210.6)	FAILSAFE=>IDLE (1495,1505,1455,1428);(1,1)
   246016	 363.9	( 246.9)	FAILSAFE=>RX_LOSS_IDLE (1495,1507,1499,1323);(0,1)
   261632	 379.5	( 262.6)	nav_state_idle (1)	 ARM|BLACKBOX|GCSNAV	
   261664	 379.6	( 262.6)	nav_state_waypoint_in_progress (17)	 ARM|BLACKBOX|NAVWP|GCSNAV	
   272416	 390.3	( 273.4)	FAILSAFE=>IDLE (1495,1505,1427,1376);(1,1)
   479776	 598.4	( 481.5)	nav_state_idle (1)	 ARM|BLACKBOX	
   551776	 670.7	( 553.8)	end of log
```

Interesting assertion of `GCSNAV` when the link comes back.

HerraMakkara, 2021-03-17:
That could be the channels then, jumping GCSNAV on and off

HerraMakkara, 2021-03-18:
Had another test two with GCSNAV completely removed from the modes. Waypoint mission resets still :(

[image]

HerraMakkara, 2021-03-20:
Any idea what this could be about?

breadoven, 2021-03-21:
I couldn't see anything obvious in the code that explains why it resets. If the option to continue the mission on failsafe is to be any use obviously it shouldn't reset the mission when you regain the signal.

The Nav system briefly went to idle when the Rx signal was regained in the log above. So either a bug or still related to the Aux channels doing strange things briefly when the signal is first regained.

HerraMakkara, 2021-03-22:
I’ll reset all the settings and test again tomorrow with clean setup.

HerraMakkara, 2021-03-23:
Ok, there is the channel issue after all. Here is a video on channels when returning (reconnecting) from failsafe.
https://youtu.be/Ri4qDIkTvbk

Channel 9 has been the NAV WP switch.

HerraMakkara, 2021-03-25:
It works after moving NAV WP to another channel. TX16S and crossfire - something odd happens with those channels..

closed 2021-03-25 by HerraMakkara