iNavFlight/inav #6719: Alternate Receiver Channel mapping

billyd60, 2021-03-17:
## Current Behavior
Currently the choices of receiver mapping limit you to standards found in typical radio transmitters, such at TAER or AETR

## Desired Behavior
At times it may be convenient, depending on the model, to use an alternate non-standard channel mapping. For example, in a Rover, I would prefer to have steering on a separate stick than the throttle. Since Rover requires stabilized Yaw to control RTH and waypoints, you cannot simply assign stabilized roll to achieve this effect. So you endup with throttle and steering on one stick of the radio, for a mode 2 setup. You could also use a radio mix, but that potentially wastes a channel that could otherwise be used.

## Suggested Solution
Allow in the CLI to enter a new map for receiver, so that any of the first four servo channels can be set to any of the 4 basic radio functions of Throttle, Aileron, Elevator and Rudder.

## Who does this impact? Who is this for?
I think this would help those who use INAV for models that don't fit the typical aircraft or copter model. Particularly, Boats, and Land craft.

## Additional context
I don't believe this should be difficult to implement, however that is just my layman's understanding and perhaps it is impossible. Thanks for listening!

wx4cb, 2021-03-22:
you can have whatever channel mapping you want, just type it in instead of picking.. for example, say you wanted aux1 in the middle then you can do  TR1AE234

billyd60, 2021-03-22:
That's what I thought but it just bounced it and reset it to the default. I'll try it again.

closed 2021-03-22 by billyd60