iNavFlight/inav #6725: System Message display time option ?

breadoven, 2021-03-19:
Wondering if there's interest in an option to set the System Message cycle display time. Currently this is fixed at 1 second which seems a bit short if several messages are popping up. Option would allow up to 5s perhaps.

Jetrell, 2021-03-21:
Message cycle times can be too fast when trying to absorbing what is written. And too slow, when waiting for the next message to come around. 
It all depends on the circumstances.

Jetrell, 2021-03-21:
Message cycle times can be too fast when trying to absorb what is written. And too slow, when waiting for the next message to come around.
It all depends on the circumstances.

Maybe having them dynamically time displayed. Or in order of critical importance? I wouldn't think longer than 2second would be needed.

An example of this is the DJI Craft Name repurposing. It has up to 5 elements, with each appearing for 3 seconds.
I often spend too much time waiting for the element I missed, to cycle back around.

breadoven, 2021-03-21:
Yes dynamically changing the display time and priority order depending on the message would be the ideal solution but a lot more difficult to do than just making the current 1000ms hard coded time a setting. There was talk of a new message bus https://github.com/iNavFlight/inav/issues/2211 which could have included priority and timing per message ... but it never went anywhere.

For me 2s or slightly shorter is much easier to instinctively absorb. At least with a basic setting you have means of tweaking it as required.

MrD-RC, 2021-03-24:
Could the time not be based on the length of the message? so shorter messages are displayed for say 1 second, and longer messages for more time. It may take a little playing about with to work out the multiplier based on characters. You could also have an option to change the base 1 second. The multiplier could then be calculate on that base time.

breadoven, 2021-03-25:
I think you're right MrD-RC. It's just the longer messages that need more than the current 1s. So variable display time based on message length makes sense. And possibly with an option to factor the overall time to personal choice (although maybe avoid another option if the variable timing function works well enough in itself).

closed 2022-02-15 by DzikuVx