iNavFlight/inav #6728: Inav6.2.1. Matekf405 - Flysky fs-i6x Dont work

Gianluca111, 2021-03-21:
Hi, I have upgraded from version 5.0 to 6.2.1. Everything was working perfectly. Now after the upgrade, I am no longer getting a signal from the receiver.
It is connected on Uart2 in S-Bus mode. I have to bind it again but nothing to do. I have rechecked all connections and everything is ok.
Can you help me?

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: Eachine Tyro119 - Matekf405
- INAV version string:

breadoven, 2021-03-21:
Flysky doesn't use SBUS, only IBUS or PPM.

Better if you post a Diff output from the Configurator CLI so the settings can be checked.

Gianluca111, 2021-03-21:
Unbelievable! I checked the radio and it was set to S-Bus. So that means I was using it in S-Bus before. Now I changed the setting to I-Bus and it worked!
How is this possible? In the frame I soldered the cable into the S-Bus pad....

breadoven, 2021-03-21:
Actually I'd forgotten Flysky does offer SBUS but I never use it because IBUS seems a better option, avoids issues with inverted SBUS signals.

Gianluca111, 2021-03-22:
Ok, it's works...Thanks in advance.

closed 2021-03-22 by Gianluca111