iNavFlight/inav #6730: Prearm issue with non momentary switches

breadoven, 2021-03-22:
I was using the new Prearm feature and quickly realised it's flawed. It's fine if the Prearm is on a momentary switch but flawed if it isn't because a non momentary switch can be left in the Prearm position when the battery is connected or you've just disarmed meaning you can still arm with only a single action, removing any protection from accidental arming. Not all transmitters have momentary switches.

The solution would be to block arming until the Prearm switch had been reset off after battery connect and also after a disarm, In fact the better and safer solution would be to have a time limit between selecting Prearm and then Arm. So if you don't Arm within perhaps 5s of Prearming, Arming is blocked until you reset the Prearm switch. There's no reason to not perform both actions at the same time.

harry1453, 2021-03-24:
This is a flaw I didn't think about in my implementation. I agree with the first part of your solution, however I don't like adding a timer, I think it overcomplicates the issue, especially when prearm is designed for momentary switches.

I think that there should be a flag that says that the prearm has been reset (i.e. is not active). This flag would be false on startup, and set to true whenever the prearm is not set. After disarming, this flag should be set to false. Then this flag should be required to be true in order to arm.

breadoven, 2021-03-24:
I did consider your solution but decided it doesn't work effectively because it still allows the Prearm switch to be set and inadvertently left in the Prearmed state without noticing, allowing accidental arming. The whole point of Prearming is it forces you to perform 2 actions "at the same time" to Arm. The timer forces you to do this because if you don't do both actions within the time window then you are forced to reset the Prearm switch when you do try and Arm intentionally, i.e. perform 2 actions at the same time.

harry1453, 2021-03-24:
Although I don't have the same situation as you, I can hear where you're coming from. Neither solution is perfect, so I believe the best solution is to add options for the end user. I'll work on this and update my PR.

harry1453, 2021-03-24:
@breadoven Please see my updated PR - I have fixed the reactivation issue even if you arm and disarm really quickly, and I have added an optional, configurable timeout.

breadoven, 2021-03-25:
Thanks for your efforts but can't help but think this is overcomplicating something simple. The basic timer idea has no adverse affects when using a momentary switch other than to force you to Arm within the time window which surely you're going to do anyway if you intend to Arm. And not using the timer with a non momentary switch effectively makes the Prearm feature an unreliable lottery giving a false sense of security which defeats the whole point of using it in the first place. It's hard to see why anyone would want to risk injury just to avoid the inconvenience of having to toggle the Prearm switch because they exceeded the timer limit.

harry1453, 2021-03-25:
Okay, let me put it a different way: The issue where you can instantly rearm without resetting has to be fixed. Otherwise you could disarm and rearm without prearm reset within the timer window and that 1) is jank 2) defeats the purpose of prearm.

Now, ignoring whether you want a timer or not, you can't hardcode a timer value, that isn't really a good way of doing things. So it really ought to be configurable. If it's gonna be configurable, you may as well allow users to set it to 0 to disable it, like how a lot of other configuration parameters in INAV work. I don't think adding the **option** to have a timer complicates the matter any more than the alternative of forcing the user to have a hardcoded timer.

> And not using the timer with a non momentary switch effectively makes the Prearm feature an unreliable lottery giving a false sense of security which defeats the whole point of using it in the first place

I agree, so you should use the timer when using a non-momentary switch. In my PR the timer defaults to enabled (10 seconds). I just want to be able to disable it if I know I'm using a momentary switch so won't need it.

> It's hard to see why anyone would want to risk injury

But with a momentary switch, there is no additional risk of injury, because your finger is always holding down the button. A timer only serves to frustrate the user in some situations. So, if you are using a momentary switch, you should be able to disable it.

You say it has no adverse effects apart from **the adverse effect you specified**. Sometimes I'll be waiting to go and will prearm long before I hit arm, knowing that I can always let go if I know I'm not going to arm. Perhaps I'm waiting for someone to tell me they're ready to film something. Either way, I don't want a timer. It will just be a source of frustration.

You, however, do want a timer. So we're at an impasse. We could either argue on GitHub all day, **or**, we could compromise (which is what I'm trying to do) and add an option that gets us both what we want, and gives end users more choice. Which is what I've proposed, and you've called "overcomplicating".

PR #6737 (2021-04-05): Add "prearm was reset" flag to prevent rearming without resetting prearm & add prearm timeout
Fixes #6730. Alternative to #6731.

This fixes an oversight in my implementation of Prearm: Once prearm is held, you can toggle the arm switch on and off all day. So if a non-momentary prearm is used and the quad is disarmed without resetting the prearm, it can be armed again with the single arm switch, which defeats the prearm.

This PR adds a simple flag to track whether the prearm has been reset (not engaged / RC mode not active) since the last disarm.

This **also** adds an optional setting called [... cut]
closed 2021-04-05 by DzikuVx