iNavFlight/inav #6732: SmartAudio not working with TBS Unify Pro32 HV

poi2poi, 2021-03-22:
## Current Behavior
Actually it's not possible to set VTX output power of **TBS Unify Pro32 HV** through SmartAudio because the Decibel values are set wrong, see source code file "inav-master\src\main\io\vtx_smartaudio.c" Line 120
_(I have files for 2.6.0 but shouldn't have changed in 2.6.1)_

VTX will only change power values if the received values (14, 20, 26 or 36 for max power) are sent by FC

Actual standard values are     7, 16,  25  or  40

To get more informations about this topic and why it's important, please check Joshua Bardwell´s video:
[https://www.youtube.com/watch?v=thR2XA_0PLM]

- and TBS vtx tables for betaflight -
[https://www.team-blacksheep.com/vtx-table-for-betaflight.txt]

## Expected behavior

For the described VTX the correct values for power are:
```
powerlevels 4
powervalues 14 20 26 36
powerlabels 25 100 400 1W+
```

## Suggested solution(s)

Either: Enable to optionally set vtx-tables through CLI like in Betaflight (easiest to maintain)

Or: Let user choose from most common VTX available with correct values hard coded

At least mention the issue in the docs as soon it's not fixed, i.e. here:
[https://inavfixedwinggroup.com/shopping-list/vtx-and-cameras/tbs-unify-pro32-hv/]

## Additional context
Mention somewhere in the iNav-configurator that SmartAudio first has to be activated in VTX config through TBS Agent X, because as a standard it's set to CSFR

wx4cb, 2021-03-22:
strange, it works fine for me. that's the only vtx i use. 

now that said, i havent checked the actual physical power output, but i know for a fact that it changes to at least level 4 via smart audio

Pairan, 2021-03-23:
I have the same problem- turned out, that I ran the BETA Firmware which perhaps expect these mentioned values! 
After having flashed the old Firmware (not BETA) to the VTX it worked again.

Maybe this helps

poi2poi, 2021-04-06:
> 
> 
> I have the same problem- turned out, that I ran the BETA Firmware which perhaps expect these mentioned values!
> After having flashed the old Firmware (not BETA) to the VTX it worked again.
> 
> Maybe this helps and maybe this is the same thing here

Pairan, I compiled a custom build for this issue, tell what board you're using if you want the fix.

Greetings

Pairan, 2021-04-06:
Oh ... okay ... sounds interesting. I'm running a Matek F411WSE board

poi2poi, 2021-04-06:
It's weird, actually looks like iNav is trying to get the available power values from the "SmartAudio device" but I have no idea how to use the debug features.

I compiled for your FC if you want to try it, works for me now

[inav_2.6.1_MATEKF411SE.hex.zip](https://github.com/iNavFlight/inav/files/6264746/inav_2.6.1_MATEKF411SE.hex.zip)

Pairan, 2021-04-13:
flashed it today ... will give it a try at the next occasion and report back in

Pairan, 2021-04-15:
Flew yesterday after having flashed the UnifyPro32 to BETA 1.15 !

The power went from 1 (as set UNTIL FIRST ARM) to 4 after launch - this looks good so far. will test switching in the next flights

wx4cb, 2021-04-15:
> Flew yesterday after having flashed the UnifyPro32 to BETA 1.15 !
> 
> The power went from 1 (as set UNTIL FIRST ARM) to 4 after launch - this looks good so far. will test switching in the next flights

Yes but does it go to 5? mine's always gone to 4

charlie

Pairan, 2021-04-17:
Nope ... no chance to get it up at 5!

nwpulver, 2021-05-24:
> Pairan, I compiled a custom build for this issue, tell what board you're using if you want the fix.
>
> Greetings

Hello would it be possible to get a custom build for this issue for this board 
[inav-iflight-succex-e-v2.1.zip](https://github.com/iNavFlight/inav/files/6535303/inav-iflight-succex-e-v2.1.zip)