iNavFlight/inav #6736: Support BetaFPV SMO 4k Cam remote control

Skipp1609, 2021-03-23:
## Current Behavior
There is no possibility to remote control the BetaFPV SMO Cam by flight controller connection

## Desired Behavior
Power and recording should be controlled by flight controller connection
## Suggested Solution
needs a configuration just like in BetaFlight (see attached file)

## Who does this impact? Who is this for?
possibilty to operate the Cam by remote control

## Additional context
none
[image]

Papsi77, 2021-03-23:
+1
That would be great 😉

konraddi, 2021-03-23:
+2
Absolutely needed! :)

lightpixel73, 2021-04-12:
+3
would be great.. also requested in the facebook Inac fixed wing group...

SilentZen, 2021-04-13:
+4 was hoping to have it enable on arm then realised inav didn't support it.

mattcave, 2021-05-28:
My reading of the docs suggests that the camera just needs the voltage to go high then low on the remote recording pin for the camera to start recording. If that's the case, some of the fixed wing oriented FCs from Matek already have a simple switch built into them that can be controlled by a spare channel. Would that work?

closed 2022-01-28 by DzikuVx