iNavFlight/inav #6739: Bing Maps in Mission planner

Marcus-Barcus, 2021-03-25:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:  2.6.1

This is to do with just Mission Planner
When setting up Bing Maps i cant copy and past the Key from Chrome or Notepad or 
Excel or key pad ctrl-c ctrl-v
nothing happens 
tried turning off java script still nothing as key is gone - needs Java
Is there something in the code stopping this.
No info what Proxys to employ for URL & Layer
Thanks

harry1453, 2021-03-25:
This is the issue tracker for the INAV firmware, not any GCS. Which "Mission Planner" do you mean? Ardupilot's Mission Planner?INAV Configurator? You need to find the issue tracker for the software you are using and open an issue there.

Marcus-Barcus, 2021-03-25:
Im only new to iNav - GCS? Im referring to inav  configurator that Mission Planner is part of.
If you click the wheel at top right corner it gives you application options and saw a Youtube 
on using a different Satellite map option in Mission Planner ie Bing maps 
It needs a key to configure which i got but is 35 characters long and has odd characters i 
cant produce. I cant copy and paste as its rejecting it but may try 2.6 to see if its working on 
that iNav? Bing satellite view looks way better than a Map.

avsaase, 2021-03-25:
@Marcus-Barcus I think the confusion is because you are talking about the INAV Configurator (correct?), which is not a Ground Control Station software. Mission Planner _is_ a GCS used with competing flight controller firmware ArduPilot. For problems with the INAV Configurator you should open an issue [here](https://github.com/iNavFlight/inav-configurator/issues).

Marcus-Barcus, 2021-03-25:
My apologies its Mission Control in INav not any third party software - Cheers

closed 2021-03-26 by DzikuVx