iNavFlight/inav #6746: INAV doesn´t find my FC and can´t connect to it

Urromir, 2021-03-25:
I tried to connect my new Matek F411 WSE to INAV and after a few problems i finally could flash it, but after that i wasn´t able to connect. I can only choose COM3, COM4 or manual selection and when i choose COM3 or 4 it fails to open serial port. At manual selection i need to manually write the port (which I don´t know). In every youtube video I watched about INAV they just plugged  the FC in and were able to select it. 
When i want to flash it, i can press the button on the FC and then INAV automatically switches to DFU.
Would be very nice if someone has a solution for this problem.
Btw I use the lates INAV version on a Windows PC

avsaase, 2021-03-25:
Probably a driver issue. Try the [ImpulseRC Driver Fixer](https://impulserc.com/pages/downloads).

Urromir, 2021-03-25:
okay, I did it about 5 times before you told me to do it. Now I did it again and now it works :D
Thanks a lot <3

closed 2021-03-25 by Urromir