iNavFlight/inav #6749: NAV POSHOLD - unstable attitude and altitude

MaxLansing, 2021-03-26:
Hi there,

i got multiple quads with iNav, but the latest one just acts up.
System:
Matek F405STD
BN880
inav 2.6

i just added the GPS and it's not holding POS/ALT. it flies just fine and everything else is great, it just doesn't want to steadily hold POS/ALT.
BN880 is mounted on an outrigger and should be setup correctly (270CW Flip) and calibrated.
the other quad with a similar setup on inav 2.5 just works fine.
I am at a loss here.
whenever i calibrate the compass the values differ by a lot, while doing it on the same spot and everything.
if anyone has suggestions, i would be glad.
i exchanged the BN880, same problem
any help debugging this is much appreciated!
thx

diffall:
 Mode, type 'exit' to return, or 'help'

# diff all

# version
# INAV/MATEKF405 2.6.0 Dec  4 2020 / 11:55:31 (cada68e4)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# safehome

# logic
logic 0 1 -1 4 1 8 0 0 0
logic 1 1 -1 5 1 8 0 0 0
logic 2 1 -1 6 1 8 0 0 0
logic 3 1 0 25 0 1 0 0 0
logic 4 1 1 25 0 2 0 0 0
logic 5 1 2 25 0 4 0 0 0

# gvar

# feature
feature -TX_PROF_SEL
feature SOFTSERIAL
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 2 2 115200 115200 0 115200
serial 3 4096 115200 115200 0 115200
serial 31 32 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1900 2100
aux 1 1 2 1375 1600
aux 2 11 2 1950 2100
aux 3 10 7 1925 2100
aux 4 30 1 1900 2100
aux 5 13 6 1325 1650
aux 6 38 4 1950 2100
aux 7 27 4 1425 1600
aux 8 42 5 1075 1300
aux 9 43 5 1300 1700
aux 10 44 5 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 21 0 V
osd_layout 0 1 2 6 V
osd_layout 0 2 0 0 V
osd_layout 0 6 11 13 H
osd_layout 0 7 25 0 V
osd_layout 0 9 24 13 V
osd_layout 0 11 12 10 V
osd_layout 0 12 17 15 V
osd_layout 0 15 25 7 V
osd_layout 0 25 23 5 V
osd_layout 0 28 23 15 V
osd_layout 0 30 2 11 V
osd_layout 0 32 2 8 V
osd_layout 0 38 13 15 V
osd_layout 0 87 1 15 V
osd_layout 0 105 1 0 V
osd_layout 0 107 7 15 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 110
set gyro_lpf_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 120
set acc_hardware = MPU6500
set acczero_x = -10
set acczero_y = -9
set acczero_z = 28
set accgain_x = 4093
set accgain_y = 4085
set accgain_z = 4095
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 29
set magzero_y = -70
set magzero_z = -33
set maggain_x = 571
set maggain_y = 740
set maggain_z = 580
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 16
set serialrx_inverted = ON
set blackbox_rate_denom = 2
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set throttle_idle =  3.000
set failsafe_delay = 10
set failsafe_throttle = 1200
set failsafe_procedure = RTH
set align_board_roll = 4
set align_board_pitch = -6
set vbat_scale = 1101
set current_meter_scale = 210
set model_preview_type = 3
set applied_defaults = 2
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set mc_airmode_type = THROTTLE_THRESHOLD
set nav_use_midthr_for_althold = ON
set nav_auto_speed = 400
set nav_manual_speed = 350
set nav_rth_allow_landing = FS_ONLY
set nav_mc_bank_angle = 35
set osd_alt_alarm = 145
set osd_crsf_lq_format = TYPE2
set osd_right_sidebar_scroll = ALTITUDE
set osd_sidebar_scroll_arrows = ON
set i2c_speed = 800KHZ
set name = ACQ-Predator
set vtx_band = 2
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM

# profile
profile 1

set mc_p_pitch = 44
set mc_i_pitch = 75
set mc_d_pitch = 25
set mc_i_roll = 60
set mc_p_yaw = 35
set mc_i_yaw = 80
set mc_d_yaw = 5
set mc_p_level = 35
set max_angle_inclination_rll = 450
set max_angle_inclination_pit = 450
set dterm_lpf_hz = 110
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 170
set dterm_lpf2_type = PT1
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set setpoint_kalman_enabled = ON
set setpoint_kalman_q = 200
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_yaw_expo = 60
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 75

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

set battery_capacity = 1800
set bat

avsaase, 2021-03-26:
blackbox log and/or dvr?

MaxLansing, 2021-03-27:
https://youtu.be/DG5FsuPsAlk
[LOG00020.TXT](https://github.com/iNavFlight/inav/files/6215404/LOG00020.TXT)

thx alexander, 
i will make a clean video and log later in the day. i wasn't expecting problems yesterday.
i will let it do it's thing longer in POSHOLD without touching the sticks.

MaxLansing, 2021-03-27:
https://youtu.be/gn8i9RMquMQ
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/6215963/LOG00001.TXT)

It is toiletbowling badly. and i have no idea why, the only thing i know causes this, was if the mag isn't setup correctly.

MaxLansing, 2021-03-28:
After extensive testing in the field and at the bench, for anyone experiencing similar problems, here is what screwed it up for me:
The cables for GPS ran to close to the base of the VTX Antenna mount and screwed up the SDC SDA lines for the compass.
Rookie mistake but it happens.

closed 2021-03-28 by MaxLansing