iNavFlight/inav #6754: Programming RC_CHANNEL_OVERRIDE not working in 2.7

rts18, 2021-03-28:
I cannot get `RC_Channel_Override `to work with the 2.7 development firmware.
The logic condition outputs as active in the Programming tab. But the output channel does not get overridden. Either on the device or the receiver tab.

I have tried multiple channels, values and logic states. I am reasonably sure I haven't overlooked anything.

@DzikuVx  Can I ask that you take a look?
Thanks

DzikuVx, 2021-03-31:
@rts18 it works like expected

[image]

[image]

closed 2021-03-31 by DzikuVx

rts18, 2021-03-31:
@DzikuVx Thank you for testing it out.

This is why I couldn't get it working.

[image]

Please don't take this as criticism. 
As an average Joe, I assumed  that operand 'A' would be defined as an `RC Channel` and not a `Value`, because it is that channel that is being overridden.
The confusing part, is the condition can also be met by using operand "A' `RC channel`. But it doesn't output unless operand 'A' is defined as a `Value`.

Is it possible not to have the condition met by two operand commands. So us simple people can work it out easier?

Thanks

DzikuVx, 2021-04-01:
Operand RC channel is a getter. If it would be used, you would not be able
to set rc channel based on other value.

OptimusTi, 2021-05-22:
@DzikuVx Any chance to change operand name of "RC Channel" to "Get RC Cannel"? I lost a few hours trying to figure out the same thing. Now that I know that the operand s a getter, it makes sense, but then, the docs should say that operand RC Channel “gets” instead of “points to”.

Another suggestion is to split the operand in two?

"Get RC Channel" is the getter. 
"RC Channel" sets the value.

wx4cb, 2021-07-11:
maybe rename it to RC Channel Value ? or some derivative of that to make it obvious ?