iNavFlight/inav #6762: Temperature sensor from DPS310 barometer not recognized

nickoe, 2021-03-30:
## Current Behavior

I got a new Mateksys F405-WSE flight controller, it contains a DPS310 barometer.

I have read this doc: https://github.com/iNavFlight/inav/blob/master/docs/Temperature%20sensors.md

It mentions that:
> When a new temperature sensor is connected it is automatically detected and will appear in the output of the temp_sensor command. 

But I don't see any temperature sensors, not even with the `temp_sensors` command in the CLI:

```
# version
# INAV/MATEKF405SE 2.6.1 Feb 25 2021 / 10:39:37 (64249b1a)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# temp_sensor
temp_sensor 0 0 0 0 0 0 
temp_sensor 1 0 0 0 0 0 
temp_sensor 2 0 0 0 0 0 
temp_sensor 3 0 0 0 0 0 
temp_sensor 4 0 0 0 0 0 
temp_sensor 5 0 0 0 0 0 
temp_sensor 6 0 0 0 0 0 
temp_sensor 7 0 0 0 0 0 
```

## Steps to Reproduce

1. Configure your FC for your plane, enable telemetry etc. (probably not needed)
2. Enable the DPS310 
3. Observe that the temperature values are all -100  in the "Sensors" pane. 

## Expected behavior

I would expect the temperature reading from the DPS310 to show up with a real value.

As far as I understand the support was added in 2.6.0, see mentions of DPS310 in  https://github.com/iNavFlight/inav/releases/tag/2.6.0

I enabled it as a baromter in and it appears to work as a barometer in the sensors readings as well, the setting looks like:
[image]

## Additional context

`dump`: https://pastebin.com/raw/FE0X5kev

---

- FC Board name and vendor: [F405-WSE](http://www.mateksys.com/?portfolio=f405-wse)
- INAV version string: `# INAV/MATEKF405SE 2.6.1 Feb 25 2021 / 10:39:37 (64249b1a)`

DzikuVx, 2021-03-31:
In this context, DPS310  is used as a barometer, not temperature sensor

nickoe, 2021-03-31:
@rts18 @DzikuVx I don't have FPV gear and did not look there, I see that option now.

I tried to enable it ,but I don't see any new field when I try to discover new telemetry sensors on my remote. So I can't use it as a temperature source for telemetry?

avsaase, 2021-03-31:
@nickoe AFAIK telemetry does not include any temperature data.

nickoe, 2021-03-31:
Ok, I will close thus issue then. Thank you all!

closed 2021-03-31 by nickoe