iNavFlight/inav #6764: Cruise mode not controlling throttle

Thieu-s, 2021-03-30:
Updated from 2.2.1 to 2.6.0.
Now on fixed wing the CRUISE mode is not controlling throttle, so no altitude hold.
When i enable Alt Hold and Cruise mode, the OSD shows just "AH", not "3CRS".
Something changed here?

Jetrell, 2021-03-30:
It would be useful to see your `Diff All `settings.

A good place to start looking would be `nav_overrides_motor_stop`
It was change in 2.6. It now allows for disabling of the throttle in navigation modes.

Thieu-s, 2021-03-31:
Thanks! There's a lead!
My diff all below.

`diff all

# version
# INAV/MATEKF722SE 2.6.0 Dec  4 2020 / 12:21:50 (cada68e4)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 3 950 2000 1457 -100
servo 4 950 1900 1433 100

# safehome
safehome 0 1 521939600 47010000

# logic

# gvar

# feature
feature -TX_PROF_SEL
feature -BLACKBOX
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ON_USB

# map

# serial
serial 2 2 115200 115200 0 115200
serial 3 2048 115200 115200 0 115200
serial 5 256 115200 115200 57600 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100
aux 1 1 1 1300 1700
aux 2 3 2 1300 1700
aux 3 10 0 1700 2100
aux 4 45 2 1700 2100
aux 5 12 1 1700 1900
aux 6 13 2 1050 1300
aux 7 27 1 1900 2100
aux 8 48 3 1700 2100
aux 9 42 3 1300 1700
aux 10 43 3 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 7 2 H
osd_layout 0 1 12 0 H
osd_layout 0 4 8 6 H
osd_layout 0 5 23 11 H
osd_layout 0 6 23 12 V
osd_layout 0 7 1 11 V
osd_layout 0 8 0 4 H
osd_layout 0 9 24 1 H
osd_layout 0 10 10 11 H
osd_layout 0 11 1 12 V
osd_layout 0 12 16 12 V
osd_layout 0 13 4 6 V
osd_layout 0 14 1 1 V
osd_layout 0 15 22 6 V
osd_layout 0 19 4 7 H
osd_layout 0 20 20 0 V
osd_layout 0 21 0 0 V
osd_layout 0 22 14 2 V
osd_layout 0 23 24 2 V
osd_layout 0 24 12 0 V
osd_layout 0 25 28 4 H
osd_layout 0 28 23 11 H
osd_layout 0 30 1 10 V
osd_layout 0 32 8 12 V
osd_layout 0 33 23 1 V
osd_layout 0 35 24 11 V
osd_layout 0 40 24 3 V
osd_layout 0 105 4 4 V
osd_layout 0 108 12 1 H
osd_layout 0 109 7 3 H
osd_layout 0 110 0 2 V
osd_layout 0 112 1 3 V
osd_layout 1 12 24 12 V
osd_layout 1 15 25 0 V
osd_layout 1 30 1 10 V
osd_layout 1 32 0 12 V
osd_layout 1 45 0 0 V
osd_layout 1 98 0 0 V
osd_layout 1 99 15 0 V
osd_layout 2 12 24 12 V
osd_layout 2 15 25 0 V
osd_layout 2 30 1 10 V
osd_layout 2 32 0 12 V
osd_layout 2 99 15 0 H

# master
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 40
set acc_hardware = MPU6000
set acczero_x = 14
set acczero_y = -10
set acczero_z = 164
set accgain_x = 4082
set accgain_y = 4067
set accgain_z = 4030
set mag_hardware = NONE
set mag_declination = 120
set baro_hardware = NONE
set pitot_hardware = NONE
set min_check = 1015
set max_check = 1995
set serialrx_provider = CRSF
set max_throttle = 1950
set min_command = 980
set motor_pwm_protocol = STANDARD
set throttle_idle =  7.000
set failsafe_procedure = RTH
set failsafe_mission = OFF
set align_board_roll = 5
set align_board_pitch = 30
set current_meter_scale = 250
set platform_type = AIRPLANE
set model_preview_type = 8
set small_angle = 180
set applied_defaults = 3
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_min_sats = 5
set deadband = 1
set yaw_deadband = 1
set nav_extra_arming_safety = ALLOW_BYPASS
set nav_wp_radius = 3000
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 7000
set nav_rth_home_altitude = 5000
set nav_fw_climb_angle = 25
set nav_fw_pitch2thr = 4
set nav_fw_loiter_radius = 3000
set nav_fw_control_smoothness = 2
set nav_fw_launch_max_angle = 90
set nav_fw_launch_thr = 2000
set nav_fw_launch_idle_thr = 1300
set nav_fw_launch_motor_delay = 0
set nav_fw_launch_timeout = 8000
set nav_fw_launch_climb_angle = 20
set nav_fw_allow_manual_thr_increase = ON
set osd_video_system = NTSC
set osd_time_alarm = 0
set osd_alt_alarm = 0
set osd_dist_alarm = 0
set osd_link_quality_alarm = 60
set osd_coordinate_digits = 8
set name = ARPRO
set vtx_band = 5
set vtx_channel = 5

# profile
profile 1

set fw_p_pitch = 20
set fw_i_pitch = 40
set fw_ff_pitch = 90
set fw_p_roll = 30
set fw_i_roll = 45
set fw_ff_roll = 45
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 30
set pitch_rate = 15
set yaw_rate = 9

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 320

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

# `

Thieu-s, 2021-03-31:
Thanks! There's a lead!
My diff all below.

[INAV_cli_ARPRO_20210331_084130.txt](https://github.com/iNavFlight/inav/files/6234446/INAV_cli_ARPRO_20210331_084130.txt)

Thieu-s, 2021-03-31:
Thanks! But I think that's set up properly.
My diff all below.
.
[INAV_cli_ARPRO_20210331_093945.txt](https://github.com/iNavFlight/inav/files/6234674/INAV_cli_ARPRO_20210331_093945.txt)

Thieu-s, 2021-03-31:
Thanks! It's set to ALL_NAV". I think that's correct.
My diff all below.

[INAV_cli_ARPRO_20210331_093945.txt](https://github.com/iNavFlight/inav/files/6234702/INAV_cli_ARPRO_20210331_093945.txt)

Jetrell, 2021-03-31:
There is something in the CLI about `ALL_NAV ` needing to be combine with `Motor_stop` to work.
I have mine set to `AUTO_ONLY`.
The only time I have seen `AH` appear on the OSD, in place of `3CRS`. Is when I don't have a satellite lock.

If you haven't. It may be  worth looking at` nav_overrides_motor_stop` explanation in detail.  
 https://github.com/iNavFlight/inav/blob/master/docs/Settings.md

Thieu-s, 2021-03-31:
Oh, never mind... i had the modes on either AH or CRUISE.  When I adjusted this so that both modes can be enabled simultaneously, I only did a bench test, without GPS lock...
Thanks for pointing to this!

closed 2021-04-03 by avsaase