iNavFlight/inav #6769: Matek F722MiniSE / Crossfire / Inav 2.6.1 Bug

JDFPV, 2021-03-31:
Matek F722 Mini SE
Nano Crossfire RX
Inav 2.6.1

My current dump
https://docs.google.com/document/d/1JhT0OswJOS42WvOQktOsSRxWAWvbI2oXOucWD3e1KK0/edit?usp=sharing

Currently spoken to a Matek Service Technician and Inav Developer. Both parties agreed equipment and dump files and senor readings was fine. 

Issue: Random and strange telemetry readings on altitude and amps when CRSF is used. Can be seen on the OSD and Inav lua script on tx16s radio. When I switched to SBUS the telemetry readings in osd are normal. When I flew the quadcopter using crsf or sbus. Positon Hold mode is working correctly. Have not tried RTH. 

Video showing the random numbers seen in telemetry using crsf vs sbus.
https://youtu.be/VGNJ_DliHXs

I know you guys are really busy. And I myself and others really appreciate what you do for free. I do donate when I can. Can you please take a look at this and see what a solution can be other than not using crsf 😉. I know a few other people that have this excat same issue using the same components. Thanks

OptimusTi, 2021-03-31:
Quick comment on the video:

You must properly setup the crossfire RX to SBUS if you change the serial provider in INAV to SBUS.

(Note: Using crossfire with other boards but not having the same issue)

[image]

JDFPV, 2021-03-31:
And this video is what the osd altitude does when using crsf during flight. 

https://youtu.be/dh4Aqr12_YI

OptimusTi, 2021-03-31:
Could you please setup a telemetry screen on the radio showing the altitude sensor and compare if you are getting the same numbers? Which crsf firmware are you on?

Something like this:
[image]

JDFPV, 2021-03-31:
Here's a video showing the telemetry screen crsf vs sbus

https://youtu.be/kMYrSfGjFCc

OptimusTi, 2021-03-31:
You may have done this already but try deleting all sensors and connect a battery, wait for gps lock, and rediscover them. Mine look that they are updating at a different rate.

In the inav OSD, altitude should come from the FC board and not from crsf data.

JDFPV, 2021-03-31:
Yes I have deleted sensors and discovered them again..I only get 23 sensors. And that's on all of my aircraft using different types of boards. My osd shows a random number while flying.. -34 to 100 and I'm flying 5 ft off of the ground. When I switched rx provider to sbus every works perfectly fine.

OptimusTi, 2021-04-01:
Not sure what's going on. I'm on 4.11 and OpenTX 2.3.11 and getting 24 sensors.

JDFPV, 2021-04-01:
I'm using whatever came with the tx16s. Waiting for the touch screen to be enable before I update my radio.

rts18, 2021-04-01:
@JDFPV I'm using the Matek F722miniSE and CRSF V3.78.  I'm not having those problems.
For what its worth, mine only discovers 22 sensors.

I don't know how voltage dependent the CRSF UART protocol is, nor how rugged the SE's reverse voltage protection diodes on the 4.5v USB pins are. They may be causing a larger volt drop under load. 
I'm running my crossfire RX from the 5v supply. Not the 4.5v USB supplied pins.

JDFPV, 2021-04-01:
Resolved. Haven't test flew it yet. But its looking good in position hold mold. 

https://youtu.be/I8GgRdEMRLo

JDFPV, 2021-04-04:
Got a chance to fly it today. And the osd reading is still all over the place. I don't know what else to try... But the osd altitude reading is bad off and now position hold mode is not working. In position hold mode the quad climbs up or falls out of the sky despite the throttle being centered. Previously before I disabled mavlink on uart 4 position hold worked great. So there is most definitely some type of bug in inav.

avsaase, 2021-04-04:
@JDFPV Can you share a blackbox log? Several sensors contribute to the altitude estimation so to troubleshoot we need to see what the individual sensors report.

JDFPV, 2021-04-04:
Sure. If you can give me directions to get the Blackbox info off and send it... I will be more than happy to share this info.

avsaase, 2021-04-04:
The Matek F722Mini SE has onboard storage for blackbox logs. You need to have it enabled on the blackbox tab in the configurator. After the flight, or now if you already had it enabled, go to the blackbox tab and click safe to file.

JDFPV, 2021-04-05:
Here's the link to Blackbox info. Thanks @avsaase 

https://drive.google.com/file/d/1K1TikgBbfowOMRMtHzy6vlcaLxpLkbrt/view?usp=drivesdk

avsaase, 2021-04-05:
> And this video is what the osd altitude does when using crsf during flight.
> 
> https://youtu.be/dh4Aqr12_YI

The log from this video is not included, correct? In either case, I think you have a problem with vibrations, like so many with 4-inch quads, myself included. Your baro readings also jump up and down more than they should. Do you have some open-cell foam over the baro?

JDFPV, 2021-04-05:
No log for that flight. And yes open cell foam is over the Barometer.

closed 2021-08-23 by DzikuVx