iNavFlight/inav #6774: Magnetometer is not working in Diatone Mamba Basic F722 Mini MK3

Ram-85, 2021-04-02:
Hi,

I am using this board for my Micro LR build with the Matek M8Q-5883 GPS/Compass. Everything seems to work except for the Magnetometer (red). In Diatone's website they have the following command to set the Magnetometer, but they dont seem to work with iNav:
set mag_i2c_device = 1
set mag_bustype = i2c

Is there an alternative to these commands in iNav?

Here is the wiring diagram for your reference:

https://cdn.shopify.com/s/files/1/0027/2708/4144/files/4-MAMBA_Basic_F722_MINI_MK3.pdf?v=1610714258
[https://cdn.shopify.com/s/files/1/0027/2708/4144/files/4-MAMBA_Basic_F722_MINI_MK3.pdf?v=1610714258](url)

Thanks

robertha64, 2021-04-14:
MK3 includes new dedicated pins for I2C SCL (PB8) and SDA (PB9) where the Compass is connected (not UART3). It will not work until target file for MAMBA722 will be modified to include this HW version.

Ram-85, 2021-04-18:
Thanks for explaining that, it was driving me crazy!

hobride, 2021-05-05:
Hi there,
I also bought a Mamba Basic F722MINI MK3. Notice MINI.
But I can't locate any "I2C" or "Mag" Pads. Can you show me the Pads on a picture?

DzikuVx, 2021-05-06:
This board will be supported i the next INAV release

Ram-85, 2021-05-06:
> Hi there,
> I also bought a Mamba Basic F722MINI MK3. Notice MINI.
> But I can't locate any "I2C" or "Mag" Pads. Can you show me the Pads on a picture?

@hobride the I2C pads are the SCL and SDA pads on the board just above the GPS UART, I circled them in this picture
[image]

closed 2021-06-10 by DzikuVx