iNavFlight/inav #6789: Ignore all safety checks for arming within 2 seconds of disarm

avsaase, 2021-04-04:
When the pilot accidentally disarms in flight, the procedure to arm and resume normal flight is quite involved and easy to mess up when the pilot panics after suddenly seeing the flight stats: 

1. (Disable nav mode)
2. Throttle to zero
3. Center roll/pitch stick
4. Arm
5. (Wiggle roll/pitch stick to disable launch mode)

If you don't have `small_angle = 180` or no GPS fix you are almost certainly screwed. 

There's all kind of ways to mitigate this problem, but an easy solution suggested on FB is to disable all the safety checks in the first few seconds after disarm so you can always rearm. 2 seconds seem to be a reasonable time for this because the first instinct will be to flip the arm switch again, and 2 seconds is sufficiently short to be a major safety hazard.

JulioCesarMatias, 2021-04-05:
Checking the altitude seems more viable, for example, if the altitude is a few meters higher than the ground altitude of the `baroGroundAltitude` calibration, use a configurable standard altitude `Alt_Desarm_Ignore_Checks = 3 meters`, then it would be `baroGroundAltitude + Alt_Desarm_Ignore_Checks`, if the altitude given by `baro .BaroAlt` is greater than this sum, all arming security checks will be ignored within 2 seconds after disarming will be ignored.

Jetrell, 2021-04-05:
@avsaase I've never had `nav_extra_arming_safety` turned ON. That overcame the rearming issue in flight for me. 
But I think your proposal seems to be a good way of using arming safety and preventing the hassles of rearming if the pilot gets in a fluster.

@JulioCesarMatias There are many that fly fixed wings below launch altitude. Wouldn't your idea prevent them from using this check ignore?

avsaase, 2021-04-05:
`nav_extra_arming_safety` only overrides the GPS fix and nav mode check, not any of the other checks like stick position. I think a two-second period after disarming where everything is ignored is probably safe enough. I agree that altitude should not be used for this, and it is also not necessary.

MrD-RC, 2021-04-06:
Personally, I don’t think turning nav_extra_arming_safety off so should be recommended at all. This allows launching without a home position recorded and can cause fly-aways. 

I think the best solution so far is to use the programming system to try and detect when you’re flying. If you’re flying, disable the extra safety features then. 

However, the 2 second window would be better all round. Functionality wise to the pilot l, it would work the same (actually better if it also disables launch). But this would be built in to the firmware, so much easier to use for everyone. It also means that all the safety features like nav_extra_arming_safety and disarm_kill_switch can be left on (or allow_bypass for bench testing with the former).

Just on suggestion. It may also be handy to add the mechanism that decides if the model is flying to the programming. I can see that being a useful feature. Just a simple is_flying boolean would be great.

Jetrell, 2021-04-06:
@MrD-RC You could add this to the long list of features that would benefit from `Flying/not flying `detection.
Whoever reliably implements that feature is going to be THE MAN!

MrD-RC, 2021-04-08:
I think it would only ever be an estimation, but it could be accurate enough for most purposes I think. Maybe this would be a good place to hash out some ideas?

### In flight triggers
All would be after arming. Once a threshold is met, it would stay _in-flight_ until the _no longer flying_ trigger. 
- **3D speed > 400 cm/s (14.5 kmh)** This speed is half way between average human walking and running speeds. This should mean that walking with the model would not trigger, but running would. This allows the failsafe test to still work.
- **Alt <> 1000 (10m)** This should cover aircraft like multirotors (that lift off slowly) and sloping gliders (which can have a slower ground speed, depending on the wind conditions).
- **Launch mode completes** This should be a pretty obvious one. 

### No longer flying triggers
- **Disarmed > 2 seconds**
- **Nominal movement detection** We would need some kind of detection for detecting when if the model has crashed. This could of course be hung in a tree with some movement. Another option could be to trigger _no longer flying_ in failsafe, then rely on a _flying trigger_ once it has recovered. Though that gives a false state, which I’m not overly keen on

They’re just a couple of thoughts. It would be great to get your input guys.

avsaase, 2021-04-08:
@MrD-RC These are some good ideas and I like to discuss them further. But in the interest of keeping things organized, can you copy or move this comment to https://github.com/iNavFlight/inav/issues/2331?

FPVZaphod, 2021-05-24:
Great idea, just wondering if 2 seconds is enough. Would be great to be able to choose the time span, because I use a long press of a button to arm/disarm.  5secs would be ok fine for me.

Mateyhv, 2021-06-10:
My vote goes for the simple 2 seconds solution, maybe user-configurable up to 5 seconds as FPVZaphod suggested to accomodate for personal needs. 

In a plane midair disarm it took me about 15 seconds to realize whats going on with not rearming and to reorganize all switches/modes and take care of the glide path to avoid obstacles. On a multirotor that would have been a disaster. This a welcome solution!

Mateyhv, 2021-06-10:
One caveat for fixed wings though. On rearming should be useful to apply some spool up if throttle not at 0% to avoid a sudden full power motor torque which can be dangerous in the air and on the ground. Comes to mind the AutoLaunch spool up when AL idle throttle is set >0 . Soft and fast enough, just needs to take the current stick throttle as a target.

MrD-RC, 2021-06-10:
@Mateyhv the spool up is a great idea. The setting from nav_launch could be used for the time.

Mateyhv, 2022-01-09:
Hmm, hope this issue does not fade out..

MrD-RC, 2022-01-09:
Beat me to it @Mateyhv This is definitely an issue that I was hoping would get a PR.

MrD-RC, 2022-06-18:
Now that https://github.com/iNavFlight/inav/pull/7270 has added `isFlightDetected()` to navigation.c. Maybe the 2 seconds to re-arm is something that can be revisited?

Jetrell, 2022-06-18:
@MrD-RC 
Maybe the `Landing detector check` could even be added to the logic conditions for a workaround.

MrD-RC, 2022-06-18:
@Jetrell Certainly the isFlying could be. I think the landing detection may have multiple stages to report. But they could be simple ints with explanations in the docs. I was thinking of adding the launch stages to programming. As that could be handy for automatic take off flaps.

ennob, 2022-08-14:
I just bumped my disarm switch today and lost a model... +1 for this feature to be added urgently.

k1g0r, 2023-01-23:
it's nice idea!

MrD-RC, 2023-11-19:
Fixed by #9254

closed 2023-11-19 by MrD-RC