iNavFlight/inav #6791: Incorrect Longitude displayed in DJI FPV goggles V2

kelleysislander, 2021-04-05:
I just got the V2 goggles and the longitude is incorrectly displayed, but it does follow waypoints fine, so the issue is just the displayed value.  Please see the attached images.

[image]
[image]
[image]
[image]

DzikuVx, 2021-04-05:
Since it's working in V1 and broken in V2, we have to assume it's a problem with V2 firmware, not INAV. Both types of goggles are fed the same data from INAV side

digitalentity, 2021-04-05:
Alternatively we can figure out what DJI expects and add a bit to `dji_worksarounds`. Question is - do we want to?

kelleysislander, 2021-04-06:
Hi,
Thanks for replying... Yes it is an unexpected data condition that the V2 goggles don't know how to handle, but V1's do.
It would be nice to have this displaying correctly.
My workaround is that I do have the coordinates displaying in my radio, so if I do go down I still have a way of seeing the last known position.
Cheers,
Bill
***@***.***

kelleysislander, 2021-04-10:
Ya, sorry for the duplicate ;)  I opened it originally under configurator as #6770 and a dev replied not a configurator issue so I opened it here. Should have closed that one...

pbreed, 2021-05-14:
I believe this is fixed in the latest V2 goggles software. 606 .

closed 2021-06-10 by DzikuVx