iNavFlight/inav #6804: Return to Home, At least, linear. Unusual current surges

brinuk, 2021-04-08:
****

## Current Behavior

When the quad is returning to home using the "at least, linear" option. It returns at 30km/h which has a current draw of 4-5 amps. The quad descends gradually on route. From time to time Inav appears to determine that the descent has been too much and the quad jumps up a metre or so, the current draw jumps up to 13-15 amps before returning to the normal 4-5 amps and the quad continues to descend gradually. This current surge plays havoc with the li-ion battery. This edited video shows the effect at 50% video speed

## Steps to Reproduce

1. Recorded the current surge on video edited to 50% playback speed. https://www.youtube.com/watch?v=tm4rgcwroI8

## Expected behavior

Expected that the quad will descend at a gradual rate on RTH without sudden increases in altitude and bursts of heavy current draw. 

## Suggested solution(s)

If the calculation of distance to home and altitude indicates that the descent has been too much, correct this by maintaining the current altitude until the distance to home reduces to the required figure. Do not increase the altitude of the quad.
## Additional context

https://pastebin.com/RzNu7m55
---

INAV/MAMBAF405 2.5.2 Aug  4 2020 / 10:42:50 (faaedc7cb)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
- FC Board name and vendor:
- INAV version string:

avsaase, 2021-04-08:
I noticed this as well. Does it only do this during RTH with this RTH mode?

brinuk, 2021-04-08:
Yes, only during RTH with this mode. The li-ion battery drops so much during this event that I have to take manual control and it then flies home at 30km/h, gradually descending, with about 5 amp or less current draw. I noticed on the FPV that it descended gradually for a while and then jumped up a metre or so, at which point the current draw went high. Then returned again to a gradual descent. I had assumed that it was connected with the distance to home/altitude calculation that Inav was making. Sent from Mail for Windows 10 From: Alexander van SaaseSent: Thursday, April 8, 2021 11:45 AMTo: iNavFlight/inavCc: Brian Lambert; AuthorSubject: Re: [iNavFlight/inav] Return to Home, At least, linear. Unusual current surges (#6804) I noticed this as well. Does it only do this during RTH with this RTH mode?

closed 2022-02-10 by DzikuVx