iNavFlight/inav #6809: Please add support for SPEEDYBEE F7 V2 FC target

jokeane, 2021-04-09:
## Current Behavior
It appears you already support the V2 SPEEDYBEE F7, but the SPEEDYBEE F7 V2 is slightly different.

## Desired Behavior
Support SPEEDYBEE F7 V2 please.

## Suggested Solution
Modify existing SPEEDYBEE F7 to make SPEEDYBEE F7 V2 target.

## Who does this impact? Who is this for?
Long range quad users. 

## Additional context
I've had really nice results with INAV so far on my 7" LR quads. Looking forward to replacing BF on this new FC. Thanks!

suhast222, 2021-04-24:
It would be very nice to have the speedybee f7 v2 target, any chance to get it. By 3.0?
Does the speedybee f7 work on the v2?
Regards.

krizw, 2021-05-03:
> It would be very nice to have the speedybee f7 v2 target, any chance to get it. By 3.0?
> Does the speedybee f7 work on the v2?
> Regards.

Speedybee confirmed today on FB that an Inav build is coming.

DzikuVx, 2021-08-23:
[image]

closed 2021-08-23 by DzikuVx