iNavFlight/inav #6810: Replace roll and pitch board alignment adjustments with pitch trim adjustment

avsaase, 2021-04-10:
In 3.0, the preferred method for getting the plane to fly level in angle mode is to adjust the pitch trim, not the board alignment.

rts18, 2021-04-10:
Without stepping on anyone's toes, I have to agree.

And I certainly don't think Pitch and Roll board alignment should have been removed from the multi-rotor setup?
Even though it wasn't as precise as the 1 degree increments in the CLI. I still think it was quicker and easier for the average setup.
Unless multi-rotor preset detect will be added before the final RC?

[image]

DzikuVx, 2021-04-10:
The settings are still available via CLI. Configurator is cludged with settings that are used by less than 1% of users

rts18, 2021-04-10:
@DzikuVx  I agree that the Configurator is cluttered with some settings that are rarely used.
But with all respect, From my years on RCG. I don't believe that this is one of those settings.

Many users find multi-rotor board and sensor alignment confusing. 
The Configurator is a means for them to visualize their setup. Having to do this in the CLI, will raise many future questions concerning board and magnetometer alignment in multi-rotors.

avsaase, 2021-04-11:
@rts18 do people really mount the fc in their quad upside down or sideways?

rts18, 2021-04-11:
> 
> 
>do people really mount the fc in their quad upside down or sideways?

In many 2' to 4" builds. I have noticed that its become a more common practice, due to the way some FC boards are layout. This makes it impossible to access the USB port in their designated orientation, unless the board is turn another direction. or even upside down in another direction.

closed 2022-01-28 by DzikuVx