iNavFlight/inav #6811: Resource mapping RSSI & CURRENT for FURYF4OSD is wrong

julianschweizer, 2021-04-10:
## Current Behavior
Target FURYF4OSD has wrong resource mapping. ADC current sensor does not work properly (wrong value), verified with circuit tester.

## Steps to Reproduce
1. Flash FURYF4OSD current sensor is not working properly (wrong value), verified with circuit tester
Resource mapping:
  RSSI : configured = ADC 3 (C03)
  CURRENT : configured = ADC 2 (C02)
2. Flash INAV MAMBAF405US current sensor is working (C03 and C02 switched other way round), verified with circuit tester
3. Flash Betaflight FURYF4OSD current sensor is working (C03 and C02 switched other way round), verified with circuit tester

## Expected behavior
Current sensor showing right current value instead of wrong value in accordance with with circuit tester.

## Suggested solution(s)
Switch resource mapping C03 and C02 for INAV target FURYF4OSD the other way round.

## Additional context
See discussion and screenshot @ https://intofpv.com/t-correct-current-scale-in-inav-not-possible?pid=136747#pid136747

FC: Mamba F405 MK2 (HW Rev. 3)
ESC: Mamba F40 MK2
Target: FURYF4OSD for Betaflight according to diatone.us website
INAV Version: 2.6.1
Configurator Version: 2.6.1

julianschweizer, 2021-04-11:
ok .... i tried editing  in target.h:

`#   define ADC_CHANNEL_2_PIN       PC3 
#   define ADC_CHANNEL_3_PIN       PC2 

#define VBAT_ADC_CHANNEL            ADC_CHN_1
#define CURRENT_METER_ADC_CHANNEL   ADC_CHN_2 
#define RSSI_ADC_CHANNEL            ADC_CHN_3`

which is in accordance with working BF target: 
https://github.com/betaflight/betaflight/blob/4.1-maintenance/src/main/target/FURYF4/target.h

no luck.

Any help/ideas appreciated ... .

SpencerMKSmith, 2021-06-09:
What's the next steps here? @julianschweizer why not just throw up a PR and see what people think?  I'm having this same issue..

geduxk, 2021-08-28:
same for me any solution to solve this bug ?

julianschweizer, 2021-08-29:
I did not PR it since I am not really sure how to. There is some ‚wired‘ (at least to me) if/else code for several hw revisions in the target.h and I was not able to figure out if the PR will make this worse for others.

 What I did is i compiled my own firmware (it’s easy) with the above target.h informations. Works great!

geduxk, 2021-08-29:
cant you compile for me :D 3.0.1

closed 2022-01-28 by DzikuVx

7subaru4, 2023-10-05:
Hi guys,
Same problem here, as the current seems to increase without any relation to idling or flying.
Droping :
#define ADC_CHANNEL_2_PIN PC2
#define ADC_CHANNEL_3_PIN PC3
#define VBAT_ADC_CHANNEL ADC_CHN_1
#define RSSI_ADC_CHANNEL ADC_CHN_2
#define CURRENT_METER_ADC_CHANNEL ADC_CHN_3

in the CLI did not changed anything. What did I do wrong ?

pki791, 2024-08-19:
Workaround in CLI:
```
set current_adc_channel = 3
set rssi_adc_channel = 2
save
```