iNavFlight/inav #6814: Enhance the mixer

MrD-RC, 2021-04-11:
We all know that there should be no mixing in the transmitter, and that everything should be done in iNav. For basic mixes, this can be achieved now. However, there are things missing from the mixer that could make it more useful to a wider range of models.

## What I'd like to see
- **Differential** The ability to change the throw in each direction of ailerons and elevons can have a positive impact on the handling of a plane or wing. Being able to add differential to reduce adverse yaw would be a great feature. This is a number between -100 and 100. 0 is neutral, -100 to 0 reduces throw in the < 0 range of the servo, and 0 to 100 reduces throw in the >0 range of the servo. __What could benefit?__ Any fixed wing without a symmetrical aerofoil.
- **Offset** We already have the ability to change the weights. Having the ability to change the offset would allow setting up flaps, flaperons, and spoilerons without any changes in the transmitter. This is a number between -100 and 100. __What could benefit__ Any fixed wing with a tail. I see quite a few posts of people using motor gliders now for iNav, due to their efficiency.

avsaase, 2021-04-11:
Good idea about the differential, although I always prefer to do set this mechanically as much as possible to void throwing away servo resolution. As a side note, implementing this will interfere with #6708 because it relies on a fixed mixer wright and servo rate. Not that we shouldn't implement it, just something to keep in mind.

Regarding the offset, I always setup flaperons etc. as a separate mixer rule driven by an RC channel. Does that not effectively equals an offset?

DzikuVx, 2021-04-11:
Actually, both of those things can be done right now without any code
changes:

1. Offset - add an extra servo rule with MAX input and custom weight
activated by a logic condition
2. Differential: 2 mixer rules with different weights activated by two
separate LC when one is true when stabilized pitch (example ) is > 0 and
the second one is inverted condition

Yes it's advanced usage, but both offset and differential would be used
only by power users

MrD-RC, 2021-04-11:
You both raise some good points.

@avsaase regarding #6708 So how would that work on for example elevon mixes with for example 60% on roll and 40% on pitch, like you could have on a plank? Or is it just that either side of centre on the axis is equal?

@avsaase and @DzikuVx point 1. You could set up a mixer rule with various logic conditions so that a switch in LOW activates a mixer rule that has a weight of -100, switch MID activates weight -50, and switch HIGH activates weight 0. I just think it would be a neater solution to have the offset, and not lose 3 lines of programming unnecessarily. It also doesn't allow for variable flaps, flaperons, or spoilerons on a slider. When you get people who like to slope soar using iNav, they will quite often use a slider (half the time the throttle stick) to control the flap or crow braking. Something that again would be simpler to have with an offset. I guess offsets could be done in the transmitter. But I thought the point was not to have any mixing in the transmitter, or do we only care about that for the 4 main control channels?

@DzikuVx point 2. Totally that is possible. But, people coming from a more tradition background in to iNav would find it simpler to use with a diff value in the mixer. They could understand transmitter programming to an extent, but struggle with iNav programming.

I get what you're saying about power users. But I think simple Tx programming features like this are less power user based than the whole iNav programming system. I know you're not saying that it shouldn't be implemented, because it's for more advanced pilots. Afterall, someone added programmable PID controllers ;) 

Thanks for the feedback guys :)

avsaase, 2021-04-11:
> @avsaase regarding #6708 So how would that work on for example elevon mixes with for example 60% on roll and 40% on pitch, like you could have on a plank? Or is it just that either side of centre on the axis is equal?

With differential? Idk, that's something that we would need to work out. I also didn't consider the possibility of multiple `stabilized roll` (for example) rules driving the same servo, where LC's determine when each rule is active. I think this may cause the associated servo center to be adjusted twice. Something I'll have to look into.

DzikuVx, 2021-04-11:
@MrD-RC I understand the "different way", but please remember, that adding those values to the standard mixer will complicated mixer set up for the remaining 99,5% of people that just don't care about differential throws.
Even OpenTX does not give you an option to setup differential throw out of the box without using curves, and offset is basically done the same way: by mixing additional input into the channel.

MrD-RC, 2021-04-11:
OpenTX gives easy access to diff in the mixer. Literally click edit on the mix line and it's the 6th element down. Out the box, it's there, just wind in the value. Even my old Spektrum DX6i piece of crap has differential easily accessible. I don't see how it would complicate the mixer set up for the 99.5% of people. They simply leave it on 0. Or they could add some, and have their planes fly better. Especially wings, when you have no rudder to counteract adverse yaw.

DzikuVx, 2021-04-11:
With FC differential is not that important and I stand in position that LC
way is just good enough for everyone who want to do it.

closed 2021-07-28 by DzikuVx