iNavFlight/inav #6815: Altitude reading decimal places

Energeezer, 2021-04-11:
Using Baro on 2.6 my altitude ladder is set to feet but only reads up tp 100 feet then starts at 1 again.

MrD-RC, 2021-04-11:
You need to move the OSD element. Some of the numbers are getting hidden by other elements.

Energeezer, 2021-04-11:
> You need to move the OSD element. Some of the numbers are getting hidden by other elements.

> You need to move 

> You need to move the OSD element. Some of the numbers are getting hidden by other elements.

OK. Sounds good. I'll give that a try. While I'm here is there a way to show distances in metric and altitude in feet? Seems the settings are global.

MrD-RC, 2021-04-11:
Try the UK settings. That has some things in imperial, others in metric. But I think it’s the other way around, distance in imperial and altitude in metres. I’m from the UK and prefer imperial. The only thing that would annoy me is temperature in F rather than C. I’d rather have distance and altitude in imperial though.

avsaase, 2021-05-12:
Looks like this is resolved.

closed 2021-05-12 by avsaase